LinuxCNC/linuxcnc #2206: HY_GT_VFD fault info register

persei802, 2022-12-14:
The HAL component HY_GT_VFD declares a HAL pin called hy_gt_vfd.fault-info as a FLOAT output. However, that register (0x5000) is a set of BIT values and should therefore be declared as a U32.  Then it will be possible for python to convert it to a hex value for display in a GUI.

petterreinholdtsen, 2022-12-14:
[Jim]
> The HAL component HY_GT_VFD declares a HAL pin called
> hy_gt_vfd.fault-info as a FLOAT output. However, that register
> (0x5000) is a set of BIT values and should therefore be declared as a
> U32.  Then it will be possible for python to convert it to a hex value
> for display in a GUI.

I am not sure, but src/hal/user_comps/hy_gt_vfd.c do not seem to list
fault-info as float.  How can the problem you describe be
detected/reproduced?

-- 
Happy hacking
Petter Reinholdtsen

persei802, 2022-12-15:
Not sure why it's not declared in the C file and I don't know where the header files are - I am not a C++ programmer. Anyway, the documentation says it's a FLOAT and when it's run in a GUI (qtdragon), HALSHOW shows it as a FLOAT. https://linuxcnc.org/docs/stable/html/man/man1/hy_gt_vfd.1.html
Jim

andypugh, 2022-12-15:
All the pins that correspond to a register are created as type float:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/hal/user_comps/hy_gt_vfd.c#L266
Including the fault register:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/hal/user_comps/hy_gt_vfd.c#L606

It's not _trivial_ to change the type of a particular hal pin, but neither is it particularly difficult. 
The  modbus_register_t; struct could contain a pin of type hal_u32_t added, and the handler could update the appropriate variable (possibly depending on which is non-null, though an explict "type" tag in each struct is probably better. Using a multiplier of zero would probably work as a tag too.

persei802, 2022-12-15:
Thanks Andy. I eventually figured that out upon closer study. I could mess around for days trying to figure out what you said. I don't see how the handler can convert a FLOAT to U32. Any chance you could make a modified file for me as a starting point? I can then test it out and post the results. I do think this should be implemented in the source as it is not currently possible to read the fault code bits.

petterreinholdtsen, 2022-12-16:
[Jim]
> Any chance you could make a modified file for me as a starting point?
> I can then test it out and post the results. I do think this should be
> implemented in the source as it is not currently possible to read the
> fault code bits.

I drafted a patch in pull request #2209.  Perhaps you can see if it work
as you want?
-- 
Happy hacking
Petter Reinholdtsen

persei802, 2022-12-16:
Thankyou Petter. I will test it once I figure out how to get the file. In the meantime, I have a python version that I wrote that functions perfectly well. I don't know if there is any interest from the developers in including it in linuxcnc, but if there is, I'd be glad to submit it for review.

PR #2209 (2022-12-19): Change HAL pin type for hy_gt_vfd.fault-info from float to u32
Restructure code to handle two types of HAL pin types for modbus register.

Fixes #2206.
commit 4128c3e0 (2022-12-19): Change HAL pin type for hy_gt_vfd.fault-info from float to u32

Restructure code to handle two types of HAL pin types for modbus register.

Fixes #2206.
closed 2022-12-19 by andypugh