LinuxCNC/linuxcnc #2210: gmoccapy.ini from config/sim has multiple HOMEs per joint

smoe, 2022-12-17:
For sections [JOINT_0], [JOINT_1] and [JOINT_2] (i.e. all axes for that default 3-axis mill) the HOME entry is given twice. For the first joint this is https://github.com/LinuxCNC/linuxcnc/blob/6fc2d5beab2cf6cc14793f7b34ed616eb9e64987/configs/sim/gmoccapy/gmoccapy.ini#L112 and https://github.com/LinuxCNC/linuxcnc/blob/6fc2d5beab2cf6cc14793f7b34ed616eb9e64987/configs/sim/gmoccapy/gmoccapy.ini#L123 . 

The effect is not that the second HOME assignment is overwritten, but it is an array forwarded to GMOCCAPY, triggering a respective error message when selecting that configuration.

That "double HOME" is also seen for the 4-axis variant (https://github.com/LinuxCNC/linuxcnc/blob/master/configs/sim/gmoccapy/gmoccapy_4_axis.ini#L100 and https://github.com/LinuxCNC/linuxcnc/blob/master/configs/sim/gmoccapy/gmoccapy_4_axis.ini#L111), the 5-axis ones are all fine, but messages has them (https://github.com/LinuxCNC/linuxcnc/blob/6fc2d5beab2cf6cc14793f7b34ed616eb9e64987/configs/sim/gmoccapy/gmoccapy_messages.ini#L142 and https://github.com/LinuxCNC/linuxcnc/blob/6fc2d5beab2cf6cc14793f7b34ed616eb9e64987/configs/sim/gmoccapy/gmoccapy_messages.ini#L153), and so do pedant (https://github.com/LinuxCNC/linuxcnc/blob/6fc2d5beab2cf6cc14793f7b34ed616eb9e64987/configs/sim/gmoccapy/gmoccapy_pedant.ini#L110 and https://github.com/LinuxCNC/linuxcnc/blob/6fc2d5beab2cf6cc14793f7b34ed616eb9e64987/configs/sim/gmoccapy/gmoccapy_pedant.ini#L121) and pyngcgui (https://github.com/LinuxCNC/linuxcnc/blob/6fc2d5beab2cf6cc14793f7b34ed616eb9e64987/configs/sim/gmoccapy/gmoccapy_pyngcgui.ini#L145 and https://github.com/LinuxCNC/linuxcnc/blob/6fc2d5beab2cf6cc14793f7b34ed616eb9e64987/configs/sim/gmoccapy/gmoccapy_pyngcgui.ini#L156)

@hansu?

hansu, 2022-12-17:
Interesting, that has been there for 10 years and nobody noticed that :D
With error message you an error printed to the console?
I think we should remove the duplicates to avoid confusion.

smoe, 2022-12-18:
Yes, you get an error that arrays cannot be handled and the two configured values are printed for each axis.

andypugh, 2022-12-19:
````
andypugh@rm-one:~/linuxcnc-dev/configs$ grep -r -Plz "HOME[^_]*=[^\[]*HOME[^_]*="
sim/gmoccapy/plasma_config/gmoccapy_plasma.ini
sim/gmoccapy/lathe_configs/lathe_imperial.ini
sim/gmoccapy/messages.ini
sim/gmoccapy/gmoccapy.ini
sim/gmoccapy/4_axis.ini
sim/gmoccapy/gmoccapy_plasma/gmoccapy_plasma.ini
sim/gmoccapy/gmoccapy_messages.ini
sim/gmoccapy/gmoccapy_4_axis.ini
sim/gmoccapy/gmoccapy_pedant.ini
sim/craftsman/craftsmancnc-axis.ini
sim/axis/vismach/puma/puma_cube.ngc
sim/gscreen/silverdragon/silverdragon.ini
````
Fixed by e714b3ac2cb11046050b6320dfa7e78984e12ce9

closed 2022-12-19 by andypugh