LinuxCNC/linuxcnc #2212: Clang 15 error with LinuxCNC

NTULINUX, 2022-12-18:
## Here are the steps I follow to reproduce the issue:

 1. Build LinuxCNC with Clang
 2. make
 3. wait

## This is what I expected to happen:

Successful build (worked a few months ago, maybe something changed with Clang 15?

## This is what happened instead:

error: unknown warning option `-Wno-stringop-truncation`; did you mean `-Wno-string-concatenation`? [-Werror,-Wunknown-warning-option]

## It worked properly before this:

LinuxCNC from 4 months ago with Clang 14

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Custom Gentoo
 * I am using this kernel version (shown by `uname -a`): 6.1
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [X] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): 6fc2d5beab2cf6cc14793f7b34ed616eb9e64987
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): N/A
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): N/A

NTULINUX, 2022-12-18:
Rats... another one just popped up:

Compiling realtime objects/hal/components/sim_home_switch.c
hal/components/reset.comp:76Compiling realtime objects/hal/components/sim_matrix_kb.c
:5: error: void function '_' should not returnCompiling realtime objects/hal/components/sim_parport.c
 a value [-Wreturn-type]
    return 0;

Not sure how to properly fix this one..

NTULINUX, 2022-12-18:
I don't know if -Wno-unknown-warning-option is the right fix here, feel free to amend or make suggestions!

petterreinholdtsen, 2022-12-18:
The correct fix for reset is to drop the return statement, I believe.

petterreinholdtsen, 2022-12-18:
[NTU]
> I don't know if -Wno-unknown-warning-option is the right fix here,
> feel free to amend or make suggestions!

I suspect a better fix is to move the GCC only options to
src/configure.ac and only activate activate it for gcc.

-- 
Happy hacking
Petter Reinholdtsen

NTULINUX, 2022-12-18:
> [NTU]
> I don't know if -Wno-unknown-warning-option is the right fix here, feel free to amend or make suggestions!
> I suspect a better fix is to move the GCC only options to src/configure.ac and only activate activate it for gcc.

Unfortunately I can't test GCC builds because my entire system is built against Clang. There's usually a lot of compiling errors and undefined references when you build code that links to other libraries (especially Boost) with a different compiler.

petterreinholdtsen, 2022-12-18:
[NTU]
> Unfortunately I can't test GCC builds because my entire system is
> built against Clang. There's usually a lot of compiling errors and
> undefined references when you build code that links to other libraries
> (especially Boost) with a different compiler.

Do not really need GCC to do this, but I decided to give it a go.  My
branch to test clang now include a few fixes to get the clang build
working, in conflict with your proposed changes.

--
Happy hacking
Petter Reinholdtsen

petterreinholdtsen, 2022-12-18:
This issue was fixed in 5a5786b1bf1e0d31974346c4bdeab001f55b4959 .

closed 2022-12-18 by petterreinholdtsen

NTULINUX, 2022-12-19:
> [NTU]
> Unfortunately I can't test GCC builds because my entire system is built against Clang. There's usually a lot of compiling errors and undefined references when you build code that links to [...]

What I meant is that without testing the GCC build after said changes, I'm liable to break it. You're much better at this than me because I wouldn't have done as good a job. Clang build works now, thank you!

petterreinholdtsen, 2022-12-19:
[NTU]
> What I meant is that without testing the GCC build after said changes,
> I'm liable to break it. You're much better at this than me because I
> wouldn't have done as good a job. Clang build works now, thank you!

I have more confidence in you that you do, apparently. :)

Anyway, now github CI will validate clang build for every commit and
pull request, allowing us to discover such build and runtime errors
earlier next time.

-- 
Happy hacking
Petter Reinholdtsen