LinuxCNC/linuxcnc #2218: Several typos discovered by Debian lintian

petterreinholdtsen, 2022-12-20:
The debian build tool lintian discovered several typos in the source debian package.  Here is a script to fix the detected issues.  I do not see the point of only running it on the source repository, as several of the typos are in files not present in the linuxcnc repository.

```
#!/bin/sh
#
# Fix several typos discovered by lintian in the linuxcnc source package:
#
# W: linuxcnc-doc-es: spelling-error-in-changelog embeded embedded
# W: linuxcnc-doc-es: spelling-error-in-changelog Portugese Portuguese
# W: linuxcnc-doc-es: spelling-error-in-changelog more more (duplicate word) more
# W: linuxcnc-doc-es: spelling-error-in-changelog vesion version
# W: linuxcnc-doc-es: spelling-error-in-changelog unecessary unnecessary
# W: linuxcnc-doc-es: spelling-error-in-changelog commmand command
# W: linuxcnc-doc-es: spelling-error-in-changelog authorative authoritative
# W: linuxcnc-doc-es: spelling-error-in-changelog interpeter interpreter
# W: linuxcnc-doc-es: spelling-error-in-changelog crtical critical
# W: linuxcnc-doc-es: spelling-error-in-changelog consistant consistent
# W: linuxcnc-doc-es: spelling-error-in-changelog overriden overridden
# W: linuxcnc-doc-es: spelling-error-in-changelog Futher Further
# W: linuxcnc-doc-es: spelling-error-in-changelog hightlighting highlighting
# W: linuxcnc-doc-es: spelling-error-in-changelog folowing following
# W: linuxcnc-doc-es: spelling-error-in-changelog incase in case
# W: linuxcnc-doc-es: spelling-error-in-changelog mutiple multiple
# W: linuxcnc-doc-es: spelling-error-in-changelog upto up to
# W: linuxcnc-doc-es: spelling-error-in-changelog occurences occurrences
# W: linuxcnc-doc-es: spelling-error-in-changelog occassionally occasionally
# W: linuxcnc-doc-es: spelling-error-in-changelog substract subtract
# W: linuxcnc-doc-es: spelling-error-in-changelog possibilty possibility
# W: linuxcnc-doc-es: spelling-error-in-changelog missmatch mismatch
# W: linuxcnc-doc-es: spelling-error-in-changelog documention documentation
# W: linuxcnc-doc-es: spelling-error-in-changelog relatd related
# W: linuxcnc-doc-es: spelling-error-in-changelog formated formatted
# W: linuxcnc-doc-es: spelling-error-in-changelog calulation calculation
# W: linuxcnc-doc-es: spelling-error-in-changelog menues menus
# W: linuxcnc-doc-es: spelling-error-in-changelog inadvertantly inadvertently
# W: linuxcnc-doc-es: spelling-error-in-changelog surounding surrounding
# W: linuxcnc-doc-es: spelling-error-in-changelog commited committed
# W: linuxcnc-doc-es: spelling-error-in-changelog accessable accessible
# W: linuxcnc-doc-es: spelling-error-in-changelog follwing following
# W: linuxcnc-doc-es: spelling-error-in-changelog intial initial
# W: linuxcnc-doc-es: spelling-error-in-changelog lables labels
# W: linuxcnc-doc-es: spelling-error-in-changelog exising existing
# W: linuxcnc-doc-es: spelling-error-in-changelog openning opening
# W: linuxcnc-doc-es: spelling-error-in-changelog documenation documentation
# W: linuxcnc-doc-es: spelling-error-in-changelog distrubution distribution
# W: linuxcnc-doc-es: spelling-error-in-changelog requiere require
# W: linuxcnc-doc-es: spelling-error-in-changelog temparary temporary
# W: linuxcnc-doc-es: spelling-error-in-changelog consistancy consistency
# W: linuxcnc-doc-es: spelling-error-in-changelog intialization initialization
# W: linuxcnc-doc-es: spelling-error-in-changelog transistion transition
# W: linuxcnc-doc-es: spelling-error-in-changelog guage gauge
# W: linuxcnc-doc-es: spelling-error-in-changelog bultin builtin
# W: linuxcnc-doc-es: spelling-error-in-changelog compatiblity compatibility
# W: linuxcnc-doc-es: spelling-error-in-changelog exemple example
# W: linuxcnc-doc-es: spelling-error-in-changelog informtion information
# W: linuxcnc-doc-es: spelling-error-in-changelog accelleration acceleration
# W: linuxcnc-doc-es: spelling-error-in-changelog allows to allows one to
# W: linuxcnc-doc-es: spelling-error-in-changelog Allow to Allow one to

replace() {
    key="$1"
    sed="$2"
    sed -i "$sed" $(grep -rl "$key" bin configs COPY* debian docs gnupg/ include ISSUE* lib meson* nc_files README* rtlib scripts share src tcl tests TODO VERSION)
}

replace embeded 's/\<embeded\>/embedded/g'
replace Portugese 's/\<Portugese\>/Portuguese/g'
replace 'more more' 's/\<more more\>/more/g'
replace vesion 's/\<vesion\>/version/g'
replace unecessary 's/\<unecessary\>/unnecessary/g'
replace commmand 's/\<commmand\>/command/g'
replace authorative 's/\<authorative\>/authoritative/g'
replace interpeter 's/\<interpeter\>/interpreter/g'
replace crtical 's/\<crtical\>/critical/g'
replace consistant 's/\<consistant\>/consistent/g'
replace overriden 's/\<overriden\>/overridden/g'
replace Futher 's/\<Futher\>/Further/g'
replace hightlighting 's/\<hightlighting\>/highlighting/g'
replace folowing 's/\<folowing\>/following/g'
replace incase 's/\<incase\>/in case/g'
replace mutiple 's/\<mutiple\>/multiple/g'
replace upto 's/\<upto\>/up to/g'
replace occurences 's/\<occurences\>/occurrences/g'
replace occassionally 's/\<occassionally\>/occasionally/g'
replace substract 's/\<substract\>/subtract/g'
replace possibilty 's/\<possibilty\>/possibility/g'
replace missmatch 's/\<missmatch\>/mismatch/g'
replace documention 's/\<documention\>/documentation/g'
replace relatd 's/\<relatd\>/related/g'
replace formated 's/\<formated\>/formatted/g'
replace calulation 's/\<calulation\>/calculation/g'
replace menues 's/\<menues\>/menus/g'
replace inadvertantly 's/\<inadvertantly\>/inadvertently/g'
replace surounding 's/\<surounding\>/surrounding/g'
replace commited 's/\<commited\>/committed/g'
replace accessable 's/\<accessable\>/accessible/g'
replace follwing 's/\<follwing\>/following/g'
replace intial 's/\<intial\>/initial/g'
replace lables 's/\<lables\>/labels/g'
replace exising 's/\<exising\>/existing/g'
replace openning 's/\<openning\>/opening/g'
replace documenation 's/\<documenation\>/documentation/g'
replace distrubution 's/\<distrubution\>/distribution/g'
replace requiere 's/\<requiere\>/require/g'
replace temparary 's/\<temparary\>/temporary/g'
replace consistancy 's/\<consistancy\>/consistency/g'
replace intialization 's/\<intialization\>/initialization/g'
replace transistion 's/\<transistion\>/transition/g'
replace guage 's/\<guage\>/gauge/g'
replace bultin 's/\<bultin\>/builtin/g'
replace compatiblity 's/\<compatiblity\>/compatibility/g'
replace exemple 's/\<exemple\>/example/g'
replace informtion 's/\<informtion\>/information/g'
replace accelleration 's/\<accelleration\>/acceleration/g'
replace 'allows to' 's/\<allows to\>/allows one to/g'
replace 'Allow to' 's/\<Allow to\>/Allow one to/g'
```

The runtime on my machine is around 1 minute.

smoe, 2022-12-25:
I am fine with correcting typos also for earlier releases of LinuxCNC.

andypugh, 2022-12-25:
The problems seem clear, but I am not sure what solution you are proposing? 
Are you saying that this can't be a PR because some of the files are generated? In that case the spelling issue are likely to be in commit comments, and I am not sure that it makes sense to try to correct those.

petterreinholdtsen, 2022-12-25:
[andypugh]
> The problems seem clear, but I am not sure what solution you are
> proposing?  Are you saying that this can't be a PR because some of the
> files are generated?

I believe at least the debian/changelog file are is a different git
repostory.  So those who know which repo should run the script.

-- 
Happy hacking
Petter Reinholdtsen

andypugh, 2022-12-25:
This one? https://github.com/LinuxCNC/linuxcnc/blob/master/debian/changelog ?

petterreinholdtsen, 2022-12-25:
[andypugh]
> This one?
> https://github.com/LinuxCNC/linuxcnc/blob/master/debian/changelog ?

Nope, do not have all the typos reported by lintian.

-- 
Happy hacking
Petter Reinholdtsen

andypugh, 2022-12-26:
Looking further, all the examples I tried looking for appeared in commit messages, and I suspect that this is where Lintian is finding them. 
I have done some googling and I get the feeling that rewriting the commit history to fix typos would probably be a bad idea. 

I am curious why it is only finding the typos in linuxcnc-doc-es ?

andypugh, 2022-12-26:
If the suggestion is to run this script on the actual .deb file after it has been generated, I think this would require it to be unpacked, edited and re-packed (with the MD5 sums updated accordingly)

petterreinholdtsen, 2022-12-26:
[andypugh]
> If the suggestion is to run this script on the actual .deb file after
> it has been generated, I think this would require it to be unpacked,
> edited and re-packed (with the MD5 sums updated accordingly)

The suggestion is to run the script to edit the debian/changelog file
I assume is in some git repo somewhere.  If it is generated from scratch
every upload, this approach will not work.

> I have done some googling and I get the feeling that rewriting the
> commit history to fix typos would probably be a bad idea.

Yes, it is a horrible idea which will cause problems for all developers
and other users of the git repository.

> I am curious why it is only finding the typos in linuxcnc-doc-es ?

No idea.

I guess a good start is to fix the typos in the linuxcnc git repository,
and fix the remaining ones in debian/changelog if possible.
--
Happy hacking
Petter Reinholdtsen

andypugh, 2022-12-26:
Where do you see this Lintian report?

andypugh, 2022-12-26:
http://buildbot.linuxcnc.org/buildbot/builders/4041.deb-buster-rtpreempt-amd64/builds/2432/steps/shell_4/logs/warnings%20%2849%29

For example, only complains about "embeded" (so I have fixed that)