LinuxCNC/linuxcnc #2222: PnCconf needs some eye candy / user directions

smoe, 2022-12-25:
I just gave my new 7I96s a spin with PnCconf. It technically worked, but it did not feel good to use. My expectations for that tool are likely a bit off, but here we go with my noob's concerns:
 * it was not clear if the card should be connected or not - no need, but it would nonetheless be nice to state early.
 * "no firmware downloaded" should also show where to download them
 * inconsistent spelling of PnCconf and Mesa cards, i.e. lower-case "i" different from Mesa web pages
 * documentation on https://linuxcnc.org/docs/devel/html/config/pncconf.html describes the user interface but does not guide the user
 * help within PnCconf is not context-sensitive, shows nice diagrams and is not guiding the user either
 * missing a description of a workflow to approach the setup prior to a connection to the mill/lathe
 * the overall design was problematic for me since e.g. the many 0s for the servo thread's periodicity did not all fit into the textbox reserved for it, but the window has plenty of gray empty space available.
 * text labels have no spacer to the borders of the man window (the documentation shows such padding, though)

What now? Is there an agreement that PnCconf should make the first installation a bit easier? Since PnCconf and LinuxCNC are unlikely to be run at the same time - should we be a bit wasteful with resources like images and pointers to the documentation on the internet? Or can we somehow share a documentation between what is shown in PnCconf and what is presented as a PDF?

c-morley, 2022-12-25:
It never occurred to me that people would think the card needed to be connected, though in fact that (optional) capability has been programmed in later. i certainly seen many people do think it does need to be connected though.

There is contextual help. Press the help button on different pages usually gives different information. 
Maybe this should work differently?

mesa cards are spelled inconsistently over the years - probably could hide that though.

entrybox problems - The conversion to python3/gtk3 probably wasn't perfect.

Dev testing with feedback as you did, really helps.

Pncconf is a super useful (when it works properly) yet is largely a neglected program.
It is a critical part of the user experience.

I understand JT has built a configuration program outside of linuxcnc.
If we are not interested in working on pncconf maybe we should see if JT is interested in moving his project into linuxcnc.

If we are interested in pncconf then maybe it's time to reassess what we want it to do and how to get there.

silopolis, 2022-12-27:
Hi,

Thanks Steffen for this report, very much needed 👍

Le dim. 25 déc. 2022 à 22:55, c-morley ***@***.***> a écrit :

> Dev testing with feedback as you did, really helps.
>
Absolutely

Pncconf is a super useful (when it works properly) yet is largely a
> neglected program.
> It is a critical part of the user experience.
>
1000%

I understand JT has built a configuration program outside of linuxcnc.
> If we are not interested in working on pncconf maybe we should see if JT
> is interested in moving his project into linuxcnc.
>
> If we are interested in pncconf then maybe it's time to reassess what we
> want it to do and how to get there.
>
1000% again!

@smoe that'd be awesome if you could give @jethornton/mesact a try a
provide a comparative UX report to feed that discussion 🙏

Hope yall had a nice Christmas and have a well recovering lever 😅😘

smoe, 2022-12-28:
I had in the meantime overcome the problem to run AXIS and GMOCCAPY on bookworm/unstable (https://github.com/LinuxCNC/linuxcnc/issues/1599), so I had them giving me some first light, too. The eye-candy problem with labels not placed where I like them also affects those applications. I need to investigate this more in the upcoming weeks. In parts this likely is triggered by different lengths of words once translated, but again, I need to chase this up myself a bit more and then see about what I can come up with. I am just a bit swamped with things to do and (guess why it looks like it looks) those eye candies are not the highest priority for me :)

I had a look at mesact and think we should also package it for Debian. I had other problems with it, like the window's height not fitting on my monitor. Somehow our web pages are more reactive to screen sizes than our user applications these days. And weirdly enough, I found PnCconf to be more intuitive at a first sight. But I liked it.

We should also somehow mention mesact in the LinuxCNC documentation. Just - where? I am still stuck with my description of HAL :)

smoe, 2023-01-10:
To give you an impression.
I see the upper left corner of axis_mm as in
[image]
which has "Achse" too much oriented to the left
(start as "LANG=de linuxcnc" to reproduce)
while LANG=C linuxcnc gives me the expected setup
[image]
I have removed the glBitmap instructions to prevent the crash on Debian bookworm. Since the English variant seems fine, I do not think this to be the problem. No idea where the missing colon went. I presume that the corresponding weblate entry was already corrected.