LinuxCNC/linuxcnc #2226: .deb: Cannot find the linuxcnc Python module

smoe, 2022-12-27:
Without that module, any "import linuxcnc" fails in Python. Any idea anyone?

In our source tree, it is located in lib/python/, so to help myself I just bent PYTHONPATH to my local build, worked, but ... am I missing something?

@petterreinholdtsen ?

Thanks!
Steffen

petterreinholdtsen, 2022-12-27:
[Steffen Möller]
> Without that module, any "import linuxcnc" fails in Python. Any idea
> anyone?

I see it using 'dpkg -L linuxcnc-uspace|grep -i dist-packages'.  Is it
not there on your machine?

-- 
Happy hacking
Petter Reinholdtsen

smoe, 2022-12-27:
I eventually found
`/usr/lib/python3/dist-packages/linuxcnc.cpython-310-x86_64-linux-gnu.so`
and on a new day in a test script the "import linuxcnc" just works.
Thank you!

closed 2022-12-27 by smoe

SebKuzminsky, 2022-12-27:
Possibly a python version mismatch, 3.9 vs 3.10?