LinuxCNC/linuxcnc #2227: LCNC 2.8 - PNCconf - Error when more 6 SSerial

zz912, 2022-12-27:
Hello,

I made next configuration for PNCconf.
I made 7i96_7i74d.bin
https://forum.linuxcnc.org/27-driver-boards/47764-ask-7i96-7i74d-bin

In private_data.py I added:
`    ['7i96-Internal Data', '7i96', '7i96_7i74d', '7i96', 'hm2_eth',
        1,3, 0,0, 0,3, 0,0, 5,2, 1,8, [],0,0,0,0,0,0,0, 1, 34, 33, 200, [1, 2, 3],
        # TAB 1
        [S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],
        [S.GPIOI, 0],[S.SSR0, 100],[S.SSR0, 101],[S.SSR0, 102],[S.SSR0, 103],[S.SSR0, 104],[S.SSR0, 105],
        [S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],
        # TAB 2
        [S.STEPB,0],[S.STEPA,0],[S.STEPB,1],[S.STEPA,1],[S.STEPB,2],[S.STEPA,2],[S.STEPB,3],[S.STEPA,3],[S.STEPB,4],[S.STEPA,4],
        [S.ENCA,0],[S.ENCB,0],[S.ENCI,0],[S.RXDATA0,0],[S.TXDATA0,0],[S.TXEN0,0],[S.NUSED,0],
        [S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],
        # TAB 3
        [S.RXDATA1, 0],[S.RXDATA2, 0],[S.RXDATA3, 0],[S.RXDATA4, 0],[S.TXDATA1, 0],[S.TXDATA2, 0],[S.TXDATA3, 0],[S.TXDATA4, 0],[S.RXDATA5, 0],[S.RXDATA6, 0],
        [S.RXDATA7, 0],[S.NUSED, 0],[S.TXDATA5, 0],[S.TXDATA6, 0],[S.TXDATA7, 0],[S.NUSED, 0],[S.NUSED, 0],
        [S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],],`
but it did not working.
It types:
[image]

So I made configuration only for 6 SSerial and it is working.
`    ['7i96-Internal Data', '7i96', '7i96_7i74d', '7i96', 'hm2_eth',
        1,3, 0,0, 0,3, 0,0, 5,2, 1,6, [],0,0,0,0,0,0,0, 1, 34, 33, 200, [1, 2, 3],
        # TAB 1
        [S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],
        [S.GPIOI, 0],[S.SSR0, 100],[S.SSR0, 101],[S.SSR0, 102],[S.SSR0, 103],[S.SSR0, 104],[S.SSR0, 105],
        [S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],
        # TAB 2
        [S.STEPB,0],[S.STEPA,0],[S.STEPB,1],[S.STEPA,1],[S.STEPB,2],[S.STEPA,2],[S.STEPB,3],[S.STEPA,3],[S.STEPB,4],[S.STEPA,4],
        [S.ENCA,0],[S.ENCB,0],[S.ENCI,0],[S.RXDATA0,0],[S.TXDATA0,0],[S.TXEN0,0],[S.NUSED,0],
        [S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],
        # TAB 3
        [S.RXDATA1, 0],[S.RXDATA2, 0],[S.RXDATA3, 0],[S.RXDATA4, 0],[S.TXDATA1, 0],[S.TXDATA2, 0],[S.TXDATA3, 0],[S.TXDATA4, 0],[S.RXDATA5, 0],[S.NUSED, 0],
        [S.NUSED, 0],[S.NUSED, 0],[S.TXDATA5, 0],[S.NUSED, 0],[S.NUSED, 0],[S.NUSED, 0],[S.NUSED, 0],
        [S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],],`

zz912, 2022-12-30:
I am looking for my Issue report again, and I see, that I explained it not enough.

I want  make next configuration for PNCconf. Now for 7i96 card with 7i74 card, later for 7i96S card with 7i74 card.

The card 7i96 can use 8 SSerial channels, 1 port.
1 SSerial is used in 7i96 and 7 SSerial is used in 7i74.

The card 7i96S can use 9 SSerial channels, 2 port.
1 SSerial is used in 7i96 and 8 SSerial is used in 7i74.
 
I found out that I am the first to need more than 6 SSerial in PNCconfig.

When I did the first configuration for all 8 SSerials, it started reporting an error when running in PNCconfig, see the image above.

In order to work now, I created a configuration with 6 SSerial. This configuration works.

I would like ask for help with solving the problem, because I would like to do the correct configuration for 7i96_7i74 and for 7i96S_7i74. After that I would like to make a Pull Request for these configurations.

I asked PCW how many cards was made. He wrote me:
"There have been about 3000 7I96 +7I96S's made so far."

The PNCconf is fantastic helper for me. 

I think this configuration would also help other 7i96(S) card users who would like to buy a 7i74 card to go with their card.

I know PNCconf won't solve all problems, but it can be used in two ways.

Previously, I only used PNCconf to create a skeleton, which I modified.

Today I am trying to configure the LCNC so that anything can be edited in PNCconfig at any time. I put everything that cannot be done in PNCconfig into the custom.hal file.

c-morley, 2022-12-31:
This is a big problem. The way pncconf was designed does not scale very well.
In the glade file; mesa0 and mesa1, the combobox names are used to identify the serial port and serial channel.
ie. mesa0sserial0_5pin13 = mesa card 0, sserial port 1, channel 6

These comboboxes are prebuilt rather then made at run time.
currently there are only comboboxes for channel 1-6 on port 1
Adding 2 more channels is not too bad but adding all possible sserial ports/channels to pncconf would have the program much much bigger.

It would have been better to use a general name scheme for the comboboxes and  code the port/channel numbers later at run time but now that would be a huge job.

I'm not sure what to do.

zz912, 2022-12-31:
Maybe it would be enough for me if the error message was just removed and the user was informed that the next SSerial should be filled in custom.hal. For me, it is important that PNCconf generates the correct configuration files for LCNC. Mainly the "loadrt hostmot2 ....." command to be fine. Missing comboboxes for SSerial >6 are acceptable to me. What do you think?

c-morley, 2023-01-01:
Ok in 2.9 I trapped the error.
Now it will tell you it partially failed but will let you continue.
I added your sample firmware

c-morley, 2023-01-01:
Ok I added 2 more sserial tabs now there are 8. Please let me know of any problems you encounter.

zz912, 2023-01-01:
I will look at it and I will give you feedback.

c-morley, 2023-01-01:
Do you have a firmware data for the 7i95s that uses 2 ports? I could use it to test with.
The PIN file would work too.

zz912, 2023-01-02:
Do you think 7i96s?  Use standard firmware for 7i96s and 7i74 in mesanet store net. PCW explane it here: https://forum.linuxcnc.org/27-driver-boards/47764-ask-7i96-7i74d-bin#260422

c-morley, 2023-01-03:
> Do you think 7i96s? Use standard firmware for 7i96s and 7i74 in mesanet store net. PCW explane it here: https://forum.linuxcnc.org/27-driver-boards/47764-ask-7i96-7i74d-bin#260422

To test, I need the PIN file. I don't have the 7i96s board so cannot create the PIN file mayself.
You can create the PIN file with MesaFlash, if you are familiar with it.
 I believe the --readhmid and --sserial options is the info i need.

zz912, 2023-01-03:
I downloaded it from:
http://www.mesanet.com/software/parallel/7i96s.zip
[7i96s_7i74d.pin.txt](https://github.com/LinuxCNC/linuxcnc/files/10334881/7i96s_7i74d.pin.txt)

c-morley, 2023-01-03:
Thank you. Now I see the bulk of the sserials is on port 1.
It would require me to add 8 more tabs each with 60 pins.
I think this is too much. I'll think some more on it.

zz912, 2023-01-03:
I wanted try your changes in VirtualPC, but PNCconf dont work now.
`zdenek@CNC:~/linuxcnc/linuxcnc-2.9_2023-01-03/bin$ ./pncconf
Traceback (most recent call last):
  File "./pncconf", line 52, in <module>
    from multifilebuilder import MultiFileBuilder
ModuleNotFoundError: No module named 'multifilebuilder'`

I use RIP instalation. Branche 2.9

c-morley, 2023-01-03:
Did you run:
./scripts/rip-environment 
before running pncconf?

Less likely:
Did you use make clean before make? If you compile  2.8 and 2.9 from the same GIT, that might be the problem.

zz912, 2023-01-03:
I dont use git. I only download branche in zip.

rip-environment not work too.

`zdenek@CNC:~/linuxcnc/linuxcnc-2.9_2023-01-03/scripts$ ./rip-environment 
This script can be used in one of two ways.

It can be loaded in the context of your shell, by executing
    . ./rip-environment
after this, commands like 'linuxcnc', 'halrun' and so on refer to the
version in this directory, instead of to an installed version of
linuxcnc (if any)

Second, it can be used to run a command in this directory without
modifying the environment of the calling shell:
    ./rip-environment command...
such as
    ./rip-environment halrun foo.hal
    ./rip-environment linuxcnc configs/.../foo.ini
zdenek@CNC:~/linuxcnc/linuxcnc-2.9_2023-01-03/scripts$ pncconf
bash: pncconf: příkaz nenalezen
zdenek@CNC:~/linuxcnc/linuxcnc-2.9_2023-01-03/scripts$ linuxcnc
bash: linuxcnc: příkaz nenalezen
zdenek@CNC:~/linuxcnc/linuxcnc-2.9_2023-01-03/scripts$ halrun
bash: halrun: příkaz nenalezen
zdenek@CNC:~/linuxcnc/linuxcnc-2.9_2023-01-03/scripts$ 
`

c-morley, 2023-01-04:
Sorry I assumed you compiled. I think you will need to run a compiled or installed version.

zz912, 2023-01-04:
I apologize for the confusion. Yesterday LCNC 2.9 started working properly. I don't know what I did wrong. Yesterday I managed to switch from LCNC 2.8 to 2.9 using PNCconf. I'll look into it more tonight.

zz912, 2023-01-04:
I tested your PNCconf modification. It looks good.
[image]
But SS# 6 is blank.
[image]

zz912, 2023-01-04:
I noticed that the second time I modified the parameters in PNCconf, that some boxes turned black.
[image]

c-morley, 2023-01-05:
Ok thanks - I think I got that sserial 6 tab fixed now.

zz912, 2023-01-10:
Hi Chris,

my mesa 7i84 just arrived (5pieces).  I could verify the functionality of the 7i96s+7i74+7i84 configuration. Could you please add the 7i96s+7i74 configuration to private_data.py? I understood from previous posts that you already have it ready.

I know it bothers you that there are many tables in pncconf with this configuration. I don't see a problem with that. If the 7i96s+7i74 configuration will work, I would consider this issue solved.

I assume a normal person will configure max 2 SS via PNCconf and write the rest in custom.hal. I only needed the 7i96+7i74 and 7i96S+7i74 configurations to not generate error messages and you solved that.

c-morley, 2023-01-10:
I never tested The 7i96s_7i74. It still seems to error I will try to fix that.
Since the bulk of the pins are on port 1 rather then port 0, most of the pins are not configurable.
Adding tabs slows the mesa page functions down. It also makes the GLADE files huge.
 There are probably ways to mitigate these problems if someone is interested.

zz912, 2023-01-11:
I tested it. All boxes are suppressed.
[image]

c-morley, 2023-01-11:
Yes that is expected, those sserials are on port 1.
Only port 0 can be configured currently on pncconf.
I'm investigating how difficult it will be to work around the problem.

c-morley, 2023-01-13:
I did some work and in principle got some rough code to be able to configure up to a max of 9 channels total with two ports.
Long ways from done, but is possible.

zz912, 2023-12-30:
Now 7i96 or 7i96s with 7i74 can be configured by PNCconf. 7i74 has 8 SSerial. So I am closing this Issue.

closed 2023-12-30 by zz912