LinuxCNC/linuxcnc #2228: raster.comp does not appear to compile with RTAI.

andypugh, 2022-12-28:
Running a git checkout pf LinuxCNC master with RTAI kernel 4.19.195-rtai-amd64

Commit 56367af8cb3d467cc22a95e20c58989222aef232 added a component to stream data to a laser using hal_port. 

It contains
````
include <stdlib.h>
````
Which appears to cause problems if building for RTAI. 
````
  CC [M]  /home/andypugh/linuxcnc-dev/src/objects/hal/components/raster.o
/home/andypugh/linuxcnc-dev/src/objects/hal/components/raster.c:14:10: fatal error: stdlib.h: No such file or directory
 #include <stdlib.h>
          ^~~~~~~~~~
````
@curtdutt Any thoughts? I guess you don't have a system to test this?

petterreinholdtsen, 2022-12-28:
[andypugh]
> It contains
> ````
> include <stdlib.h>
> ````
> Which appears to cause problems if building for RTAI.
> ````
>   CC [M]  /home/andypugh/linuxcnc-dev/src/objects/hal/components/raster.o [...]

I had a look, and see that <stdlib.h> is used for strtol(), strtoul()
and strtod().

-- 
Happy hacking
Petter Reinholdtsen

andypugh, 2022-12-28:
Yes. We have simple_strtol in rtapi
https://github.com/LinuxCNC/linuxcnc/blob/2.9/src/rtapi/rtapi.h#L835
But nothing equivalent to strtoul and strtod.

curtdutt, 2022-12-28:
I'll check it out. We could add some strto.... equivalents to rtapi or make
some changes to the component.

andypugh, 2022-12-28:
The RTAI build can use kstrtol (and friends) but there is no string-to-double (functions found in usr/src/linux-headers......./include/linux )
But those don't work for uspace (which is why rtapi defines them for uspace builds in rtapi.h

It almost might be simpler to just write one..
````
val = 0
bool point = False
bool sign = False
switch (char) 
    case '0':
    case '1':
    (case.....)
        if (! point) val = val * 10 + (char - '0');
   case '.':
   case ',': // For Germans etc
       point = True;
   case '-':
       sign = True;
   case else
       panic();
}
````

curtdutt, 2022-12-28:
Maybe add an rtapi_sdlib.h which is where we could either supply stdlib.h
functions for uspace, kernel functions for realtime code, or an actual
implementation if none exists?

Perhaps kernel hasn't included kstrotod because no one has needed it up
until now?

curtdutt, 2023-01-10:
I have a debian buster machine that I can install an RTAI kernel on to get
this worked out. I'm not up to date with the latest  Andy do you have a
link to a page for the process you follow to build RTAI kernels?

andypugh, 2023-01-11:
You should only need to do this: 
http://linuxcnc.org/docs/stable/html/getting-started/getting-linuxcnc.html#cha:Installing-RTAI
There are pre-built RTAI kernel packages. (and matching LinuxCNC, but you will be compiling from source anyway so don't need that)

andypugh, 2023-01-11:
In kernel space there are kstrto* functions for integer and boolean types, but nothing for float.
For the purposes of this component I think that a simple decimal number parser would probably work as we (you) have control of the data source too.

petterreinholdtsen, 2023-01-11:
[andypugh]
> In kernel space there are kstrto* functions for integer and boolean
> types, but nothing for float.
> For the purposes of this component I think that a simple decimal
> number parser would probably work as we (you) have control of the data
> source too.

Could work, but it would be best to detect and error out if someone in
the future used it in an supported way, to avoid unexpected behaviour.

-- 
Happy hacking
Petter Reinholdtsen

andypugh, 2023-01-11:
I think the only difference between my simplistic version and the real one is in not handling numbers of the form 314.159E-2. (apparently e, E, d, D are all valid exponent indicators) 
That and that the real one actually uses the sign, and doesn't forget to subtract '0' from the decimal places. 

A likely reason for there being no  kstrtod or kstrtof
https://stackoverflow.com/questions/13886338/use-of-floating-point-in-the-linux-kernel

andypugh, 2023-10-04:
Fixed by https://github.com/LinuxCNC/linuxcnc/commit/70f3bc12d2daac3ed3c233d0138cd8577a896e9e and a few follow-up commits

closed 2023-10-04 by andypugh