LinuxCNC/linuxcnc #2230: hm2_eth

Andrew8487, 2022-12-29:
Good afternoon.
I'm studying Linux CNC source codes. Driver hm2_eth.c
Tell me please.
How the hm2_eth_read function is called.
The function is called via hal_export_funct. Or a task is created for it via rtapi_task_new.
Thank you.

andypugh, 2022-12-29:
The exported functions are called by being added to a real-time thread in the HAL file

Andrew8487, 2022-12-29:
I am developing an Internet driver for STM32.
4 - encoder
4-pwm
32 - inputs
16 - exits.
INI file SERVO_PERIOD = 700000
I implemented the connection on sockets. But when running in a servo stream in (HAL file), the time of the last function call is very large, the value is ~600000. Is this value normal or not?