LinuxCNC/linuxcnc #2247: linuxcnc 2.8.4 crashes wayland on Buster

sebastiankuzminsky, 2023-01-05:
I made a fresh Debian 10 Buster VM (amd64), upgraded all the packages from debian.org, and rebooted.

The little gear icon on the login screen that lets you select the session type was on the default, "GNOME":

[image]

When I logged in, `echo $XDG_SESSION_TYPE` showed that I was running Wayland:

[image]

I built & installed the tip of 2.8, and when I ran `linuxcnc` and picked a config, it killed the wayland session and put me back at the login screen (no screenshot of this happening).

This time I selected "GNOME on Xorg":

[image]

This gave me an X11 session:

[image]

And as expected, linuxcnc ran fine.

So something is horribly wrong running 2.8 on wayland on Buster.

Running 2.9 or master on wayland on Bullseye works fine for me.

rodw-au, 2023-01-14:
The linuxcnc  .ISO is built on XFCE
The easy workaround is to just only install the XFCE desktop when installing Buster from a Debian ISO to remain consistent with the original environment.
This is a manifestation of  #2264  Does the environment variable fix it?
When Bookworm is released in a few months, Buster looses its oldstable status and falls off the Debian radar so resources should be focussed on getting a Bookworm release.

SebKuzminsky, 2023-01-16:
> This is a manifestation of #2264 Does the environment variable fix it?

This is *not* a manifestation of #2264, it's some other problem.  Setting the `PYOPENGL_PLATFORM` environment variable does not fix this issue.

Buster runs LinuxCNC 2.8, Bullseye and newer run LinuxCNC 2.9 or master.  Those versions use OpenGL in totally different ways.