LinuxCNC/linuxcnc #2257: PNCconf - delete tool.tbl during modification  configuration

zz912, 2023-01-10:
I would like to request that the tool.tbl file generated by PNCconf behaves the same as the custom.hal file. I would need PNCconf to create a tool.tbl file during the first configuration, but not overwrite the tool.tbl file when modifying an already created configuration. It's annoying that I have to refresh the tool table after every machine configuration modification.

c-morley, 2023-01-11:
Ya that seems reasonable - I'll see what I can do.

c-morley, 2023-01-14:
ok this should be fixed now.

zz912, 2023-01-15:
I wanted test it, but PNCconf doesnt work now:
 
[image]
[image]
[image]
[image]
[image]
[image]

zz912, 2023-01-15:
I made new test with old Mesa card 7i96 not with 7i96s and I found that PNCconf works.

I tested overwriting tool.tbl and verified that it works.

Therefore, I would like to thank you for this modification and hereby close this Issue.

closed 2023-01-15 by zz912