LinuxCNC/linuxcnc #2273: Configuration warnings by sim.gmoccapy

smoe, 2023-01-15:
Hello, Just to make a good impression on new users (and convey trust for those transitioning to the new release) we should avoid warnings on stdout/stderr. I came across

```
**** GMOCCAPY INFO ****
**** Entering make macro button
found 15 Macros
Macro go_to_position x-pos y-pos z-pos has image link
Image = /home/moeller/linuxcnc/configs/sim.gmoccapy/macros/images/goto_x_y_z.png
get button with image
Macro i_am_lost has image link
Image = /home/moeller/linuxcnc/configs/sim.gmoccapy/macros/images/i_am_lost.png
get button with image
Macro macro_8 has image link
Image = /home/moeller/linuxcnc/configs/sim.gmoccapy/macros/images/macro_8.png
get button with image
**** GMOCCAPY INFO ****
```
Why is this shown? Maybe a debug level wrongly set? The files are existing:
```
-rw-r--r-- 1 moeller moeller  24520 26. Dez 18:38 /home/moeller/linuxcnc/configs/sim.gmoccapy/macros/images/goto_x_y_z.png
-rw-r--r-- 1 moeller moeller  14074 26. Dez 18:38 /home/moeller/linuxcnc/configs/sim.gmoccapy/macros/images/i_am_lost.png
-rw-r--r-- 1 moeller moeller 222151 26. Dez 18:38 /home/moeller/linuxcnc/configs/sim.gmoccapy/macros/images/macro_8.png
```
I could open these files in gimp (macro_8 is very cute). So, maybe just not print this?

```
**** GMOCCAPY INFO ****
**** Place in table
/usr/bin/gmoccapy:1565: DeprecationWarning: Gtk.Table.resize is deprecated
  self.widgets.tbl_DRO.resize(rows, cols)
**** GMOCCAPY INFO ****
```
https://docs.gtk.org/gtk3/method.Table.resize.html mentions that this is deprecated since 3.4 and GtkGrid would resize automatically. Such dynamics may be interesting - after all, reactive web pages are a thing, too. No idea how will this works for us, though.

```
**** GMOCCAPY INFO ****
**** Gremlin Einrichtung gestartet ****
Found an error!
The following information may be useful in troubleshooting:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gladevcp/hal_gremlin.py", line 164, in do_set_property
    if self.lathe_option:
       ^^^^^^^^^^^^^^^^^
AttributeError: 'HAL_Gremlin' object has no attribute 'lathe_option'
```
The default "sim.gmoccapy" is a 3-axis mill and that is why that lathe_option is not present?!?
Directly underneath it states the following, which may be related (even though it does not look like it):
```
not enough values to unpack (expected 2, got 1)
'btn_sel_next'
'btn_sel_prev'
'btn_select'
'btn_home'
'btn_dir_up'
'btn_jump_to'
not enough values to unpack (expected 2, got 1)
not enough values to unpack (expected 2, got 1)
'IconFileSelection' object has no attribute 'on_btn_jump_to'
not enough values to unpack (expected 2, got 1)
not enough values to unpack (expected 2, got 1)

(gmoccapy:101189): Gtk-CRITICAL **: 22:43:43.955: gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed

(gmoccapy:101189): Gtk-CRITICAL **: 22:43:44.047: gtk_entry_set_text: assertion 'text != NULL' failed
Attribute error <GParamString 'icon-theme-path'> and <class 'str'> sans 10
Attribute error <GParamString 'icon-theme-name'> and <class 'str'> sans 10
0  =  /usr/bin/gmoccapy
1  =  -ini
2  =  /home/moeller/linuxcnc/configs/sim.gmoccapy/gmoccapy.ini
Entry = trivkins
Entry = coordinates=xyz
found the following coordinates xyz
```
Harmless:
```
**** virtual keyboard program found : <onboard>
14:09:30.842 WARNING Config: mousetweaks GSettings schema not found, mousetweaks integration disabled.
/usr/bin/gmoccapy:3318: DeprecationWarning: Gtk.FontButton.get_font_name is deprecated
  self.notification.set_property('font', self.widgets.fontbutton_popup.get_font_name())
/usr/bin/gmoccapy:381: DeprecationWarning: Gtk.FontButton.set_font_name is deprecated
  self.widgets.fontbutton_popup.set_font_name(self.prefs.getpref("message_font", "sans 10", str))
/usr/bin/gmoccapy:476: DeprecationWarning: Gtk.ColorButton.set_color is deprecated
  self.widgets.abs_colorbutton.set_color(Gdk.color_parse(self.abs_color))
```
which has been deprecated with version 3.4 (according to https://athenajc.gitbooks.io/python-gtk-3-api/content/gtk-group/gtkcolorbutton.html) and we are requesting 3.11+ as a dependency.

Similarly:
```
**** GMOCCAPY INFO ****
**** Entering make touch button
/usr/bin/gmoccapy:972: DeprecationWarning: Gtk.Widget.override_background_color is deprecated
  btn.override_background_color(Gtk.StateFlags.ACTIVE, Gdk.RGBA(1.0, 1.0, 0.0, 1.0))
**** GMOCCAPY INFO ****
```
Maybe just give default settings?
```
**** GMOCCAPY GETINIINFO **** 
[KINS] KINESTYPE is trivkins
**** GMOCCAPY GETINIINFO **** 
No MIN_ANGULAR_VELOCITY entry found in [DISPLAY] of INI file
Using default value of 0.1 degree / min
**** GMOCCAPY GETINIINFO **** 
No DEFAULT_ANGULAR_VELOCITY entry found in [DISPLAY] of INI file
Using default value of 360 degree / min
**** GMOCCAPY GETINIINFO **** 
No MAX_RAPID_OVERRIDE entry found in [DISPLAY] of INI file 
 Default settings 100 % applied!
**** GMOCCAPY GETINIINFO **** 
```

hansu, 2023-01-15:
Introducing of debug levels in Gmoccapy is a well-known to-do.
Most of the Gtk deprecation warnings I was able to remove, but for some of them it's really hard to find the root.
But I am wondering more where this comes from
```
[QtDesigner][INFO]  Logging to: /home/cnc/qtdesigner.log (logger.py:107)
[QtDesigner.QTVCP.QT_ISTAT][DEBUG]  Machine is METRIC based. unit Conversion constant=0.03937007874015748 (qt_istat.py:150)
[QtDesigner.QTVCP.QT_ISTAT][DEBUG]  TRAJ COORDINATES: XYZ (qt_istat.py:161)
[QtDesigner.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No MIN_LINEAR_VELOCITY Entry in DISPLAY, Using: 1 (qt_istat.py:477)
[QtDesigner.QTVCP.QT_ISTAT][DEBUG]  DEFAULT_LINEAR_VELOCITY = 6000.0 (qt_istat.py:330)
[QtDesigner.QTVCP.QT_ISTAT][DEBUG]  MIN_LINEAR_VELOCITY = 60.0 (qt_istat.py:331)
[QtDesigner.QTVCP.QT_ISTAT][DEBUG]  MAX_LINEAR_VELOCITY = 9960.0 (qt_istat.py:332)
[QtDesigner.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No DEFAULT_SPINDLE_0_SPEED Entry in DISPLAY, Using: 200 (qt_istat.py:477)
[QtDesigner.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No MIN_SPINDLE_0_SPEED Entry in DISPLAY, Using: 100 (qt_istat.py:477)
[QtDesigner.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No MAX_SPINDLE_0_SPEED Entry in DISPLAY, Using: 2500 (qt_istat.py:477)
[QtDesigner.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No MAX_SPINDLE_0_OVERRIDE Entry in DISPLAY, Using: 1 (qt_istat.py:477)
[QtDesigner.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No MIN_SPINDLE_0_OVERRIDE Entry in DISPLAY, Using: 0.5 (qt_istat.py:477)
```

I have no idea how Qt-stuff is related to Gmoccapy.

smoe, 2023-01-15:
Can you reproduce this?

hansu, 2023-01-15:
> Can you reproduce this?

Just running the basic Gmoccapy sample config

c-morley, 2023-01-15:
Qt warnings will come from the hal_action library.
ACTION and INFO libraries were added
# 4772b687

hansu, 2023-01-15:
Ah ok, thanks.

c-morley, 2023-01-15:
GMoccapy could use qtvcp's logger - then you could change the prefix name of messages..
And you get nice messages.

hansu, 2023-01-19:
> GMoccapy could use qtvcp's logger - then you could change the prefix name of messages.. And you get nice messages.

Is it just adding this to Gmoccapy?

```
# Set up logging
from qtvcp import logger
LOG = logger.getLogger(__name__)

# Set the log level for this module
#LOG.setLevel(logger.INFO) # One of DEBUG, INFO, WARNING, ERROR, CRITICAL
```

c-morley, 2023-01-20:
Actually this to GMoccapy:
`from qtvcp import logger
LOG = logger.initBaseLogger('GMOCCAPY', log_file=None, log_level=logger.ERROR)
`
The example you posted is for the files you wish to have logging in.

hansu, 2023-01-21:
When adding this I still get a lot of QtVCP log messages with WARNING, DEBUG, INFO and ERROR

c-morley, 2023-01-21:
Did you put it before this line?
import gladevcp.makepins

hansu, 2023-01-21:
> Did you put it before this line? import gladevcp.makepins

No. Do I have to?

c-morley, 2023-01-21:
Yes you must set the log level before the file are imported.

hansu, 2023-01-21:
I set your two lines at the very beginning of the imports, but it didn't change anything.

c-morley, 2023-01-21:
Did you compile it? It definitely works here.

hansu, 2023-01-21:
Sorry my fault, I tried it on a configuration where I used cam_align.
For the default sim config it works :)

How can I suppress the warning when I run cam_align in an embedded tab in gmoccapy?
I guess the cam_align widget needs to be modified in this case?

c-morley, 2023-01-21:
What command are you using to load cam_align?

c-morley, 2023-01-21:
if the command is something like:
halcmd loadusr -Wn qtvcp_embed qtvcp  -d -c qtvcp_embed  -x {XID} -o camnumber=0 -o size=500,600 -o rotincr=.2 cam_align

remove the -d

You'll still get some warnings that can't easily be removed.

c-morley, 2023-01-21:
I just pushed a change: -q will quiet all but error or critical logging

hansu, 2023-01-21:
>if the command is something like:
halcmd loadusr -Wn qtvcp_embed qtvcp -d -c qtvcp_embed -x {XID} -o camnumber=0 -o size=500,600 -o rotincr=.2 cam_align

Wes :+1: 

Is now really quite with the -q option, Thanks!

I think it would be good if these options are also documented in the man page. I can do that...

hansu, 2023-01-22:
>I think it would be good if these options are also documented in the man page. I can do that...

Done in https://github.com/LinuxCNC/linuxcnc/pull/2295

smoe, 2023-02-11:
I just ran the latest upload to Debian and stumbled over those warnings. Should the "-q" not be the default? Or should that be another option to check at start?

c-morley, 2023-02-11:
I would suggest no, -q should not be default.
Don't run linuxcnc from the command line and you won't see them anymore :)
We could fix the INI to fix the warnings - ideally teach qtvcp and gmoccapy to use the same entries.

petterreinholdtsen, 2023-02-11:
[c-morley]
> I would suggest no, -q should not be default.
> Don't run linuxcnc from the command line and you won't see them
> anymore :)

Just remember that stdout and stderr from programs started from the QUI
end up in ~/.xsession-errors and can fill up the disk.  I have seen
programs spew out so much output during run that the user home directory
filled up and everything stopped working.

-- 
Happy hacking
Petter Reinholdtsen

c-morley, 2023-02-11:
Yes but aren't we talking of a dozen lines here?
Also we reason there are warnings is we hope the user will fix them.

hansu, 2023-02-11:
> I just ran the latest upload to Debian and stumbled over those warnings. Should the "-q" not be the default? Or should that be another option to check at start?

That's because the latest debian upload missed the log-related changes by half an hour :see_no_evil: 
With #2333 we get a nice clean output with the default settings:

```
1   LINUXCNC - 2.9.0~pre1
2   pickconfig: installed aux examples: /usr/share/linuxcnc/aux_examples/NativeCAM
3   Machine configuration directory is '/home/cnc/linuxcnc/linuxcnc-dev-master/configs/sim/gmoccapy'
4   Machine configuration file is 'gmoccapy.ini'
5   Starting LinuxCNC...
6   linuxcnc TPMOD=tpmod HOMEMOD=homemod EMCMOT=motmod
7   Note: Using POSIX realtime
8   Found file(REL): ./core_sim.hal
9   Unexpected realtime delay on task 0 with period 100000
10  This Message will only display once per session.
11  Run the Latency Test and resolve before continuing.
12  Found file(REL): ./spindle_sim.hal
13  Found file(REL): ./simulated_home.hal
14  [Gmoccapy.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No MIN_LINEAR_VELOCITY Entry in DISPLAY, Using: 0 (qt_istat.py:500)
15  [Gmoccapy.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No DEFAULT_SPINDLE_0_SPEED Entry in DISPLAY, Using: 200 (qt_istat.py:500)
16  [Gmoccapy.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No MIN_SPINDLE_0_SPEED Entry in DISPLAY, Using: 100 (qt_istat.py:500)
17  [Gmoccapy.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No MAX_SPINDLE_0_SPEED Entry in DISPLAY, Using: 2500 (qt_istat.py:500)
18  [Gmoccapy.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No MAX_SPINDLE_0_OVERRIDE Entry in DISPLAY, Using: 1 (qt_istat.py:500)
19  [Gmoccapy.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No MIN_SPINDLE_0_OVERRIDE Entry in DISPLAY, Using: 0.5 (qt_istat.py:500)
20  
21  (gmoccapy:19596): Gtk-CRITICAL **: 08:30:57.074: gtk_entry_set_text: assertion 'text != NULL' failed
22  Attribute error <GParamString 'icon-theme-path'> and <class 'str'> sans 10
23  Attribute error <GParamString 'icon-theme-name'> and <class 'str'> sans 10
24  [Gmoccapy.GMOCCAPY.GETINIINFO][WARNING]  No MIN_ANGULAR_VELOCITY entry found in [DISPLAY] of INI file. Using default value of 0.1 degree / min. (getiniinfo.py:274)
25  [Gmoccapy.GMOCCAPY.GETINIINFO][WARNING]  No DEFAULT_ANGULAR_VELOCITY entry found in [DISPLAY] of INI file. Using default value of 360 degree / min. (getiniinfo.py:258)
26  [Gmoccapy.GMOCCAPY.GETINIINFO][WARNING]  No MAX_RAPID_OVERRIDE entry found in [DISPLAY] of INI file. Default settings 100 % applied! (getiniinfo.py:311)
27  /home/cnc/linuxcnc/linuxcnc-dev-master/bin/gmoccapy:992: DeprecationWarning: Gtk.Widget.override_background_color is deprecated
28    btn.override_background_color(Gtk.StateFlags.ACTIVE, Gdk.RGBA(1.0, 1.0, 0.0, 1.0))
29  /home/cnc/linuxcnc/linuxcnc-dev-master/bin/gmoccapy:1587: DeprecationWarning: Gtk.Table.resize is deprecated
30    self.widgets.tbl_DRO.resize(rows, cols)
31  task: main loop took 0.101714 seconds
32  [Gmoccapy][WARNING]  No audio available! PYGST library not installed? Is python-gstX.XX installed? (gmoccapy:2051)
33  [Gmoccapy][WARNING]  No virtual keyboard installed, we checked for <onboard>. Try 'sudo apt-get install onboard'. (gmoccapy:2125)
```
Those two on line 27 & 29 can't be avoided without converting some parts of Gmoccapy.
For 
`21  (gmoccapy:19596): Gtk-CRITICAL **: 08:30:57.074: gtk_entry_set_text: assertion 'text != NULL' failed
`
I have no idea where it is coming from.

For those missing INI value warnings - there I have to figure out the suitable values and add them to the sample configs.

c-morley, 2023-02-11:
I bet Gmoccapy doesn't use:

16  [Gmoccapy.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No MIN_SPINDLE_0_SPEED Entry in DISPLAY, Using: 100 (qt_istat.py:500)
17  [Gmoccapy.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No MAX_SPINDLE_0_SPEED Entry in DISPLAY, Using: 2500 (qt_istat.py:500)
18  [Gmoccapy.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No MAX_SPINDLE_0_OVERRIDE Entry in DISPLAY, Using: 1 (qt_istat.py:500)
19  [Gmoccapy.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No MIN_SPINDLE_0_OVERRIDE Entry in DISPLAY, Using: 0.5 (qt_istat.py:500)

But maybe we could change that?

hansu, 2023-02-11:
Yes apparently not.

>But maybe we could change that?

The `MIN_SPINDLE_0_SPEED` and `MAX_SPINDLE_0_SPEED` are set in the preferences:
[image]
Possible to move that to the INI file, yes.

`MIN_SPINDLE_0_OVERRIDE` and `MAX_SPINDLE_0_OVERRIDE` only  have another name:
```
MAX_SPINDLE_OVERRIDE =  1.2
MIN_SPINDLE_OVERRIDE = 0.5
```
To change that I think we have to increase the INI file version to upgrade the names automatically.

I think both is more something for master than 2.9.

smoe, 2023-02-12:
This is kind of the wrong thread for a warning with the imperial axis.sim and the 9-axis .sim, while axis_mm is fine:

```
LINUXCNC - 2.9.0~pre1+git20230208.f1270d6ed7
Machine configuration directory is '/home/moeller/linuxcnc/configs/sim.axis'
Machine configuration file is 'axis_9axis.ini'
Starting LinuxCNC...
linuxcnc TPMOD=tpmod HOMEMOD=homemod EMCMOT=motmod unlock_joints_mask=0x10
Note: Using POSIX non-realtime
Found file(LIB): /usr/share/linuxcnc/hallib/basic_sim.tcl
setup_kins: cmd=loadrt trivkins
motion.c: Creating unlock hal pins for joint 4
Found file(lib): /usr/share/linuxcnc/hallib/locking_indexer.hal
link (updating variable file): Datei oder Verzeichnis nicht gefunden
note: MAXV     max: 1.200 units/sec 72.000 units/min
note: LJOG     max: 1.200 units/sec 72.000 units/min
note: LJOG default: 1.000 units/sec 60.000 units/min
note: AJOG     max: 90.000 units/sec 5400.000 units/min
note: AJOG default: 45.000 units/sec 2700.000 units/min
```
This "Dir or File not found" error on the "link(..)" line is my concern.  Any idea anyone? And could the error message be a bit more informative? A quick grep pointed me to 
```
src/emc/rs274ngc/rs274ngc_pre.cc:    perror("link (updating variable file)");
```
Thanks!

hansu, 2023-02-12:
>This is kind of the wrong thread for a warning with the imperial axis.sim and the 9-axis .sim, while axis_mm is fine:

Indeed. At least I have no idea.

hansu, 2023-02-20:
> I bet Gmoccapy doesn't use:
>
> 16 [Gmoccapy.QTVCP.QT_ISTAT][WARNING] INI Parsing Error, No MIN_SPINDLE_0_SPEED Entry in DISPLAY, Using: 100 (qt_istat.py:500) 17 [Gmoccapy.QTVCP.QT_ISTAT][WARNING] INI [...]

@c-morley I would suggest removing those warnings. Because the docs say frst it's only for Qt based GUIs and second they say it's deprecated:
```
DEFAULT_SPINDLE_0_SPEED = 100 - The default spindle RPM when the spindle is started in manual mode. On multi spindle machine there will be entries for each spindle number. Only used by the QtVCP-based user interfaces.

    deprecated - use the [SPINDLE_n] section instead.
```

c-morley, 2023-02-20:
The docs are wrong. Someone misunderstood the entries.
The SPINDLE_N section is for HAL file substitution only. It's a convention started by pncconf.

[DISPLAY] is setting that relate to the GUI. 
MAX_SPINDLE_0_SPEED is a GUI setting not a setting for anything else.

The setting in DISPLAY are not depreciated not sure who or why they said that.
AXIS and Gmoccapy do not use [DISPLAY_N]

MAX_SPINDLE_0_SPEED (etc)  are qtvcp only because qtvcp is the only program so far that considers that there might be more then one spindle.

c-morley, 2023-02-20:
So I have to correct ,myself. Looking in the source Andy has added some setting that are used by motion? - still looking into it.
It's certainly annoying when something major is changed and not reported for all to adjust to.

At best that means some settings are in DISPLAY some in SPINDLE_N

c-morley, 2023-02-20:
I guess we need to decide if having different (manual) upper and lower spindle speeds then the hardware is capable is something we wish to keep.
We do this for jogging for instance.

hansu, 2023-02-21:
Yes maybe. Gmoccapy uses spindle speed limits for scaling of the sindle speed bar display. But it doesn't take the values from the INI file

hansu, 2023-03-14:
Resolved by https://github.com/LinuxCNC/linuxcnc/pull/2383

PR #2323 (2023-02-04): gmoccapy -add log levels to gmoccapy and getiniinfo
This will allow the user to control the many debug messages sent to the terminal. 
It uses qtvcp's logging library so no reinventing the wheel. 
Other files should probably be converted to use logging too. 

Resolves #2273
PR #2330 (2023-02-09): GMOCCAPY 3.4.2
Swapped position of settings button with user tabs button to a more logical order.
Log levels can now be specified in the INI file (#2323).

Closes #2273
closed 2023-03-14 by hansu