LinuxCNC/linuxcnc #2276: Problem with tkinter on updated Bookworm

hansu, 2023-01-15:
I did a fresh Bullseye install and built LinuxCNC from source and got a working RIP environment.
Then I updated to Bookworm (and installed python3.11-dev like mentioned in https://github.com/LinuxCNC/linuxcnc/issues/2263).
When starting LinuxCNC I get this weird error:

```
cnc@bullseye:~/linuxcnc/src$ linuxcnc
LINUXCNC - 2.9.0~pre1
Machine configuration directory is '/home/cnc/linuxcnc/configs/sim/axis'
Machine configuration file is 'axis_mm.ini'
Starting LinuxCNC...
linuxcnc TPMOD=tpmod HOMEMOD=homemod EMCMOT=motmod
Note: Using POSIX non-realtime
Found file(lib): /home/cnc/linuxcnc/lib/hallib/core_sim.hal
Found file(lib): /home/cnc/linuxcnc/lib/hallib/sim_spindle_encoder.hal
Found file(lib): /home/cnc/linuxcnc/lib/hallib/axis_manualtoolchange.hal
Traceback (most recent call last):
  File "/home/cnc/linuxcnc/bin/hal_manualtoolchange", line 61, in <module>
    import nf, rs274.options
  File "/home/cnc/linuxcnc/lib/python/nf.py", line 19, in <module>
    import tkinter
ModuleNotFoundError: No module named 'tkinter'
/home/cnc/linuxcnc/lib/hallib/axis_manualtoolchange.hal:1: waitpid failed hal_manualtoolchange hal_manualtoolchange
/home/cnc/linuxcnc/lib/hallib/axis_manualtoolchange.hal:1: hal_manualtoolchange exited without becoming ready
Shutting down and cleaning up LinuxCNC...
Traceback (most recent call last):
  File "/home/cnc/linuxcnc/bin/axis-remote", line 28, in <module>
    import tkinter
ModuleNotFoundError: No module named 'tkinter'
Note: Using POSIX non-realtime
LinuxCNC terminated with an error.  You can find more information in the log:
    /home/cnc/linuxcnc_debug.txt
and
    /home/cnc/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal
```

But the tkinter package seems to be available for both python versions:
```
cnc@bullseye:~/linuxcnc/src$ python3 --version
Python 3.10.9
cnc@bullseye:~/linuxcnc/src$ python3.11 --version
Python 3.11.1
cnc@bullseye:~/linuxcnc/src$ python3 -c "import tkinter"
cnc@bullseye:~/linuxcnc/src$ python3.11 -c "import tkinter"
```

Not sure if this is relevant.

rodw-au, 2023-01-17:
What happens if you install Bookworm non-free iso-cd from the daily builds?
https://cdimage.debian.org/cdimage/unofficial/non-free/cd-including-firmware/daily-builds/sid_d-i/current/amd64/
In the absence of official debs for Bullseye, your installation path is unusual.
A more normal path would be to install Bookworm and use the included linuxcnc-uspace package.
That would ensure only one python version exists.

I got as far as pulling down a current image today but got interrupted before i got it installed on a VM.

hansu, 2023-01-17:
If I follow a more straight forward path like do a fresh Bookworm install, I don't get this problem. 
I only consider the case that some user is updating from Bullseye.

rodw-au, 2023-01-17:
That is my point. I started to use Bullseye before it was released because my long standing installation  on Mint was broken by the multiple python versions and Mint not keeping pace with Python versions required by Linuxcnc so I exclusively use Debian now. More recently because linuxcnc is included, I try and use Bookworm.

Lets say your hardware is not supported on Buster or you need a feature that is not available in V2.8. eg a Plasma table or Custom homing.

Your choices are:
1.  Install on Bookworm which is still in testing status. Its hard to find a good image to work from. But there is a packaged distribution of Linuxcnc available once you get past that.
2. Install Bullseye and accept that the only way you can run linuxcnc on Bullseye is to install from source. 
You might follow a recipe like this: https://www.qtpyvcp.com/install/bullseye.html#
Or a script like this: https://github.com/joco-nz/lcnc-bullseye-installer
Or do it the traditional way like this: https://youtu.be/Oe4Hfq3G2_A

Once you've made your decision, you choose Bullseye and you get Linuxcnc installed and operational, there is no compelling reason to upgrade to Bookworm.  If you have to adopt Bookworm to support your hardware (unlikely in my experience), it won't matter if you have to start from scratch.

I personally think where the project is right now, we should not be wasting resources on Bullseye. I'd much rather see that effort expended on solving network latency issues which are far more likely to be a support issue than Bullseye support. (It spans Bullseye and Bookworm - kernels 5.10 to 6.10)

hansu, 2023-01-17:
>Once you've made your decision, you choose Bullseye and you get Linuxcnc installed and operational, there is no compelling reason to upgrade to Bookworm.

Yes not in the next time, but maybe at some point when the supports for Bullseye ends.

I agree that the relevance is not very high, but I wanted to mention it.
So I close this issue?

rodw-au, 2023-01-17:
Well right now Bullseye is not a supported platform for Linuxcnc so support for Bullseye has never started.
Closing is not my call

hansu, 2023-01-17:
But this is not Bullseye, only a Bookworm derived from Bullseye ;-)

petterreinholdtsen, 2023-01-17:
[Hans Unzner]
> But this is not Bullseye, only a Bookworm derived from Bullseye ;-)

Most Debian bookworm installations will be derived from bullseye.  Do we
need to adjust some package relations to make sure upgrades work
flawlessly?

-- 
Happy hacking
Petter Reinholdtsen

hansu, 2023-01-18:
Maybe this happens only when using a RIP. I will also test install it from a deb package.

hansu, 2023-01-18:
>Install on Bookworm which is still in testing status. Its hard to find a good image to work from. But there is a packaged distribution of Linuxcnc available once you get past that.

Where can I find that?

hansu, 2023-01-18:
Okay this happens only in a RIP environment. When installing from the package via apt, all works fine.

closed 2023-01-18 by hansu