LinuxCNC/linuxcnc #2277: gladevcp XML/PNG files are not in debs

c-morley, 2023-01-16:
User on the forum is having problems getting glade to work on an instaled version.
Seems the requires hal_python.xml  plus all the widget image files are not installed on the system.

/usr/lib/python3/dist-packages/gladevcp  is missing the .xml and .png files.

These are found in the source at:
lib/python/gladevcp

affects 2.9 and master for sure. Probably older too.

c-morley, 2023-01-16:
I should say that installing the XML file and setting the envromental variable allowed me to load widgets.
It's been awhile but maybe the XML can go somewhere else - rings a bell....

c-morley, 2023-01-16:
ok here we go.
the XML and images are in:
lib/share/glade3
and it seems they now need to be in
 usr/lib/glade

Haven't got the images to work yet...

c-morley, 2023-01-16:
You can also set the catalog from the Preference dialog - should add that to docs.

rodw-au, 2023-01-17:
What version of Debian?
What version of Linuxcnc?
URL to forum post?

It got harder now with 3 versions of linuxcnc and 3 versions of Debian in play.

c-morley, 2023-01-17:
1) no idea - I was using Mint-My guess is the source is wrong.
2) 2.9 and master (I was using 2.9)
3) The forum page led me to the issue - the fellow in the end was using RIP and forgot to run the init script.

blazini36, 2023-01-31:
> ok here we go. the XML and images are in: lib/share/glade3 and it seems they now need to be in usr/lib/glade
> 
> Haven't got the images to work yet...

Images are an issue here too. I've gotten gladeVCP to work in a couple of different setups and both spit out this
`HdyCarousel	needs an icon named 'widget-libhandy-carousel'`
for every widget

smoe, 2023-02-10:
I just started gmoccpy on Debian arm64 (February 8, 2003, official package) and run in that problem, too:
./lib/python/gladevcp/pi.png
is not installed, expected as
/usr/lib/python3/dist-packages/gladevcp/pi.png

The linuxcnc.install.in instructions seem fine (they just install everything in usr/lib/python3), have not tried it, but
```
--- a/src/Makefile
+++ b/src/Makefile
@@ -800,6 +800,7 @@ install-python: install-dirs
        $(FILE) ../lib/python/qtvcp/plugins/*.py $(DESTDIR)$(SITEPY)/qtvcp/plugins
        $(TREE) ../lib/python/qtvcp/lib/* $(DESTDIR)$(SITEPY)/qtvcp/lib
        $(FILE) ../lib/python/gmoccapy/*.py $(DESTDIR)$(SITEPY)/gmoccapy
+       $(FILE) ../lib/python/gladevcp/*.png $(DESTDIR)$(SITEPY)/gladevcp
        $(FILE) ../lib/python/gladevcp/*.py $(DESTDIR)$(SITEPY)/gladevcp
        $(FILE) ../lib/python/gladevcp/*.glade $(DESTDIR)$(SITEPY)/gladevcp
        $(FILE) ../lib/python/stepconf/*.py $(DESTDIR)$(SITEPY)/stepconf
```
would be something I feel like trying. @hansu ?

Did not encounter anything else that would be missing.

hansu, 2023-02-10:
@smoe Yeah, go for that!

hansu, 2023-04-21:
Fixed in https://github.com/LinuxCNC/linuxcnc/pull/2334

closed 2023-04-21 by hansu