LinuxCNC/linuxcnc #2281: Network latency breaks Mesa ethernet connection

rodw-au, 2023-01-17:


petterreinholdtsen, 2023-01-17:
Sound correct.  Make sure you have a low latency network and network
card (preferable a dedicated one), to avoid unpredictable latency.

-- 
Happy hacking
Petter Reinholdtsen
***@***.***

rodw-au, 2023-01-18:
> Sound correct. Make sure you have a low latency network and network card (preferable a dedicated one), to avoid unpredictable latency.
> […](#)
> -- Happy hacking Petter Reinholdtsen ***@***.***

Its not as simple as that. The hardware is low latency and works on kernels up to 4.9. The network with mesa is point to point on dedicated network segment so is low latency by definition. It fails from bullseye on with any Debian provided kernel.

petterreinholdtsen, 2023-01-18:
[Rod Webster]
> Its not as simple as that. The hardware is low latency and works on
> kernels up to 4.9. The network with mesa is point to point on
> dedicated network segment so is low latency by definition. It fails
> from bullseye on with any Debian provided kernel.

OK.  The original message was scarse on information, and I guess we
still need more information to understand how this is an issue with
linuxcnc.  It seem like you are saying that Linux kernels above 4.9
introduce unwanted latency.  If so, I would guess that have to be
address in the Linux kernel, not LinuxCNC.  Can you tell why you believe
this is an issue with LinuxCNC?

-- 
Happy hacking
Petter Reinholdtsen

rodw-au, 2023-01-18:
> It seem like you are saying that Linux kernels above 4.9 introduce unwanted latency. If so, I would guess that have to be address in the Linux kernel, not LinuxCNC. Can you tell why you believe this is an issue with LinuxCNC?
> […](#)
> -- Happy hacking Petter Reinholdtsen

Yes, The Debian RT kernel patches commencing with the RT 5.10 kernel introduce network latency that is not present in the upstream kernel.org sources. Where Linuxcnc is communicating in realtime to devices on an ethernet connection and latency exceeds acceptible limits, an "error finishing read" is generated and the device is disabled. This "bricks" Linuxcnc rendering it useless. A program restart is required to clear the error but this does not solve the problem and it will reocurr. I have experienced this on 4-5 PC's personally and many other experienced early adopters have been affected by this issue. It has held me up on one project for many months while seeking a resolution. In many instances later kernels are required to provide driver support to the affected PC's so staying on 4.x kernels is not a viable option.

Once Linuxcnc rolls out Version 2.9, it will be on the Bookworm or Bullseye platform. A large number of users will be impacted as existing working PC hardware will fail on the later Debian kernels. This will reflect poorly on this project. 

Clearly, the Linuxcnc project will be significantly impacted if no action is taken when Bookworm is released.

It would be prudent for the Linuxcnc project to take up this upstream issue with the Debian linux-image-rt developers in an attempt to have it resolved before the impending hard freeze.

I don't have the skills to review the kernel patches. If I was to hazzard a guess at the problem, I would investigate the "lazy preemption" patches which I think were introduced in RT kernel 5.10.

rodw-au, 2025-12-09:
Ooops, I have no Idea how this was closed. That was not my  intention. Perhaps it can be reopened?