LinuxCNC/linuxcnc #2286: Qtdragon NGCGui not rendering UI

ihabmmali, 2023-01-18:
## Here are the steps I follow to reproduce the issue:

 1. Compile latest version of linuxcnc 2.1 from git source
 2. Launch the qtdragon_hd sim configuration
 3. Navigate to Utils->NGCGUI Tab

## This is what I expected to happen:

See the usual NGCGui dialogue for creating a features

## This is what happened instead:

We get an empty table with a single line text input box that seems to be disabled:

[image]

## It worked properly before this:
Pre 2.9->2.1 merge

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
 Distributor ID:	Debian
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye

 * I am using this kernel version (shown by `uname -a`):Linux linuxcnctest 5.18.0-2-rt-amd64 #1 SMP PREEMPT_RT Debian 5.18.5-1 (2022-06-16) x86_64 GNU/Linux

 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [X ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):v2.10.0-pre0-566-gaa545be959

 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): qtdragon_hd
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): NA, testing UI in sim mode

c-morley, 2023-01-19:
Can you run in a terminal and see if there are relative debug information?

ihabmmali, 2023-01-19:
I have been running from command line and I cant see any unusual debug level errors:
[image]

ihabmmali, 2023-01-19:
I played around with the ini and chanced the NGCGUI_SUBFILE_PATH from the default of the sim file to the fully qualified path(so  ~/linuxcnc/nc_files/examples/ngcgui_lib instead of the relative file path) and the ngcgui tabs load properly. I think it wasn't actually finding the sub files for some reason when using the relative path.

c-morley, 2023-01-20:
Ok tha's for that. I think it would be better if it showed a message on the tab that the path couldn't be found.

c-morley, 2023-01-21:
My guess here is that as installed files get removed around, the relative path doesn't point in the right place anymore - it works fine in RIP.
I'll try to get back to this.

rmu75, 2025-12-18:
Is this still relevant? To me it's unclear if there has been a real problem in the first place.