LinuxCNC/linuxcnc #2287: qtdragon/hd: startup message about missing /media/xxx dir

sensille, 2023-01-18:
On the master branch (commit aa545be95967) qtdragon/hd complains on each startup about a missing /media/xxx dir.

`File Manager error - No such path: /media/arne`

I found a commit that seems to fix the issue for qtdragon:

https://github.com/LinuxCNC/linuxcnc/commit/e93baec245ebe9ee19859ac0e44e33a286156b05

Can this patch also be applied to qtdragon/hd? The code looks just like the previous version of the non-hd dragon.

Or is there a another way to get rid of that message? 

I cannot say if this is a new behavior, as I haven't used qtdragon on older versions of LinuxCNC

ihabmmali, 2023-01-19:
This has been the behavior since I started using qtdragon/qtddragon_hd since earlier versions of 2.9.
 I never have the /media/USERNAME folder created when installing Debian from source(at least Debian 11) so my solution was to edit line 49 of  /usr/lib/python3/dist-packages/qtvcp/widgets/file_manager.py after I install a new release of linuxcnc

from   
self.media_path = (os.path.join('/media', user))
to
 self.media_path = (os.path.join('/home', user))

c-morley, 2023-01-21:
What is the path to media on Debian 11?

phillc54, 2023-01-21:
/media

c-morley, 2023-01-21:
Excellent, I can fix that too.

c-morley, 2023-01-27:
I'm closing this. Please reopen if there is still a problem.

closed 2023-01-27 by c-morley