LinuxCNC/linuxcnc #2292: Create section in documentation for supported hardware

hansu, 2023-01-21:
We don't really have an overview of the hardware LinuxCNC can be used with in the docs.

So this is about creating a page in the docs based on this wiki page: http://wiki.linuxcnc.org/cgi-bin/wiki.pl?LinuxCNC_Supported_Hardware and extend/update it.

Maybe a separate page describing the Mesa hardware would be useful.

andypugh, 2023-01-23:
There is http://linuxcnc.org/docs/stable/html/drivers/hostmot2.html though it's not great.

hansu, 2023-01-23:
> There is http://linuxcnc.org/docs/stable/html/drivers/hostmot2.html though it's not great.

Yeah and the most popular cards are not mentioned :smile: .
Here is also a bit about the Smart Serial devices: https://linuxcnc.org/docs/stable/html/man/man9/sserial.9.html

petterreinholdtsen, 2023-01-23:
[Hans Unzner]
> So this is about creating a page in the docs based on this wiki page:
> http://wiki.linuxcnc.org/cgi-bin/wiki.pl?LinuxCNC_Supported_Hardware
> and extend/update it.
>
> Maybe a separate page describing the Mesa hardware would be useful.

Instead of supported hardware, what about a list if of various hardware
and their support status and driver references?  This way one could also
find non-supported hardware and hardware with out-of-tree drivers?  We
would have one place to point people to when they wonder if their
hardware work. :)

-- 
Happy hacking
Petter Reinholdtsen

hansu, 2023-01-23:
Hardware that has been supported or is one the way being supported could be added, yes.
And hardware with external drivers as well.