LinuxCNC/linuxcnc #2301: [Enhancement request] jog-speed python interface

spumco1234, 2023-01-26:
[forum discussion: [https://forum.linuxcnc.org/qtpyvcp/48020-probe-basic-jog-slider-status?start=0](url)

## Here are the steps I follow to reproduce the issue:

- halui.axis.jog-speed and halui.joint.jog-speed are connected to rotary encoder (MPG) and selector switch.
- Attempted to connect both pins to GUI hal slider, but unlike SRO/FRO/RRO this does not work.
- Jog-speed adjustments via encoder are accurate in LCNC, but the GUI has no way to display the adjustment/value made in halui

After some discussion with the QtPyVCP devs, they discovered that there is no python interface for jog-speed.  There is a python interface for spindle, feed, and rapid overrides, but jog-speed (and maybe max-vel?) appears to have been overlooked.

I think it'd be helpful to have a way to connect physical controls to the GUI for jog-speed.  I believe this is not limited to QtPyVCP, but affects all GUIs.

I don't know anyting about python, but I'm happy to do any testing if someone has the time to work on this feature request.

c-morley, 2023-01-27:
This is because (aside from MPG) linuxcnc does not have an internal concept of 'current jog speed' , 'current jog increment'  or 'selected axis' - each ui keeps track of it's own.
So it's not missing from the python module - it's missing from linuxcnc.

It's not immediately clear to me why this was decided - possibly for best flexibility or for least worst problems.
It's tough to get it right for all control paradigms ie using a physical selector switche on one ui makes things unclear in any other.
In qtvcp and gladevcp we extend the python interface to include a 'voluntary' jog rate/increment.
unfortunately because of limitations of python, it can only be easily accessed by programs in the same python context. i have worked around that with ZMQ messages or using HAL pins...

spumco1234, 2023-01-27:
Thank you.  I dont know the 'backstage' stuff, but here's what's rumbling around in my head based on your explanation:

LCNC holds values for feed/speed/rapid overrides, and these values can be retrieved by the GUI via python.  I believe these are attributes available through linuxcnc.stat(?) using the python interface.

In these cases, external controls connected to halui adjust the LCNC values, and GUI (sliders, status displays) can both follow _and_ adjust the LCNC values through the python interface.  Two way communication.

In the case of jog-speed, there is no internal value held by LCNC, so there is no corresponding linuxcnc.stat attribute.  Jog-speed is always set by the GUI when a jog command is sent from GUI to LCNC.  One-way communication.

And what you mentioned above - HAL pins - is what you did in this commit:

[https://github.com/LinuxCNC/linuxcnc/commit/da89022b89fe1d493f7a38fde4003786d6e475bd](url)

It appears that you created a GUI HAL pin which accepts a value from external sources - i.e. an MPG or whatever.

So, if I'm understanding the above (big IF)

If halui can adjust the LCNC jog-speed value even though the GUI doesn't 'see' it, doesn't that imply jog-speed is stored _somewhere_ in LCNC?  My MPG/encoder -> halui is certainly adjusting the jog-speed value even if the GUI has no idea.  Something is receiving that value and storing it for the next jog.

If not... would what I'm requesting be better directed at the GUI dev(s) rather than LCNC devs? Perhaps they could duplicate your commit in one of the other GUI frameworks.

c-morley, 2023-01-27:
`If halui can adjust the LCNC jog-speed value even though the GUI doesn't 'see' it, doesn't that imply jog-speed is stored somewhere in LCNC? My MPG/encoder -> halui is certainly adjusting the jog-speed value even if the GUI has no idea. Something is receiving that value and storing it for the next jog.
`
Yes - in halui. halui is just another GUI without the G.
Anything halui does, could be done in a GUI.

Typically yes we do it in each GUI and deal with the problems created by any other ui used at the same time.
For instance I could duplicate all of halui in qtvcp and then 'qtui' could directly talk to qtvcp and have access to the current jogspeed. Then I would ask people not to use halui. 

Not optimal for sure, but easiest, as penetrating linuxcnc's core code is painful

Your reference to a commit didn't show anything so I can't comment on it directly.

spumco1234, 2023-01-28:
Sorry about the bad link, I'm new at the github thing.

Commit was by you, August 20, 2021.  number da89022b89fe1d493f7a38fde4003786d6e475bd

I just searched the LCNC repo for "JOG_RATE" and found it on the first page of commits.

spumco1234, 2023-01-28:
halui being a 'UI' makes more sense now, and explains why/where the jog-speed value is stored.

spumco1234, 2023-01-31:
I think this can be closed as '_more-or-less-impossible_', and I'll look for other avenues to get an on-screen display of the external-input jog-speed value.

closed 2023-01-31 by spumco1234