LinuxCNC/linuxcnc #2312: GladeVCP/qtVCP should be loadable outside of LinuxCNC environment

blazini36, 2023-01-29:
Manjaro, and I'd assume any Arch based distro can no longer run LinuxCNC, some recent package completely broke it. Though it's not the "supported" distro, I've run LinuxCNC on Manjaro for years with no issue. Because Arch can no longer actually run LinuxCNC GladeVCP and QtVCP can't properly run the RIP environment script to enable Glade and Qt Designer to load the widgets.

LinuxCNC has alot of moving parts so I don't expect anyone to be worried about the fact that it can't be run on Arch. The UI development tools though should be a bit more flexible. A System capable of running designer and Glade shouldn't really have to be able to run LinuxCNC just to load some widgets. Having to have some instance of LinuxCNC running to do UI work has always been one of the more annoying aspects of trying to do UI work. Is there no way to get these assets loadable without the environment?

ve7it, 2023-01-29:
Hello,
I am currently running linuxcnc-sim on arch.

you can see the history of the issue on the arch package for linuxcnc-sim

https://aur.archlinux.org/packages/linuxcnc-sim

a quick summary:
- minigl.error: (1285, 'out of memory')  reported after updates in Dec 
  2022
- this is due to issues with built-in Intel video card (I use an i7 4770 
cpu)

- to get linuxcnc-sim running again on arch, I had to install mesa-amber 
and lib32-mesa-amber packages

- command line fix... pacman -S mesa-amber lib32-mesa-amber

I hope this helps get you back online with Manjaro.

cheers
Lawrence VE7IT

c-morley, 2023-01-29:
gladevcp and qtvcp directly interact with linuxcnc, so it makes little sense to me to put a lot of effort into making them work without linuxcnc's environment. I don't understand why you would want to do that, but I'm sure you have your reasons.
Qtvcp doesn't need linuxcnc running to edit the ui code - I'm pretty sure that Gladevcp is the same.

GLADE it's self is not getting much love these days and the python2/3 changes has lead to some unusual stability problems (well in glades case it's always been cranky)

Could you edit the environmental script to just set up enough for the widget editor?

blazini36, 2023-01-29:
> Hello, I am currently running linuxcnc-sim on arch. you can see the history of the issue on the arch package for linuxcnc-sim https://aur.archlinux.org/packages/linuxcnc-sim a quick summary: - [...]

Thanks for the info, I did not think to check the linuxcnc-sim package comments. The systems I run actual LinuxCNC on are Intels with built in graphics but I also get these issues with all AMD systems that I'm just doing GUI work on. I tried the amber packages on my desktop which is a 3900x and 6700xt and they result in not being able to load the desktop. I had to remove the amber packages and go back to the normal Mesa package, but I will try this on my Intel systems

This is the error I'm getting on the desktop and my laptop (AMD 4500u)
`python3.10: ../mesa-22.3.3/src/mesa/state_tracker/st_texture.c:72: st_texture_create: Assertion `width0 > 0' failed.
`