LinuxCNC/linuxcnc #2327: Gmoccapy crashes on invalid EMBED_TAB configuration (qt_istat.py)

hansu, 2023-02-07:
Taking a Gmoccapy sample configuration and add in [DISPLAY]:
`EMBED_TAB_NAME = test`
without providing the corresponding TAB_COMMAND or TAB_LOCATION.
### Result:
```
[Gmoccapy][ERROR]  Found an error!
The following information may be useful in troubleshooting:

Traceback (most recent call last):
  File "/home/cnc/linuxcnc/linuxcnc-dev-master/bin/gmoccapy", line 5766, in <module>
    import gladevcp.makepins
  File "/home/cnc/linuxcnc/linuxcnc-dev-master/lib/python/gladevcp/__init__.py", line 1, in <module>
    from .hal_pythonplugin import *
  File "/home/cnc/linuxcnc/linuxcnc-dev-master/lib/python/gladevcp/hal_pythonplugin.py", line 30, in <module>
    from .hal_gremlin import HAL_Gremlin
  File "/home/cnc/linuxcnc/linuxcnc-dev-master/lib/python/gladevcp/hal_gremlin.py", line 43, in <module>
    from .hal_actions import _EMC_ActionBase
  File "/home/cnc/linuxcnc/linuxcnc-dev-master/lib/python/gladevcp/hal_actions.py", line 31, in <module>
    from gladevcp.core import Info, Action
  File "/home/cnc/linuxcnc/linuxcnc-dev-master/lib/python/gladevcp/core.py", line 34, in <module>
    INI = Info()
  File "/home/cnc/linuxcnc/linuxcnc-dev-master/lib/python/qtvcp/qt_istat.py", line 75, in __init__
    self.update()
  File "/home/cnc/linuxcnc/linuxcnc-dev-master/lib/python/qtvcp/qt_istat.py", line 427, in update
    if self.TAB_NAMES is not None and len(self.TAB_NAMES) != len(self.TAB_CMDS):
TypeError: object of type 'NoneType' has no len()
 (gmoccapy:64)
```
**Note:** this doesn't happen in LinuxCNC 2.8. I have no idea why.

### Solution:
Add something like    
``` python
if self.TAB_CMDS is None or self.TAB_LOCATIONS is None:
   return # or do something other
```
before qt_istat.py:422 (in _update()_)
https://github.com/LinuxCNC/linuxcnc/blob/05eb2be0133ad0b1d35de8e00bdc1b62fbbe5b46/lib/python/qtvcp/qt_istat.py#L422

Or put this in a try except block:
``` python
        if self.TAB_NAMES is not None and len(self.TAB_NAMES) != len(self.TAB_CMDS):
            log.critical('Embedded tab configuration -invalid number of TAB_NAMES vs TAB_CMDs')
        if self.TAB_NAMES is not None and len(self.TAB_LOCATIONS) != len(self.TAB_NAMES):
```

@c-morley this is more or less your code, so I thought it's best if you add a fix according to your flavor ;-)

c-morley, 2023-02-08:
TAB_LOCATIONS will never be None but could be [] 
That could be changed too, so locations can be None, for consistency

I like the check better then the try block, just so it can report to the user what the actual problem is.

What are legal tab combinations for Gmoccapy?

hansu, 2023-02-08:
> TAB_LOCATIONS will never be None but could be []

Okay then it was only _TAB_CMDS_ which was _None_

> What are legal tab combinations for Gmoccapy?

In the documentation somewhere it states that `TAB_LOCATIONS` is not mandatory (defaults to the main area); however a test resulted that it didn't show the VCP.

c-morley, 2023-02-09:
http://linuxcnc.org/docs/2.9/html/gui/gmoccapy.html#gmoccapy:configuration-of-tabs-and-side-panels
Seems to indicate that gmoccapy has the same requirements as qtvcp. AXIS and TOUCHY are slightly different:
http://linuxcnc.org/docs/2.9/html/gui/gladevcp.html#gladevcp:embedding-tab
I wonder if in the future we should make them all the same?

Anyways, I will push a fix for the error you described, thanks for the suggestions and discussion.

c-morley, 2023-02-09:
Ok pushed to 2.9/master

hansu, 2023-02-09:
> I wonder if in the future we should make them all the same?

Yeah that would be good so far that is possible.

closed 2023-02-09 by hansu