LinuxCNC/linuxcnc #2329: Probe Screen does not revert to machine units after running .ngc

Stutchbury, 2023-02-09:
## Here are the steps I follow to reproduce the issue:

 1. Install Probe Screen NG (there are many, many forked repositories but I used this one: https://github.com/andypugh/probe-screen-ng )
 2. Load a .ngc file and run it (I chose spiral.ngc and stopped it part way through)
 3. Switch to the Probe Screen and attempt to probe X+.

## This is what I expected to happen:
Probe will move in the machine units using the values specified in the Probe Screen.

## This is what happened instead:
Probe moved using the units of the .ngc file (inches) rather than the machine units (mm) used in the Probe Screen.

## It worked properly before this:
I have not yet been able to get the Probe Screen to work (see: https://forum.linuxcnc.org/38-general-linuxcnc-questions/48184-probe-screen-v2-python-file-not-open-errors#264062 )

## Information about my hardware and software:
* I am using a Raspberry Pi 4 (2Gb).
 * I am using this kernel version: Linux linuxcnc 4.19.71-rt24-v7l+ #1 SMP PREEMPT RT Fri Jan 1 21:15:16 GMT 2021 armv7l GNU/Linux (the update manage still says I need to update to the exact same kernel version but then fails with a permissions error if I try)
 * I am running ...
   * [ ] A binary version from linuxcnc.org
 * I am using this LinuxCNC version : 2.8.4
 * I am using this user interface (GUI): gmoccapy
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): Mesa 7i96 

I'm opening this issue at the suggestion (instruction?) of Andy Pugh ( https://forum.linuxcnc.org/38-general-linuxcnc-questions/48184-probe-screen-v2-python-file-not-open-errors#264082 )

As an aside, if there is anything I can do to help get probing 'into the core' I'd be more than happy to do so - it is arguably a pretty fundamental function of a modern CNC system?
The Probe Screen looks very complete but there are dozens of Probe Screen V2 and NG git forks. I have trawled through many of them to try to get the Probe screen working on 2.8.4 but it's proving a bit of a challenge to work out if it's version incompatibility, bugs in the code or configurations issues that I'm facing.

c-morley, 2023-02-11:
Since probe screen is not part of linuxcnc, it seems weird Andy suggested you add this issue here.

linuxcnc actually has a qt version of probe screen (called versa probe) which is based on version 2 included. 
http://linuxcnc.org/docs/stable/html/gui/qtvcp_widgets.html#_dialog_versaprobe
Another closely related one is 'basic probe'.
http://linuxcnc.org/docs/stable/html/gui/qtvcp_widgets.html#_basicprobe

It has since been improved and made a bit easier to integrate in 2.9.
In 2.9 it is natively available in qtdragon and should be embeddable in other screens.

The 2.8 version would need careful testing, error handling is much better on the 2.9 version.
In 2.8 a user must build the qt panel to embed it into other screens, in 2.9 there is a prebuilt one.

So probe routines are coming.

Stutchbury, 2023-02-12:
Thank you for your reply Chris. Good to hear probe routines are coming but your reply has raised a couple of slightly off topic questions. The Probe Screen is/was good but clearly not maintained within the core of LinuxCNC. Of the GUIs, Axis is the 'official' one (according to the documentation) with gmoccapy taking a close second - but with versa probe (created with qtvcp) poised to be the official probing solution, does that mean qtdragon is poised to replace Axis as the official UI in 2.9 (ie, will qtvcp be available in the standard images)?
I have an old NUC on the way, so will try to use that as a test for 2.9.

c-morley, 2023-02-12:
'official' is pretty loose term. 'popular' is probably a better term.
AXIS is not going anywhere, nor Gmoccapy. QtDragon was added in 2.8
Each has it's merits and annoyances.
qtvcp is available in 2.8

andypugh, 2023-05-08:
Should be fixed by https://github.com/andypugh/probe-screen-ng/commit/a6fb0d20ee1a1658606f5cf18d2ae14cafb47766

dpslwk, 2023-07-24:
Do QtDragon and QtDragon_hd's probing routings also have this issue?

c-morley, 2023-07-24:
2.9 has code to switch units back and forth. 2.8 I think just switches the units to what the probe routine uses. (qtdragon)