LinuxCNC/linuxcnc #2331: Hostmot2 encoder counts and HAL encoder counts are 32 bits and can wrap.

andypugh, 2023-02-09:
The stepgens hold a step accumulator internally in an s64 (long long) and the smart-serial encoder counter does the same. 

HAL now supports 64-bit pins, so we could simply change the type of counts and rawcounts pins, but this is likely to break existing configs. 

Possibly the answer is to create counts64 and rawcounts64 pins, use those to calculate (64-bit float) position and then truncate the result into the existing pins. 

Or we could simply hide the internal accumulator, like the smart-serial driver does:

andypugh, 2023-02-09:
See https://forum.linuxcnc.org/38-general-linuxcnc-questions/48181-correct-handling-of-a-wrapping-s32-number-in-a-custom-comp#264182 for the context that brought this up.

petterreinholdtsen, 2023-02-10:
Staying backwards compatible for a few releases seem like a good idea,
perhaps even with a runtime warning if the old 32-bit counters are used.
This would mean new names for the 64 bit values, perhaps 'totalcounts'
something similar instead of adding type information in the name?

-- 
Happy hacking
Petter Reinholdtsen

pcw-mesa, 2023-02-12:
Do utilities like halshow know how to display 64 bit integers?

andypugh, 2023-02-21:
> Do utilities like halshow know how to display 64 bit integers?

Probably not, I don't see any changes to them in this commit:
https://github.com/LinuxCNC/linuxcnc/commit/7a4c0908567e27f034c7cb2fe2f6c5b1341c933c

hansu, 2023-05-21:
Yes it does not:
[image]
But the problem is here that even halcmd cannot handle 64 bit types:
 ```
$ halcmd show pin abs-s64
Component Pins:
Owner   Type  Dir         Value  Name
    40  undef IN      undef      abs-s64.0.in
    40  bit   OUT         FALSE  abs-s64.0.is-negative
    40  bit   OUT         FALSE  abs-s64.0.is-positive
    40  undef OUT     undef      abs-s64.0.out
    40  bit   OUT         FALSE  abs-s64.0.sign
    40  s32   OUT           434  abs-s64.0.time

$ halcmd getp abs-s64.0.out
unknown_type
```

hansu, 2023-05-21:
With https://github.com/LinuxCNC/linuxcnc/pull/2500 halshow also supports the 64 bit integer types.