LinuxCNC/linuxcnc #2335: Gmoccapy: unify INI file settings

hansu, 2023-02-11:
Make Gmoccapy using those INI values in [DISPLAY] to suppress warnings:

```
DEFAULT_SPINDLE_0_SPEED
MIN_SPINDLE_0_SPEED 
MAX_SPINDLE_0_SPEED
MAX_SPINDLE_0_OVERRIDE
MIN_SPINDLE_0_OVERRIDE
```

See also: https://github.com/LinuxCNC/linuxcnc/issues/2273#issuecomment-1426653896

gmoccapy, 2023-10-07:
I would suggest to change "getiniinfo.py" from gmoccapy to look first for:
`DEFAULT_SPINDLE_0_SPEED
MIN_SPINDLE_0_SPEED 
MAX_SPINDLE_0_SPEED
MAX_SPINDLE_0_OVERRIDE
MIN_SPINDLE_0_OVERRIDE`

and if not finding this entries look for:
`DEFAULT_SPINDLE_SPEED
MAX_SPINDLE_OVERRIDE
MIN_SPINDLE_OVERRIDE
`
and if both are not in the INI use the mentioned default values from getiniinfo.

That way users using the actual version do not get errors after an update.
If we than release 2.10 we may drop the old entries.

Second solution would be that we check all INI entries at gmoccapy start and if the are not in the INI we add them and inform the user using a popup window, so he is able to adapt to his needs.

hansu, 2023-10-08:
First solution sounds like a good idea.

>Second solution would be that we check all INI entries at gmoccapy start and if the are not in the INI we add them and inform the user using a popup window, so he is able to adapt to his needs.

I would rather throw a warning on every start until the user adds the values manually. I wouldn't add default values to the ini values by linuxcnc. And also the INI file shouldn't be written by LinuxCNC as it is normally read only and it could damage the the file.

gmoccapy, 2023-10-08:
As this issue is assigned to you, I will leave the decision on your side ;-)
just give me a message, so I can review and push.

Norbert

andypugh, 2023-10-11:
I would prefer the GUI to look in the [SPINDLE] section for this. 
http://linuxcnc.org/docs/2.9/html/config/ini-config.html#sub:ini:sec:spindle-num

gmoccapy, 2023-10-16:
IMHO we should get away from gmoccapy/getiniinfo and use the Gladevcp library INFO 

http://linuxcnc.org/docs/2.9/html/gui/gladevcp-libraries.html

This would save some code in gmoccapy and also not duplicate stuff.

Whats your opinion?

c-morley, 2023-10-17:
@andypugh  GUI spindle settings can be different from hardware settings. Ie in manual mode maybe you wish to limit the max spindle speed (but only in manual mode). Similar idea as limiting max jogging speed to less then max axis velocity.
Though I probably agree if there are no [DISPLAY] spindle settings it should use the [SPINDLE] settings.

hansu, 2023-10-17:
@gmoccapy 
 >IMHO we should get away from gmoccapy/getiniinfo and use the Gladevcp library INFO
This would save some code in gmoccapy and also not duplicate stuff.
Whats your opinion?

Sounds good. Let me have look how it would look like...

Another thing: Currently gmoccapy looks in DISPLAY for `MIN_SPINDLE_0_SPEED` and `MAX_SPINDLE_0_SPEED` and take those values as start values for the GUI settings ["Spindle bar min" and "Spindle bar max" ](https://user-images.githubusercontent.com/67957916/218247320-42093476-2707-416f-ae7e-7bc8714f2eef.png) in case those are non existent in the pref file. After that the values from the INI file are not used any more. Makes sense somehow but also a bit unexpected. Maybe remove those settings from the INI file and take the value from the [SPINDLE] section als start values for the GUI entries?

What do you think?

@andypugh @c-morley 
> Though I probably agree if there are no [DISPLAY] spindle settings it should use the [SPINDLE] settings.

I agree

hansu, 2024-09-22:
@c-morley  I started to use the GladeVCP Info lib to get the INI values.

1. Multiplying the value by 100 makes the library not really universally usable for others. I think the library should not do any conversions by itself:
    ```
            self['MAX_SPINDLE_{}_OVERRIDE'.format(i)] = float(
                self.get_error_safe_setting("DISPLAY", "MAX_SPINDLE_{}_OVERRIDE".format(i), 1)) * 100
            self['MIN_SPINDLE_{}_OVERRIDE'.format(i)] = float(
                self.get_error_safe_setting("DISPLAY", "MIN_SPINDLE_{}_OVERRIDE".format(i), 0.5)) * 100   

2. The way that `_IStat ` provides own default values is also sub-optimal - there should be an option to set the defaults by the GUI which uses this class. 
2. The class `_IStat `is used by gladevcp/core.py but is located in qtvcp/qt_istat.py. As it is commonly used - it might be better if it is located in a common directory.
3. As it is currently a subclass of QTvcp - the Log messages would look like this:
`[Gmoccapy.QTVCP.QT_ISTAT][WARNING]  ...`. This is a bit confusing as Gmoccapy does not use QTVcp in the default configuration.

gmoccapy, 2024-09-22:
@hansu Do you think it is a good idea to mix QT and Glade stuff?
IMHO not, If INI reading should be repoved from GUI code, we should integrate that in the LinuxCNC python handler.

Norbert

hansu, 2024-09-22:
> @hansu Do you think it is a good idea to mix QT and Glade stuff? IMHO not, If INI reading should be repoved from GUI code, we should integrate that in the LinuxCNC python handler.
> 
> Norbert

No, but it is already mixed. I just followed your suggestion from https://github.com/LinuxCNC/linuxcnc/issues/2335#issuecomment-1764722814 and was also a bit surprised that using the GladeVCP Info Library invokes QTVCP classes.

c-morley, 2024-09-22:
1) We just have to agree on units - nobody wants to change their code. I chose percent because people understand 50%, better the .5 of maximum. Qtvcp/Gladevcp is made for users to code customization. It would be a lot more work to change Qtvcp and all the GUIs it has. One possibility is that a screen could import the common class and then rescale what it wants.

 2) Again, if we agree on reasonable defaults then things are simpler. Otherwise we need to switch to something like function calls for each read INI entry that would include a default. the idea of the warnings of missing INI entries is to help users actually set the entries and not use defaults.

3) yes a common set of classes in it's own directory would be nicer. It would promote GUI builders to think collaboratively more I suppose 

4) I'm sure that could be changed. The log class could be in a common directory too. But I personally don't think that's a first priority problem, only developers read that first part, and it just tells you were the log came from. That's extra work just to remove Qt from the name :)

As for Qtvcp in Gladevcp etc - I think we get hung up too much in a name. GMoccapy is not lesser because it uses Qtvcp libraries. As discussed in 3) we can make that a different printed name if you like, but it doesn't really change anything. Where the log came from is less important then what it is actually telling you.

> 
quote>>
Another thing: Currently gmoccapy looks in DISPLAY for MIN_SPINDLE_0_SPEED and MAX_SPINDLE_0_SPEED and take those values as start values for the GUI settings ["Spindle bar min" and "Spindle bar max" ](https://user-images.githubusercontent.com/67957916/218247320-42093476-2707-416f-ae7e-7bc8714f2eef.png)in case those are non existent in the pref file. After that the values from the INI file are not used any more. Makes sense somehow but also a bit unexpected. Maybe remove those settings from the INI file and take the value from the [SPINDLE] section als start values for the GUI entries?
<<
> 

What if there isn't a [SPINDLE] entry? You still need a default.
[DISPLAY] is for the screen
[SPINDLE] is for hardware
Maybe a user wants to limit the screen min/max while in manual mode vrs when in auto.
This is how you do that.

The way I would do it:
read/record the INI [DISPLAY] entry in ISTAT -adding a message if it's missing
read/record the INI [SPINDLE] entry ISTAT
In the GUI code decide which I want to use as the default

hansu, 2024-09-22:
1.
>I chose percent because people understand 50%, better the .5 of maximum. 

Yeah, but you have 0.5 in the INI file and use 50 % internally in the code. So the user won't see only the 0.5 ...

2.
>Again, if we agree on reasonable defaults then things are simpler. 

 Ok let's start with this.

c-morley, 2024-09-23:
Well we could take this opportunity to use percent in the ini and the GUI code.

So right now istat reads the INI and sets the gatiahkes in one big function.
If we broke that up into many individual functions then it allows subclassing for different function behavior.  I'm just afraid that will be abused. I would really like all the GUIS to use  the same INI definitions for common stuff so that changing Guis isn't so tough on users.

Using functions would easily allow different defaults if required.

I guess we need a list of important common INI variables.

gmoccapy, 2024-09-23:
Percentage is fine for me.
Looking for spindle entries in the display section is a very old relict from the beginning where gmoccapy was based on screen code, as far as I remeber.

Norbert

hansu, 2024-09-23:
>Well we could take this opportunity to use percent in the ini and the GUI code.

Changing all INI files of the sim config and breaking all existing configs - I don't think that is a good idea....

andypugh, 2024-09-23:
> > Well we could take this opportunity to use percent in the ini and the GUI code.
> 
> Changing all INI files of the sim config and breaking all existing configs - I don't think that is a good idea....

We might be able to do it in the update_ini script. 

It seems likely that any scale < 1 is fractional and anything > 1 is percentage.

hansu, 2024-09-23:
> We might be able to do it in the update_ini script.

Of course we can do this, but then we can't switch back to 2.9 and that just for some kind of aesthetic reason ...

c-morley, 2024-09-26:
WelI think it would be helpful for users if we used scaling that makes sense, but that is the reason I didn't change the INI scaling but did change it inside Qtvcp.
If we are going to change it, master doesn't guarantee backwards compatibility.
It's a tough decision.

hansu, 2024-09-26:
[Andy]
>It seems likely that any scale < 1 is fractional and anything > 1 is percentage.

But > 1 could also be an absolute value, so a fractional value < 1 makes totally sense for me.

`MIN_SPINDLE_0_OVERRIDE = 50 ` for example could also be interpreted as a limit of the overriden spindle speed (even though if not very likely)

Stutchbury, 2024-11-03:
> [Andy]
>
>
> But > 1 could also be an absolute value, so a fractional value < 1 makes totally sense for me.
>
> `MIN_SPINDLE_0_OVERRIDE = 50 ` for example could also be interpreted as a limit of the [...]

Linked here from [Aciera's post](https://forum.linuxcnc.org/38-general-linuxcnc-questions/54327-ini-file-clarification-roadmap#313717) (thank you) 

For *_OVERRIDE options, the end result would always be a percent (ie both 50 and 0.5 would result in 50%) but coincidentally we have recently been discussing the ability to set SPINDLE_INCREMENT as either an absolute RPM value (> 1) or a percentage value (< 1) - in order to enable logarithmic-like spindle control in both manual mode.

Also, is there any value having each GUI use its own section, reserving the [DISPLAY] section for core 'official' options? I am likely going to create a [MANUALMATIC] section in the future, partly because of the varying usage of the ini file by the different GUIs but also because the pendant is effectively a second GUI/input.

Stutchbury, 2024-11-07:
> @Stutchbury I think it is good to have common values in the DISPLAY section to make it easier to switch between or change to another GUI.

Absolutely - but the key is _common values_. The `[DISPLAY]` section should contain solely a curated list of entries that are common to all GUIs. All entries that are not in this list should be in a custom section (eg `[GMOCCAPY]`).

We currently have multiple similarly named entries so when changing a value it will often not affect your chosen GUI.

hansu, 2024-11-07:
... and additionally almost each GUI have some preferences you can change in the GUI itself.

Stutchbury, 2024-11-07:
Hehe - yes, gmoccapy has spindle values in preferences that appear to ignore the ini file `[SPINDLE_0]` limits! But that's another issue...

hansu, 2024-11-07:
Yes and my machine config doesnt even have a SPINDLE or SPINDLE_0 section and it works also. :D

c-morley, 2024-11-08:
> @c-morley This for example is confusing
> 
> https://github.com/LinuxCNC/linuxcnc/blob/f14661683ca143e360319f3f3e43ada301e8d064/lib/python/qtvcp/qt_istat.py#L358
> 
> because the uppercase style lets one think that this is the name from the INI file which is correct in the majority of cases. Otherwise the name could also have been `self.linear_jog_vel`.
> Can we change it that way?
> 
> ```diff
> + self.MIN_LINEAR_VELOCITY = float(self.get_error_safe_setting("DISPLAY", "MIN_LINEAR_VELOCITY", 0)) * 60
> - self.MIN_LINEAR_JOG_VEL = float(self.get_error_safe_setting("DISPLAY", "MIN_LINEAR_VELOCITY", 0)) * 60
> ```
> 
> The same for
> 
> ```
> self.DEFAULT_LINEAR_JOG_VEL
> self.MIN_LINEAR_JOG_VEL
> self.MAX_LINEAR_JOG_VEL
> self.DEFAULT_ANGULAR_JOG_VEL
> self.MIN_ANGULAR_JOG_VEL
> ```

They have JOG in the name so as not to be confused with AXIS velocities.
If you had to change it, I'd prefer to change the INI entry to include  the word JOG.

hansu, 2024-11-08:
>They have JOG in the name so as not to be confused with AXIS velocities.

Sure, but as it only gives the values of the DISPLAY section, the confusion is minimal

>If you had to change it, I'd prefer to change the INI entry to include the word JOG.

I agree that this would be better, but would also lead to a bigger change affecting multiple GUIs...

c-morley, 2024-11-09:
INFO does not just give DISPLAY section values. It supplies information from any INI section a GUI might need.

I think we should put the effort into consolidating the GUI info for all the screens.
We can mitigate the transition by allowing old entries too - just adding a warning about it.

Sigma1912, 2024-11-09:
> I think we should put the effort into consolidating the GUI info for all the screens.

This _sounds_ sensible but GUIs are not (and will never be) created equal. How would this solve the issue of having a bunch of entries in the documentation of the [DISPLAY] section that my particular GUI might not even use?

Stutchbury, 2024-11-09:
> I think we should put the effort into consolidating the GUI info for all the screens. We can mitigate the transition by allowing old entries too - just adding a warning about it.

+1 For a curated list of common `[DISPLAY]` options that GUI developers *should* use (and migrate to) with the guidance that any options that are not present in the curated list should not be added by GUI developers to the `[DISPLAY]` section but to a custom section (eg `[GMOCCAPY]` or `[MAMNUALMATIC]`. 

The existing anomalies should be marked as deprecated in the main `ini` file documentation as well as in the GUI specific docs because anyone using Axis is unlikely to go searching through gmoccapy docs! Eventually the main docs would therefore include only common options with the 'extension' options being documented (only) in the GUI's docs.