LinuxCNC/linuxcnc #2339: Missing parentheses in homing.c?

hansu, 2023-02-11:
Compiling gives this warning:

```
emc/motion/homing.c:1345:18: warning: suggest parentheses around operand of ‘!’ or change ‘&’ to ‘&&’ or ‘!’ to ‘~’ [-Wparentheses]
             if ( ! H[joint_num].home_flags & HOME_ABSOLUTE_ENCODER) {
                  ^~~~~~~~~~~~~~~~~~~~~~~~~
```

I guess your intention with

```c
if ( ! H[joint_num].home_flags & HOME_ABSOLUTE_ENCODER) {
```
was
```c
if ( !(H[joint_num].home_flags & HOME_ABSOLUTE_ENCODER) ) {
```
not
```c
if ( (!H[joint_num].home_flags) & HOME_ABSOLUTE_ENCODER ) {
```
which is actually equal to the first one.

andypugh, 2023-02-23:
Yes, as you say, it is meant to be ! (flags & mask)
Though, it works as bitwise ! of the flags and then and-ing the mask is equivalent to masking the flags and then inverting.

andypugh, 2023-02-23:
Fixed via 26e2ce16cfe95e519460415b08303c448b2c6028

closed 2023-02-23 by andypugh