LinuxCNC/linuxcnc #2353: Pncconf Wizard in 2.9x: Spindle StepGen wrong HAL settings for .maxvel and .maxaccel

thomam04, 2023-02-20:
When using Pncconf Wizard in LCNC 2.9/2.10 to configure a Spindle via StepGen, it created the following HAL settings:

setp   hm2_7i76e.0.stepgen.03.maxaccel         [SPINDLE_0]MAX_ACCELERATION
setp   hm2_7i76e.0.stepgen.03.maxvel             [SPINDLE_0]MAX_VELOCITY

MAX_VELOCITY / MAX_ACCELERATION in the INI is now in RPM, but the STEPGEP still requires RPS. LCNC will show the following error when you start the Spindle:

hm2/hm2 stepgen.03.maxvel is too big for current step timings & position-scale, clipping to max possible.

The correct settings in HAL should be:

setp   hm2_7i76e.0.stepgen.03.maxaccel         [SPINDLE_0]STEPGEN_MAXACCEL
setp   hm2_7i76e.0.stepgen.03.maxvel             [SPINDLE_0]STEPGEN_MAXVEL

c-morley, 2023-02-20:
Ok I switched the setting in 2.9 and master - thank you.

c-morley, 2023-02-21:
This may still be wrong. STEPGEN_MAXVEL is probably still be in RPS.
Can you confirm?

pcw-mesa, 2023-02-21:
STEPGEN_MAXVEL is now correct (in RPS)
but MAX_VELOCITY is wrong (it should be in RPM)

thomam04, 2023-02-21:
Example for 24000 RPM spindle:

[SPINDLE_0]
# Max spindle speed in RPM (LinuxCNC GUI)
MAX_VELOCITY = 24000
# Max spindle acceleration in RPM per second (LinuxCNC GUI)
MAX_ACCELERATION = 5000

# Hardware StepGen requires values in RPS (revolution per second)
STEPGEN_MAXVEL = 400
STEPGEN_MAXACCEL = 500

The general StepGen comment below should probably be revised as well - for a Spindle StepGen. 

# The values below should be 25% larger than MAX_VELOCITY and MAX_ACCELERATION
# If using BACKLASH compensation STEPGEN_MAXACCEL should be 100% larger.

thomam04, 2023-02-21:
<pre>Example for 24000 RPM spindle:

[SPINDLE_0]
# Max spindle speed in RPM (LinuxCNC GUI)
MAX_VELOCITY = 24000
# Max spindle acceleration in RPM per second (LinuxCNC GUI)
MAX_ACCELERATION = 5000

# Hardware StepGen requires values in RPS (revolution per second)
STEPGEN_MAXVEL = 400
STEPGEN_MAXACCEL = 500

The general StepGen comment below should probably be revised as well - for a Spindle StepGen. 

# The values below should be 25% larger than MAX_VELOCITY and MAX_ACCELERATION
# If using BACKLASH compensation STEPGEN_MAXACCEL should be 100% larger."</pre>

SebKuzminsky, 2023-04-23:
@c-morley Is the remaining work here captured by #2354 ?  Can we close this one?

grandixximo, 2026-10-05:
The HAL wiring was fixed in b9e449a625 (2023-02-20, in 2.9.0): spindle stepgens now get setp ... .maxvel [SPINDLE_0]STEPGEN_MAXVEL and .maxaccel [SPINDLE_0]STEPGEN_MAXACCEL. The remaining MAX_VELOCITY units question is tracked in #2354. Closing this one.

closed 2026-10-05 by grandixximo