LinuxCNC/linuxcnc #2354: pncconf uses [SPINDLE_0] in INI and so does motion (?) - they assume different units

c-morley, 2023-02-20:
Apparently motion uses [SPINDLE_0] for spindle settings, so did Pncconf.
MAX_VELOCITY is in RPM for motion and RPS in Pncconf.
Has caused some user problems.

pcw-mesa, 2023-02-20:
The issue I see is not with MAX_VELOCITY (which is always in RPM for spindles) 
but rather STEPGEN_MAXVEL which is in RPS for spindles, so if a stepgen controlled
spindle is setup, the stepgen values need to be divided by 60.

c-morley, 2023-02-21:
Well the first problem is Pncconf already used [SPINDLE_0] in the INI, so when motion  was changed to use it - name clashes were  bound to happen.

As for MAX_VELOCITY vrs STEPGEN_MAXVEL I put a fix in, but now I probably got it wrong based on what you just said.. lol

This issue is really just so it can be remembered to reassess pnccond's spindle entries in the future.

SebKuzminsky, 2023-04-23:
@c-morley Do you think this is something that needs to be fixed before the 2.9 release?