LinuxCNC/linuxcnc #2358: Python problem with 2.9 version

TadyTheFish, 2023-02-23:
I have been trying to make this python based project to work with the 2.9 version of LinuxCNC
The author of the project made it with the 2.8 version and it works.
`https://github.com/Stutchbury/Manualmatic-Pendant`
If I run this with the 2.9 (2.10) version, LinuxCNC throws this...
`AttributeError: 'linuxcnc.stat' object has no attribute 'axes'. Did you mean: 'axis'?`
I added `print(dir(linuxcnc.stat))` to the component file to see the available attributes of the object linuxcnc.stat, and I get this...
`['__class__', '__delattr__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', 'acceleration', 'active_queue', 'actual_position', 'adaptive_feed_enabled', 'ain', 'angular_units', 'aout', 'axis', 'axis_mask', 'block_delete', 'call_level', 'command', 'current_line', 'current_vel', 'cycle_time', 'debug', 'delay_left', 'din', 'distance_to_go', 'dout', 'dtg', 'echo_serial_number', 'enabled', 'estop', 'exec_state', 'feed_hold_enabled', 'feed_override_enabled', 'feedrate', 'file', 'flood', 'g5x_index', 'g5x_offset', 'g92_offset', 'gcodes', 'homed', 'ini_filename', 'inpos', 'input_timeout', 'interp_state', 'interpreter_errcode', 'joint', 'joint_actual_position', 'joint_position', 'joints', 'kinematics_type', 'limit', 'linear_units', 'lube', 'lube_level', 'max_acceleration', 'max_velocity', 'mcodes', 'misc_error', 'mist', 'motion_id', 'motion_line', 'motion_mode', 'motion_type', 'num_extrajoints', 'optional_stop', 'paused', 'pocket_prepped', 'poll', 'position', 'probe_tripped', 'probe_val', 'probed_position', 'probing', 'program_units', 'queue', 'queue_full', 'queued_mdi_commands', 'rapidrate', 'read_line', 'rotation_xy', 'settings', 'spindle', 'spindles', 'state', 'task_mode', 'task_paused', 'task_state', 'tool_from_pocket', 'tool_in_spindle', 'tool_offset', 'tool_table', 'velocity']`

It seems that the axes attribute is missing. Can someone check this? I can't find where in the source code could this be defined.
Also if I remove the code that needs the attribute axes, it complains that the linuxcnc.stat.poll needs an argument ( 2.8 version did not )
Thank you

andypugh, 2023-02-23:
It seems that there was a warning that this would disappear in 2.8:
https://github.com/LinuxCNC/linuxcnc/blob/2.8/src/emc/usr_intf/axis/extensions/emcmodule.cc#L687

And it was removed by this commit:
https://github.com/LinuxCNC/linuxcnc/commit/ffe664c0422887c899be5cd59de6b1cb1d1bbc4c

Which also includes some indication of what has replaced it (and why) 

Your simplest answer, assuming a trivial kinematics machine with no missing axes would be to simply replace each instance with the highest axis index on your machine as a number (ie 2 for an xyz machine). 

The right solution would be to iterate through the axis mask and check the bits.

TadyTheFish, 2023-02-23:
Thank you! 
I saw your comment on the Manualmatic git page:)

closed 2023-02-23 by TadyTheFish

TadyTheFish, 2023-02-23:
Maybe update the docs. If it's obsolete add a note to the docs or remove the description