LinuxCNC/linuxcnc #2364: "gcc version 12.2.1 too old: skipping hal/user_comps/xhc-whb04b-6" when minimum version is 4.7

InnerBushman, 2023-02-26:
GCC version is higher than 'MIN_VERSION := 4.7' yet the compilation of this module is skipped.
submakefile in question: https://github.com/LinuxCNC/linuxcnc/blob/2.8/src/hal/user_comps/xhc-whb04b-6/Submakefile

## Here are the steps I follow to reproduce the issue:
 1. Compile with GCC 12.x (probably any version since 10.x)

## This is what I expected to happen:
Compilation should proceed as normal

## This is what happened instead:
gcc version 12.2.1 too old: skipping hal/user_comps/xhc-whb04b-6

## Information about my hardware and software:

 * I am using this Linux distribution and version:
 Distributor ID:	ManjaroLinux
 Release:	22.0.2

 * I am using this LinuxCNC version:
linuxcnc 2.8.4 https://aur.archlinux.org/packages/linuxcnc

dwrobel, 2023-03-07:
This is fixed since 2020 by fb0335fce47df94b8ef5d97870f478feb7a97c26 and 45d3c95d2143ea3951f57c04834c685005f12fb0 commits and is available in both `master` and `2.9` branches.

SebKuzminsky, 2023-04-23:
I think we can just delete this check.

andypugh, 2023-04-24:
cherry-picked the check from 2.9 instead of deleting. I put the test in, and feel that I must have had a reason.