LinuxCNC/linuxcnc #2367: Raspberry Pi 4 SPI does not work on Debian Bookworm

SebKuzminsky, 2023-02-27:
The hm2_rpspi driver does not work for me on Raspberry Pi 4 with Debian Bookworm: rtapi_app gets "permission denied" when trying to open `/dev/mem`, even after `sudo make setuid`. I think it's because the kernel is built with [CONFIG_STRICT_DEVMEM](https://man7.org/linux/man-pages/man4/mem.4.html). 

And the normal hm2_spi driver doesn't work, because the Pi's SPI hardware is disabled by the default DTB.

petterreinholdtsen, 2023-02-28:
[Sebastian Kuzminsky]
> The hm2_rpspi driver does not work for me on Raspberry Pi 4 with
> Debian Bookworm: rtapi_app gets "permission denied" when trying to
> open `/dev/mem`, even after `sudo make setuid`. I think it's because
> the kernel is built with
> [CONFIG_STRICT_DEVMEM](https://man7.org/linux/man-pages/man4/mem.4.html).

Are you sure?   I read the mem(4) page to mean that the device should be
available for opening, but not all memory locations are available for
reading.  Which part of /dev/mem do rtapi_app need to read?

> And the normal hm2_spi driver doesn't work, because the Pi's SPI
> hardware is disabled by the default DTB.

This should be easy to fix by updating the config file and rebooting?

-- 
Happy hacking
Petter Reinholdtsen

SebKuzminsky, 2023-03-01:
> Are you sure? I read the mem(4) page to mean that the device should be available for opening, but not all memory locations are available for reading. Which part of /dev/mem do rtapi_app need to read?

I'm not sure why it fails, but I'm sure *that* it fails.  hm2_rpspi successfully opens /dev/mem, then fails to mmap it with EPERM, even when running as root, even with CAP_SYS_RAWIO.

https://github.com/LinuxCNC/linuxcnc/blob/2.9/src/hal/drivers/mesa-hostmot2/hm2_rpspi.c#L1031-L1037

The `rtapi_open_as_root()` succeeds, and the `mmap()` fails.  It's trying to map the whole peripheral, using the offset and size it reads from the device tree:

https://github.com/LinuxCNC/linuxcnc/blob/2.9/src/hal/drivers/mesa-hostmot2/hm2_rpspi.c#L1330-L1348

After turning up the logging output on the driver i get this:

~~~
hm2_rpspi: Platform: Raspberry Pi 4 Model B Rev 1.2
hm2_rpspi: Base address 0xfe000000 size 0x01800000
hm2_rpspi: Can't map peripherals: Operation not permitted (1)
~~~

The base address and size are straight out of `/proc/device-tree/soc/ranges`, and matches what I see there with hexdump.  That's apparently trying to map the whole periperhal, not just the SPI registers.

> > And the normal hm2_spi driver doesn't work, because the Pi's SPI hardware is disabled by the default DTB.

> This should be easy to fix by updating the config file and rebooting?

I would have thought so too, but the Raspberry Pi OS instructions of adding `dtparam=spi=on` to `/boot/config.txt` (which is `/boot/firmware/config.txt` on Debian) and rebooting does not have any effect.  `/proc/device-tree/soc/spi@*/status` remains `disabled`.

Decompiling, editing, and re-compiling the DTB got me slightly further, but still no joy.  <https://lists.debian.org/debian-arm/2022/10/msg00059.html>

If you know of an easy fix I'm all ears :-)

petterreinholdtsen, 2023-03-01:
[Sebastian Kuzminsky]
> If you know of an easy fix I'm all ears :-)

I have used the recipe in
<URL: https://github.com/lightside-instruments/ice4pi/blob/master/example/README >,
"Make sure your Pi has SPI enabled (e.g. use raspi-config)".

-- 
Happy hacking
Petter Reinholdtsen

SebKuzminsky, 2023-03-01:
The `raspi-config` tool is available in Raspberry Pi OS but not in Debian.  I jumped through some hoops to install it in Debian (described somewhat in that thread on debian-arm linked above) but found out it relies on out-of-tree kernel patches that are part of the Raspberry Pi OS kernel but not part of the Debian kernel.

SebKuzminsky, 2023-03-01:
For contrast, I just installed Ubuntu 22.04 on my Pi and SPI works out of the box.  I'm trying to see if the SPI support there is something that can be easily moved to Debian...

SebKuzminsky, 2023-03-01:
(Oooh, ice4pi look interesting!)

SebKuzminsky, 2023-03-01:
Found this on the debian wiki: https://wiki.debian.org/RaspberryPi4#GPIO_support

And indeed, with `iomem=relaxed` on the kernel cmdline, the hm2_rpspi driver works on Debian Bookworm.

This kernel command line argument is described here: https://www.kernel.org/doc/html/v4.14/admin-guide/kernel-parameters.html#

samcoinc, 2023-03-01:
Are graphics working correctly?  Or a better question - how are you
installing bookworm on the pi4

SebKuzminsky, 2023-03-01:
> Are graphics working correctly?

As far as I can tell: mostly, yes.  Axis runs and OpenGL works.  There is a little bit of intermittent glitchy-looking video noise that flashes up on the bottom of the screen sometimes, and immediately goes away.  Doesn't seem to hurt anything.   

> how are you installing bookworm on the pi4

I'm using the linuxcnc disk image here, there's a pre-built image in the Actions tab, click on the latest green build and scroll down to Artifacts: https://github.com/SebKuzminsky/image-specs/ 

That image is based on the semi-official debian.org raspberry pi image here: https://raspi.debian.net/

SebKuzminsky, 2023-03-01:
This issue is fixed in the latest image (commit a7e037a) in https://github.com/SebKuzminsky/image-specs .

closed 2023-03-01 by SebKuzminsky