LinuxCNC/linuxcnc #2374: Gmoccapy DRO-size-4axis - doesnt work after restart

zz912, 2023-03-03:
Hello,

in 3 axis Gmoccapy this problem is not.

Simulate problem:
1) run linuxcnc (branche 2.9) 
/configs/sim/gmoccapy/gmoccapy_4_axis.ini
[image]
2) change DRO - size
[image]
3) DRO is correct changed
[image]
4) linuxcnc exit
5) run linuxcnc
6) The DRO is not changed, but it should be.
[image]
7) DRO - size value in setting is correct

hansu, 2023-03-03:
Yeah it seems that the DRO is always displayed smaller than the value from the preferences file.
As a workaround you can set 67 and after restart you have something like 50 ;-)

hansu, 2023-03-07:
The font size which is actually applied to the DRO elements is reduced if more than 3 DROs are present. But those factors are not taking into account when refreshing the GUI after the setting is changed. So currently it needs a restart to display they correctly.

closed 2023-03-14 by hansu