LinuxCNC/linuxcnc #2376: rip-environment - ignore halcomp

zz912, 2023-03-03:
Hello,

If I use RIP instalation, i have to write:
`zdenek@CNC:~/linuxcnc$ sudo ./linuxcnc-2.8_2020-10-21/bin/halcompile --install sincos.comp`
only:
`zdenek@CNC:~/linuxcnc$ sudo halcompile --install sincos.comp`
doesnt work,
If I use rip-environment, halrun works, but halcompile not.

In /scripts/rip-environment.in:
----------------------------------------------------------------------------------------------
```
!/bin/bash
 Execute this file in the context of your shell, such as with
  . @EMC2_HOME@/scripts/rip-environment
 and your shell environment will be properly configured to run commands like
 halcompile, halcmd, halrun, iosh, and python with the emc modules available.
```
----------------------------------------------------------------------------------------------

Debian instalation 2.8.4:
----------------------------------------------------------------------------------------------
```
zdenek@CNC:~/linuxcnc$ sudo halcompile --install sincos.comp
Compiling realtime sincos.c
Linking sincos.so
cp sincos.so /usr/lib/linuxcnc/modules/
```
----------------------------------------------------------------------------------------------

RIP instalation (Branche 2.8 or 2.9)
----------------------------------------------------------------------------------------------
```
zdenek@CNC:~/linuxcnc/linuxcnc-2.8_2020-10-21/scripts$ source rip-environment
This script only needs to be run once per shell session.
zdenek@CNC:~/linuxcnc$ halrun
halcmd: show comp
Loaded HAL Components:
ID      Type  Name                                            PID   State
     2  User  halcmd1393                                       1393 ready

halcmd: exit
Note: Using POSIX non-realtime
zdenek@CNC:~/linuxcnc$ sudo halcompile --install sincos.comp
[sudo] heslo pro zdenek: 
sudo: halcompile: příkaz nenalezen
zdenek@CNC:~/linuxcnc$ 
```
----------------------------------------------------------------------------------------------
příkaz nenalezen = command not founded

SebKuzminsky, 2023-03-03:
`sudo` sanitizes the environment, including PATH:
~~~ 
$ which halcompile
# (no output, command not found)

$ source linuxcnc-dev/scripts/rip-environment

$ which halcompile
/home/seb/linuxcnc-hacking/linuxcnc-dev/bin/halcompile

$ echo $PATH
/home/seb/linuxcnc-hacking/linuxcnc-dev/bin:/home/seb/linuxcnc-hacking/linuxcnc-dev/scripts:/home/seb/linuxcnc-hacking/linuxcnc-dev/tcl:/home/seb/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/usr/local/sbin:/usr/sbin:/sbin

$ sudo which halcompile
[sudo] password for seb:
# (no output)

$ sudo env | grep PATH
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
~~~

closed 2023-03-03 by SebKuzminsky

zz912, 2023-03-03:
What your Responce mean? Am I do something wrong?

```
jarda@CNC1:~$ which halcompile
/home/jarda/linuxcnc/linuxcnc-2.9/bin/halcompile

jarda@CNC1:~$ echo $PATH
/home/jarda/linuxcnc/linuxcnc-2.9/bin:/home/jarda/linuxcnc/linuxcnc-2.9/scripts:/home/jarda/linuxcnc/linuxcnc-2.9/tcl:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games

jarda@CNC1:~$ sudo which halcompile
[sudo] heslo pro jarda: 

jarda@CNC1:~$ sudo env | grep PATH
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
jarda@CNC1:~$ 
```

SebKuzminsky, 2023-03-03:
What i meant is, `sudo` undoes much of what `rip-environment` does, so that it can't find `halcompile`.

Try running `halcompile --install sincos.comp` without `sudo` (because in a rip-environment, you don't need to be root to install your comp).