LinuxCNC/linuxcnc #2379: Guide for Installing LinuxCNC with RealTime Kernel on a Raspberry Pi 3 Model A+

Gubby1, 2023-03-04:
############################################################################################

INSTALL RASPBERRY PI OS LEGACY (Debian 10 amd64 armhf) WITH DESKTOP ENVIRONMENT FROM THE PI IMAGER AND CONFIGURE SSH IN CASE YOU DONT HAVE DISPLAY

############################# BUILDING REALTIME KERNEL ##################################

sudo apt-get update
sudo apt-get install git bc mc libncurses5-dev
git clone --depth=1 https://github.com/raspberrypi/linux -b rpi-4.14.y-rt

cd linux
KERNEL=kernel7
make bcm2709_defconfig
make menuconfig

Kernel features | Preemption Model | FULLY PREEMPTABLE KERNEL #RT
exit
exit

make -j4 zImage modules dtbs //You can use -j3 or whatever number you want (if -j3, it will take around 5h to make)

sudo make modules_install
sudo cp arch/arm/boot/dts/.dtb /boot/
sudo cp arch/arm/boot/dts/overlays/.dtb* /boot/overlays/
sudo cp arch/arm/boot/dts/overlays/README /boot/overlays/
sudo cp arch/arm/boot/zImage /boot/$KERNEL.img

######################### BUILDING LINUXCNC##############################################

cd
git clone git://github.com/linuxcnc/linuxcnc.git linuxcnc-dev
sudo apt-get install dpkg-dev
cd linuxcnc-dev
git fetch origin 2.8
git checkout 2.8
cd debian

./configure uspace
cd ..

dpkg-checkbuilddeps

#BUILD LINUXCNC
cd
cd linuxcnc-dev/src

#REPEAT THIS TWO UNTIL ALL DEPENCIES ARE INSTALLED
./autogen.sh
./configure --with-realtime=uspace

######################################################################################
#IF YOUR SYSTEM RUNS OUT OF RAM, YOU CAN USE AN USB TO SET UP SWAP VIRTUAL MEMORY WITH THIS:
#FIND THE PARTITION YOU WANT TO USE AS SWAP MEMORY

sudo fdisk -l

#FORMAT THE PARTITION TO BE USED AS SWAP MEMORY

sudo mkswap /dev/sda1

#ACTIVATE THE SWAP PARTITION

sudo swapon /dev/sda1

########################################################################################

make
sudo make setuid

cd
cd linuxcnc-dev/
. scripts/rip-environment
linuxcnc

https://www.linuxcnc.org/dists/buster/2.8-rtpreempt/binary-armhf/linuxcnc-uspace_2.8.4_armhf.deb
https://www.linuxcnc.org/dists/buster/2.8-rtpreempt/binary-armhf/linuxcnc-uspace_2.8.1_armhf.deb

cd Downloads
sudo dpkg -i linuxcnc-uspace_2.8.4_armhf.deb

#Now you should be able to see the LinuxCNC on the main menu

andypugh, 2023-03-04:
You have raised this as an issue, but it seems to be purely informative. 
What action would you like from us to close this issue?

Gubby1, 2023-03-05:
> You have raised this as an issue, but it seems to be purely informative. What action would you like from us to close this issue?

That's the intention, to be informative for those who may need it

SebKuzminsky, 2023-04-23:
We put this on the wiki here: http://wiki.linuxcnc.org/cgi-bin/wiki.pl?RaspberryPi3

Thank you for your contribution!

closed 2023-04-23 by SebKuzminsky