LinuxCNC/linuxcnc #2387: Gmoccapy: Window width too big because of too many displayed h-buttons

hansu, 2023-03-08:
This affects the configurations "8-axis" and "9-axis":

[image]

After clicking one of the arrow-buttons, the number of displayed buttons is reduced and the window width can the decreased manually.

Brought to light by @SebKuzminsky's [config-test](https://github.com/SebKuzminsky/linuxcnc-config-test) if someone doesn't know that yet.

gmoccapy, 2023-10-08:
solved and fixed

closed 2023-10-08 by gmoccapy

hansu, 2023-10-08:
I noticed that also the minimal height is 836 pixels with 9 axes. This is because of the jog buttons:
[image]

Maybe we should show those buttons in this case in 5 columns and decrease the size. So something like this:

[image]

gmoccapy, 2023-10-08:
That might be a solution, but would require a complete rework of the place jog buttons part. I do think that actually Noone is using a 9 axis machine with linuxcnc, it is just the ability to show that it would be possible.

I prefer, that we both takes the little time we have to vćlear up the code, get rid of warnings and invest more time in CSS Styling etc. I would like rundet Buttons with shapes and aluminum background and an custom CSS file etc. Al the styling to be applied automatically to embedded tabs, as otherwise the user uses a dark interface with clearlooks embedded tabs, or he has to use a python file to load the embedded tabs just to set the design, etc...

commit 7d4f3a3b (2023-10-07): gmoccapy_3_4_5 - 8 and 9 axis made to much ref button due to a show all ccommand. This fixes #2387