LinuxCNC/linuxcnc #2388: homing not killed by F2 ky cycle

WOWElectronics, 2023-03-10:
# The issue tracker is not a support forum

The LinuxCNC issue tracker is to report bugs in the software.
If you have a question about how to use the software, use one of the other methods detailed on our community support page: http://linuxcnc.org/community/

## Here are the steps I follow to reproduce the issue:

 1. Note that all my machines incorporate machine power by the F2 key,
 2. launch linuxcnc -l
 3. hit F1, to unlock, F2 to enable all machine power, wait 4 seconds for soft start of power
 4. When "pwr on" turns green, issue a home-all
 5. discover something on the table to interfere with homing. hit ESC, is ignored, hit f2 to kill machine power, works
 6. move or clear table motion blocker
 7. cycle machine power back on with another F2.
 8. Axis never homed is skipped, next axis in the ini sequence continues to home and homing continues till other 3 axis's are homed.
 9.
 10.

## This is what I expected to happen:
All homing operations canceled until next time the home_all button is clicked on. Preferably by the ESC key. But certainly by power cycling the machine with the F2 key.

## This is what happened instead:

## It worked properly before this:
(If the behavior changed after making a particular change in hardware or
software, describe the change you think is responsible.  E.g., "after upgrading
from LinuxCNC 2.7.3 to 2.7.4")

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
 * I am using this kernel version (shown by `uname -a`):
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card):

WOWElectronics, 2023-03-10:
These machines all run buildbot master, but this is a long noticed miss-operation. Years.
No editing of original post allowed, didn't see more questions below opening screen.
machine is uptodate (a/o yesterday) buster
Linux GO704 4.19.0-23-rt-amd64 #1 SMP PREEMPT RT Debian 4.19.269-1 (2022-12-20) x86_64 GNU/Linux
Axis gui. lots of pyvcp custom stuff.

Roguish000, 2023-03-31:
I've noticed this kind of problem for a very long time.
It is a bit disconcerting to have the homing routine start up again .....

WOWElectronics, 2023-03-31:
> I've noticed this kind of problem for a very long time.
> It is a bit disconcerting to have the homing routine start up again .....
not to mention that it could be dangerous. Those are 435 oz/in stepper 
driving that machines XY, and a 960 oz/in nema 34 driving z.

On all my machines, if F2 is off, ALL machine power has been removed. 
Only the controls are live. And on that machine, a 4 second time lag 
built into the power up in deference to a 20 amp breaker in that 
circuit, there is a 50 ohm, 200 watt resistor in series with the spindle 
psu. Too many microfarads in that psu. A straight turn on by SSR trips a 
30 amp breaker, yet its only around 3 amps running normally. Unless the 
spindle is working really hard.

That time lag in the resumption of homing moves gives me plenty of time 
to get a hand/arm in the way of a machine suddenly moving.  Scary.

Cheers, Gene Heskett.
-- 
"There are four boxes to be used in defense of liberty:
  soap, ballot, jury, and ammo. Please use in that order."
-Ed Howdershelt (Author, 1940)
If we desire respect for the law, we must first make the law respectable.
  - Louis D. Brandeis
Genes Web page <http://geneslinuxbox.net:6309/>

SebKuzminsky, 2023-04-23:
I just tried to reproduce this here (in Sim with Axis GUI) and could not see either problem:
1. While homing, hitting Esc stops all motion immediately.
2. While homing, hitting F2 (with or without hitting Esc first) stops the machine, then hitting F2 again to power it back on does not result in any motion.

This is with a recent 2.9 from the buildbot: 2.9.0~pre1.673.gffb4e4399.

@WOWElectronics , @Roguish000 , can you reproduce this in sim?

WOWElectronics, 2023-04-23:
On 4/23/23 18:31, Sebastian Kuzminsky wrote:
> I just tried to reproduce this here (in Sim with Axis GUI) and could not see either problem:
> 1. While homing, hitting Esc stops all motion immediately.
> 2. While homing, hitting F2 (with or without [...]
I don't think so Seb.

Cheers, Gene Heskett.
-- 
"There are four boxes to be used in defense of liberty:
  soap, ballot, jury, and ammo. Please use in that order."
-Ed Howdershelt (Author, 1940)
If we desire respect for the law, we must first make the law respectable.
  - Louis D. Brandeis
Genes Web page <http://geneslinuxbox.net:6309/>

SebKuzminsky, 2023-05-03:
Here's @WOWElectronics 's config (converted from tar to zip because that's what github wants...): [genes-go704.zip](https://github.com/LinuxCNC/linuxcnc/files/11390980/genes-go704.zip)

I glanced at it and did not notice anything unusual.  It's a 4 axis config (XYZA), with a home switch for each axis. The home switch inputs are debounced, which is not ideal but should not cause the kind of surprise motion that's being reported here.

SebKuzminsky, 2023-05-03:
@WOWElectronics and @Roguish000 , can you please try this:
1. Start LinuxCNC.
2. Begin homing, so that some joints are moving (Z and A in @WOWElectronics's case, I don't know about @Roguish000's homing config).
3. Interrupt homing by hitting F2 in Axis.

At this point, if you **were** to hit F2 again, the machine would surprisingly start moving, right?  That's how to reproduce the bug we're chasing?  Instead of doing that, run `halcmd show sig` in a terminal, and paste the complete output as a comment in this issue.

WOWElectronics, 2023-05-03:
here is that output if I can get copy/paste to work, its bulky:
gene@GO704:~$ halcmd show sig
Signals:
Type          Value  Name     (linked to)
bit           FALSE  A-enable
                         ==> hm2_5i25.0.gpio.019.out
                         ==> hm2_5i25.0.pwmgen.00.enable
                         <== joint.3.amp-enable-out
                         ==> pid.a.enable
bit           FALSE  A-home-switch
                         ==> debounce.0.3.in
                         <== hm2_5i25.0.gpio.031.in_not
float             0  A-idbin
                         ==> idb.0.in
                         <== pid.a.output
float             0  A-idbout
                         ==> hm2_5i25.0.pwmgen.00.value
                         <== idb.0.out
float     -33.55699  A-pos-cmd
                         <== joint.3.motor-pos-cmd
                         ==> pid.a.command
float     -33.55699  A-pos-fb
                         <== hm2_5i25.0.encoder.01.position
                         ==> joint.3.motor-pos-fb
                         ==> pid.a.feedback
bit            TRUE  A-pwr-up
                         <== nearA.out
                         ==> not-A-pwr.in
float             0  A-vel-cmd
                         <== joint.3.vel-cmd
                         ==> pid.a.command-deriv
float         43165  Fcounter-in
                         <== S32_float-cal.out
                         ==> mux2cal0.in0
                         ==> mux2cal1.in0
bit           FALSE  MDI-mode
                         <== halui.mode.is-mdi
s32           43165  Scounter-in
                         ==> S32_float-cal.in
                         <== hm2_5i25.0.encoder.01.rawcounts
bit           FALSE  X-home-switch
                         ==> debounce.0.2.in
                         <== hm2_5i25.0.7i76.0.0.input-00-not
bit           FALSE  Y-home-switch
                         ==> debounce.0.1.in
                         <== hm2_5i25.0.7i76.0.0.input-01-not
bit            TRUE  Z-home-switch
                         ==> debounce.0.0.in
                         <== hm2_5i25.0.7i76.0.0.input-02-not
bit           FALSE  a-index-enable
                         ==> pid.a.index-enable
bit           FALSE  a-is-homed
                         <== halui.joint.3.is-homed
float             0  adjusted_speed
                         ==> mux4cmd.in2
                         <== scale_cmd.out
bit           FALSE  align-startb
                         ==> halui.mdi-command-04
                         <== pyvcp.align_startb
bit           FALSE  align_xb
                         ==> halui.mdi-command-06
                         <== pyvcp.align_yb
bit           FALSE  align_yb
                         ==> halui.mdi-command-05
                         <== pyvcp.align_xb
bit           FALSE  aligned2x
                         <== motion.digital-out-09
                         ==> pyvcp.align2_xt
bit           FALSE  aligned2y
                         <== motion.digital-out-10
[... 387 lines cut ...]
float      2.057306  z-pos-fb
                         <== hm2_5i25.0.stepgen.02.position-fb
                         ==> joint.2.motor-pos-fb
                         ==> pid.z.feedback
float             0  z-vel-cmd
                         <== joint.2.vel-cmd
                         ==> pid.z.command-deriv
bit            TRUE  ztrack
                         <== hm2_5i25.0.gpio.031.in
                         ==> spndl_cntr.countup
float             1  ztrackF
                         <== S32_float-cntr.out
                         ==> comp_cal0.in1
                         ==> comp_cal1.in1
s32               1  ztrakS32
                         ==> S32_float-cntr.in
                         <== spndl_cntr.count

gene@GO704:~$ 

Which I'm amazed, it worked w/o having to jump thru firefox security hoops.
There is also a msg on the lcnc screen:
Cannot unhome while homing, joint 0 
Now for S&G, I'll shut down lcnc and restart it because I think I recall this surviving a full stop and restart, but it did not miss-behave again after re-enabling F2 to power up the machine.
Thanks guys.

SebKuzminsky, 2023-05-04:
I edited your comment to add "```" before and after the paste, to turn it into a code block (monospaced font, no formatting or line wrapping), so it's easier to read.

That all looks fine to me!  I don't see anything wrong there.

Ok, next experiment.  Edit the INI file and change `[EMC]DEBUG` from its current value of `0x00000000` to `0xffffffff`.  Then start linuxcnc from the terminal, start homing, interrupt homing with F2, and then re-enable the machine with F2.  The machine starts moving as soon as you re-enable it, right?  Stop the motion and quit linuxcnc.  Then copy and paste all the output from linuxcnc to this thread.

WOWElectronics, 2023-05-04:
On 5/3/23 23:40, Sebastian Kuzminsky wrote:
> I edited your comment to add "```" before and after the paste, to turn it into a code block (monospaced font, no formatting or line wrapping), so it's easier to read.
>
> That all looks fine to me!  I [...]
In the morning's not so wee hours.  Thanks Sebastian.

Cheers, Gene Heskett.
-- 
"There are four boxes to be used in defense of liberty:
  soap, ballot, jury, and ammo. Please use in that order."
-Ed Howdershelt (Author, 1940)
If we desire respect for the law, we must first make the law respectable.
  - Louis D. Brandeis
Genes Web page <http://geneslinuxbox.net:6309/>

WOWElectronics, 2023-05-04:
On 5/3/23 23:40, Sebastian Kuzminsky wrote:
> I edited your comment to add "```" before and after the paste, to turn it into a code block (monospaced font, no formatting or line wrapping), so it's easier to read.
>
> That all looks fine to me!  I [...]
Ok, here is the killer, with debug=0xffffffff, it doesn't miss-behave, I 
tried it twice.
here is the shell output from the second try:

***@***.***:~/linuxcnc/configs/GO704-5i25-7i76-camera$ linuxcnc -l
LINUXCNC - 2.10.0-pre0-1123-gfaed3092e
Machine configuration directory is 
'/home/gene/linuxcnc/configs/GO704-5i25-7i76-camera'
Machine configuration file is 'GO704-5i25-7i76-camera.ini'
Starting LinuxCNC...
(time=1683176212.643229,pid=26531): Registering server on TCP port 5005.
(time=1683176212.643426,pid=26531): running server for TCP port 5005 
(connection_socket = 3).
tooldata_read_entry():comment for toolno 6 truncated to 39 chars:
    <0.052 pcb component lead clearance dril>
tooldata_read_entry():comment for toolno 7 truncated to 39 chars:
    <placeholder pcb bigger component cleara>
linuxcnc TPMOD=tpmod HOMEMOD=homemod EMCMOT=motmod
Note: Using POSIX realtime
twopass:invoked with <> options
twopass:found ./GO704-5i25-7i76-camera.hal
twopass:pass0: loadusr -W hal_manualtoolchange
hm2: loading Mesa HostMot2 driver version 0.15
hm2_pci: loading Mesa AnyIO HostMot2 driver version 0.7
hm2_pci: discovered 5i25 at 0000:01:02.0
hm2/hm2_5i25.0: Low Level init 0.15
hm2/hm2_5i25.0: Smart Serial Firmware Version 43
Board hm2_5i25.0.7i76.0.0 Hardware Mode 0 = standard
Board hm2_5i25.0.7i76.0.0 Software Mode 0 = io_spin
Board hm2_5i25.0.7i76.0.0 Software Mode 1 = io_ana_spin
Board hm2_5i25.0.7i76.0.0 Software Mode 2 = io_enc_ana_spin_fv
hm2/hm2_5i25.0: 34 I/O Pins used:
hm2/hm2_5i25.0:     IO Pin 000 (P3-01): StepGen #0, pin Direction (Output)
hm2/hm2_5i25.0:     IO Pin 001 (P3-14): StepGen #0, pin Step (Output)
hm2/hm2_5i25.0:     IO Pin 002 (P3-02): StepGen #1, pin Direction (Output)
hm2/hm2_5i25.0:     IO Pin 003 (P3-15): StepGen #1, pin Step (Output)
hm2/hm2_5i25.0:     IO Pin 004 (P3-03): StepGen #2, pin Direction (Output)
hm2/hm2_5i25.0:     IO Pin 005 (P3-16): StepGen #2, pin Step (Output)
hm2/hm2_5i25.0:     IO Pin 006 (P3-04): IOPort
hm2/hm2_5i25.0:     IO Pin 007 (P3-17): IOPort
hm2/hm2_5i25.0:     IO Pin 008 (P3-05): PWMGen #0, pin Out1 (Dir or 
Down) (Output)
hm2/hm2_5i25.0:     IO Pin 009 (P3-06): PWMGen #0, pin Out0 (PWM or Up) 
(Output)
hm2/hm2_5i25.0:     IO Pin 010 (P3-07): Smart Serial Interface #0, pin 
tx0 (Output)
hm2/hm2_5i25.0:     IO Pin 011 (P3-08): Smart Serial Interface #0, pin 
rx0 (Input)
hm2/hm2_5i25.0:     IO Pin 012 (P3-09): IOPort
hm2/hm2_5i25.0:     IO Pin 013 (P3-10): IOPort
hm2/hm2_5i25.0:     IO Pin 014 (P3-11): Encoder #0, pin Index (Input)
hm2/hm2_5i25.0:     IO Pin 015 (P3-12): Encoder #0, pin B (Input)
hm2/hm2_5i25.0:     IO Pin 016 (P3-13): Encoder #0, pin A (Input)
hm2/hm2_5i25.0:     IO Pin 017 (P2-01): IOPort
hm2/hm2_5i25.0:     IO Pin 018 (P2-14): IOPort
hm2/hm2_5i25.0:     IO Pin 019 (P2-02): IOPort
hm2/hm2_5i25.0:     IO Pin 020 (P2-15): IOPort
hm2/hm2_5i25.0:     IO Pin 021 (P2-03): IOPort
hm2/hm2_5i25.0:     IO Pin 022 (P2-16): IOPort
hm2/hm2_5i25.0:     IO Pin 023 (P2-04): IOPort
hm2/hm2_5i25.0:     IO Pin 024 (P2-17): IOPort
hm2/hm2_5i25.0:     IO Pin 025 (P2-05): PWMGen #1, pin Out1 (Dir or 
Down) (Output)
hm2/hm2_5i25.0:     IO Pin 026 (P2-06): PWMGen #1, pin Out0 (PWM or Up) 
(Output)
hm2/hm2_5i25.0:     IO Pin 027 (P2-07): IOPort
[... 489 lines cut ...]
to jiggle 2 or 3 mm, so obviously this pid does not use the same 
internal math the old one used. Perhaps it needs further TLC?

Other than the hunting, the machine seems to run well. Its next job will 
be running a boring head, drilling the handle holes in 6 more vise 
screws. Probably at least 2 weeks on down the log as I will have to 
removed a wedge under the spindle motor mount and make a B axis jig to 
sharpen the boring bar.  That or jury rig the A axis on the go704 as 
they don't come in cemented form with a description that truethfully 
uses the word sharp.

I do note that the new at-pid does not seem to honor the debug, there is 
not any pid info in the above trace that I can recognize. So that needs 
addressed at some point.

Thanks Sebastian.

Cheers, Gene Heskett.
-- 
"There are four boxes to be used in defense of liberty:
  soap, ballot, jury, and ammo. Please use in that order."
-Ed Howdershelt (Author, 1940)
If we desire respect for the law, we must first make the law respectable.
  - Louis D. Brandeis
Genes Web page <http://geneslinuxbox.net:6309/>

petterreinholdtsen, 2023-05-04:
[WOWElectronics]
> This is after figuring out how to run the new at-pid which returns crazy 
> settings as it needs an effort setting in the millions to get the axis 
> to jiggle 2 or 3 mm, so obviously this pid does not use the same 
> internal math the old one used. Perhaps it needs further TLC?

Are you running with autotuning enabled?

The pid controller code path when autotuning is not enabled was not
changed when autotuning was added to the component, as can be seen in
commit a882b0bc9bf4d42129b6ef79379518f8dd3973f9.  If you want to test
without autotuning enabled, you can build your own component with
'#define AUTO_TUNER 1' changed to '#define AUTO_TUNER 0'.  If it make a
difference, I suspect there is a race condition somewhere affected by
changed code execution times, not changed calculations.

> I do note that the new at-pid does not seem to honor the debug, there is 
> not any pid info in the above trace that I can recognize. So that needs 
> addressed at some point.

When you say new 'at-pid', are you talking about the pid component or
the at-pid component?  The autotuning change to the pid component did
not add or remove any logging in the code.

-- 
Happy hacking
Petter Reinholdtsen

WOWElectronics, 2023-05-04:
On 5/4/23 02:02, petterreinholdtsen wrote:
> [WOWElectronics]
>
> Are you running with autotuning enabled?
I've ben running master since forever, playing the canary in the coal 
mine by keeping master up to date several times a week. I also have the 
buildbot stuff installed on 2 of my 4 machines, including the rpi4b 
running my 80 yo Sheldon 11x54.

This pid change came totally unannounced. How do I determine if its 
enabled or not? My first clue was the machine acting up, and when I 
looked at man pid, I got the man page for at-pid.

2 Email's to you was not replied to> The pid controller code path when 
autotuning is not enabled was not
> changed when autotuning was added to the component, as can be seen in
> commit a882b0bc9bf4d42129b6ef79379518f8dd3973f9.  If you want to test
> without autotuning enabled, you can build your own [...]

I am talking about the pid component now in master, which is loaded with 
the std loadrt pid "bunch of names" hal command. The only change to my 
hal files when this took place was the requirement to rename the path 
from the pid_xyza I had been using for many years, to pid.xyza style.

I do this 100% as a hobby Petter. I am a bit unusual as I have only an 
8th grade education, but I've been interested if things physical since I 
was about 6 yo. I was reading high school physics books at the same time 
I was bore out of my skull with McGuffys readers in a little1 room 
school when Pearl Harbor was bombed by the Japanese. Took the IIowa 
version of the SB IQ thing in the 7nth grade, scored 147. Wanting to get 
the military out of the way when I was old enough by having my draft 
number moved up, because in those years, drafted was for 2 years, but 
volunteers was a 4 year commit.  But I made a mistake when I took the 
AFQT test, I scored a 98, next best in 130 some boys that day was 36, so 
I was instantly 4F'd, and screwed out of any chance at the GI Bill to 
further my education. So I fixed electronic stuff for a living, wound up 
in RCSD when the FCC brought their testing to town, so I got myself a 
1st phone w/o opening a book, and worked for KOTA- until about 2 years 
after my first wife had a stroke and died. So I picked up the 
glasshopper from a local bar but she pissed off Helen Duhamel, screwing 
that up, Moved to Nebraska as the tx superviser at KXNE where I learned 
the care and feeding of klystron amplifiers for 7 years, rather handy 
when they can be destroyed by a missadjusted magnet in 5 milliseconds, 
and cost around $125,000 a copy. In '72 I saw an announcement in the 
Norfolk paper that the community college there was having a Certified 
Electronics Test session, so I walked in cold with a $20 bill for the 4 
hour test and turned it back into the prof in 45 minutes.  multiple 
choice, 125 questions that got pretty deep in theory, he laid the answer 
stencil on my copy and saw a sea of black, 123 were correct. Up to thsat 
point, none of his students had passed, I was the first. So I rattled 
around in broadcasting until the Chiefs chair at WDTV was empty in 1984, 
found my niche and stayed till I retired in 2002. My glasshopper had her 
own agenda an left in '85, I married an old maid school teacher from 
here in '89, and she died of COPD 2+ years ago. Place is paid for, I'll 
be here till I leave in a box. Or urn. 88 now and also playing with 3d 
printers. I know enough assembly, basic09, c89 to be dangerous, but have 
never really understood python. My mother was the only girl in the 1929 
class on aviation tech at Des Moines Tech high school, so I think I got 
some of my native smarts from her. Doing my own machine rebuilds to use 
linuxcnc has occupied my spare time for about 25 years, building my 
first linux box in '98. Never looked back.  Ask me anything, doesn't 
have to be linuxcnc. Like the guy cast in bronze on a park bench in 
Hannibal Mo, I've never let my schooling get it the way of my education.
Cheers, Gene Heskett.
-- 
"There are four boxes to be used in defense of liberty:
  soap, ballot, jury, and ammo. Please use in that order."
-Ed Howdershelt (Author, 1940)
If we desire respect for the law, we must first make the law respectable.
  - Louis D. Brandeis
Genes Web page <http://geneslinuxbox.net:6309/>

petterreinholdtsen, 2023-05-04:
Very happy to have such skilled and capable LinuxCNC tester
available. :)

[WOWElectronics]
> How do I determine if its enabled or not?

It is enabled by setting the pid pin tune-mode to true, and activated by
setting pid pin tune-start to true.  If tune-mode is false, that code
path is unused.  In the component code is is the 'state' variable which
is STATE_PID in the normal case and other values in autotuning is in
progress.

Because of this, I suspect the autotuning code is not related, unless
there is some strange race condition involved, but lets go where the
evidence lead us. :)

I hope Sebastian is able to get anything out of the debug code.  If not,
we can perhaps add more debug output. :)

-- 
Happy hacking
Petter Reinholdtsen

WOWElectronics, 2023-05-04:
On 5/4/23 03:39, petterreinholdtsen wrote:
> 
> Very happy to have such skilled and capable LinuxCNC tester
> available. :)
> 
Thanks for the flowers. And I should ad a comment about the handle used, 
WOWElectronics, ask to see my bare feet because they'd been told I could 
walk on water and wanted to see if my feet were webbed. They aren't but 
the WalkOnWater turned into WOWElectronics when somebody wanted a 
business name. Pulling my leg, so I return the favor. ;o)> where the ">" 
is the beard on a smiley.[WOWElectronics]
> It is enabled by setting the pid pin tune-mode to true, and activated by
> setting pid pin tune-start to true.  If tune-mode is false, that code
> path is unused.  In the component code is is the [...]
your setup code didn't work, no amount of ferror would allow it to run.

that is copy/pasted here:
#!/usr/bin/env python3
#
# Copyright 2022 Petter Reinholdtsen
#
# This program is free software; you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation; either version 2 of the License, or
# (at your option) any later version.
#
# This program is distributed in the hope that it will be useful, but
# WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
# General Public License for more details.
#
# You should have received a copy of the GNU General Public License
# along with this program; if not, write to the Free Software
# Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA
# 02110-1301 USA.

"""

LinuxCNC automated PID auto-tuner
=================================

Automatically conduct a automatic PID tuning using the pid
LinuxCNC component.

This script first home the axis, then move it to the middle of its range.
Also extend the f-error value used by the PID controller.  Next, disable
the PID controller (tune-mode=1), start the tuning process (tune-start=1)
and wait for it to finish before enabling the PID controller again.

This script assume the pid component is configured like this:

   loadrt pid names=pid.x,pid.y,pid.z

It also assume the axis ranges are correct to be able to position the axes
in their middle positions.
"""

import time

import linuxcnc
import linuxcnc_util
import hal

axis = 'x'
axisid = 0

# Some values that need to be replaced during tuning
replace = {
#    f"pid.{axis}.tune-cycles": "0x40",  # Increase number of back and 
forth movements (default 50)
[... 155 lines cut ...]

WHERE?

the output set to bias and the controller still be disabled.

If running from LinuxCNC, the FERROR setting for the axis being tuned 
may need to be loosened up, as it must be larger than the limit cycle 
 amplitude in order to avoid a following error.

To perform auto tuning, take the following steps:
Move the axis to be tuned somewhere near the center of it's travel.  
Set tune-cycles  (the default value should be fine in most cases) 
and setting tune-mode to true.

Set tune-effort to a small value.
Set enable to true.
Set tune-mode to true.
Set tune-start to true. If no oscillation occurs, or the  oscillation is
too small, slowly increase tune-effort.
Set tune-start to true. 
If  no oscillation  occurs,  or the  oscillation is too small, slowly 
increase tune-effort 

Auto tuning  can be aborted at any time by setting enable or tune-mode 
to false.

WOWElectronics, 2023-05-04:
On 5/4/23 03:39, petterreinholdtsen wrote:
> Very happy to have such skilled and capable LinuxCNC tester
> available. :)
>
> [WOWElectronics]
>
> It is enabled by setting the pid pin tune-mode to true, and activated by
> setting pid pin tune-start to [...]
Unfortunately for that path of investigation, enabling debug fixes it. I 
can't get it to miss-behave with debugging fully enabled. Or is it 
sensitive to the fact I'm doing a remote login to do the testing from 
this keyboard, not the machines own xfce4 terminal front end for bash.
I've turned that debug back off, so I'll test again since I know the 
machine is cleared to do a home all. And from the remote login, it does 
not miss-behave. It just hunts. A in particular, that is a 100% home 
made servo. Piece of junk motor, very high ratio worm, driving a chinese 
clone of a bs-1's worm I cobbled up motor mounts for.  So now I'll go to 
its own keyboard and test again, BRB.

I went to the garage, the 10 feet from the house back door to the garage 
back door was chilly in my fruits and house slippers at 05:30 & 44F out 
and I cannot make it miss-behave now, tried everything but a reboot 
restarting linuxcnc, starting and interrupting a home all with an F2 
several times. I did find one item though, I need to set the A axis 
VOLATILE_HOME = 1, F2 is not clearing its * for homed in the dro. Now 
fixed. But I have really serious doubts that that is connected to this.

What do we do now? Wait for the other shoe to drop? And go cut the P 
values in half to see if that stops the hunting. A in particular is 
rattling the gears.

Intermittents are the hardest things to find, particularly when the 
fault is an idiot designer who left a cmos nand gate he was using for an 
invertor and left the 2nd gate floating despite the warning in every 
piece of cmos literature ever printed. Took me about 5 years to find it 
in a chyron station ID gfx machine. It was even wrong in the schematic 
which is how I finally found it.

Cheers, Gene Heskett.
-- 
"There are four boxes to be used in defense of liberty:
  soap, ballot, jury, and ammo. Please use in that order."
-Ed Howdershelt (Author, 1940)
If we desire respect for the law, we must first make the law respectable.
  - Louis D. Brandeis
Genes Web page <http://geneslinuxbox.net:6309/>

petterreinholdtsen, 2023-05-04:
[WOWElectronics]
> Those instructions as reformattted are attached. Are they now wrong?

As far as I know, these are as accurate as we got them.  I am sure they
can be improved, and that there are newer and perhaps even better
algorithms to use for autotuning.  I dream of someone knowledgable to
improve the PID autotuning algoritm in LinuxCNC.

> Note that setting tune-mode to true disable the control loop.  When 
> auto tuning completes, the tuning parameters will be set,
>
> WHERE?

It will be set in the relevant HAL pins.  Perhaps that should be made
more clear in the instructions?  The user need to copy the values to the
on disk configuration for the settings to persist.  I always used the
tuning GUI and lifted the relevant PID values from there.

-- 
Happy hacking
Petter Reinholdtsen

WOWElectronics, 2023-05-04:
On 5/4/23 06:28, petterreinholdtsen wrote:
> [WOWElectronics]
>
> As far as I know, these are as accurate as we got them.  I am sure they
> can be improved, and that there are newer and perhaps even better
> algorithms to use for autotuning.  I dream [...]
Which is what I did.

Cheers, Gene Heskett.
-- 
"There are four boxes to be used in defense of liberty:
  soap, ballot, jury, and ammo. Please use in that order."
-Ed Howdershelt (Author, 1940)
If we desire respect for the law, we must first make the law respectable.
  - Louis D. Brandeis
Genes Web page <http://geneslinuxbox.net:6309/>

SebKuzminsky, 2023-05-04:
> As I mentioned, I've experience this for a long time. 2.7, 2.8, 2.9pre,.... I home axes one at a time, not with a home_all command.
> 
> Here's a simple scenario: Home X (it's long so it take a while) Home Y (a lot shorter). while both are still in motion, hit F1 or F2 or Estop (physical button) all motion stops as it should. hit F1 and/or F2 and or release Estop button, get ready for motion. Home either X or Y

@Roguish000, could you please try the three debugging steps that @WOWElectronics did above, and report your results here?

1. Post a .zip file of your complete config (github doesn't let you attach tar files...).
2. Start LinuxCNC, begin homing, and interrupt it with F2, so that if you re-enable the machine by hitting F2 again it'll make the surprise move.  Don't re-enable it, instead run `halcmd show sig` and post the output here.
3. Turn up debugging all the way (`[EMC]DEBUG = 0xffffffff` in the INI file), cause the surprise motion to happen, and post the full output here.

Thanks!

andypugh, 2023-07-24:
What is the status of this? Are we still waiting for @Roguish000  to reproduce the problem?

andypugh, 2023-10-04:
Should be fixed. Please re-open this if the problem remains.

closed 2023-10-04 by andypugh