LinuxCNC/linuxcnc #2394: Gmoccapy 3.4.2.1 - virtual keyboard

zz912, 2023-03-15:
I tested new version Gmoccapy (branche 2.9) and I found bug with virtual keyboard:
Tested with:
linuxcnc-2.9_2023-03-15/configs/sim/gmoccapy/gmoccapy.ini
[image]

zz912, 2023-03-15:
Another bug. When I click on "New file" button in terminal it types:
`{'name': 'temp.ngc', 'size': '59 bytes\n5 gcode lines', 'g0': '0.000 mm', 'g1': '0.000 mm', 'run': '0 Seconds', 'toollist': [], 'x': '228599999999999990153278795644557513325662450416567920472097684948267316844236787462200184552539815936.000 to -228599999999999990153278795644557513325662450416567920472097684948267316844236787462200184552539815936.000 = -457199999999999980306557591289115026651324900833135840944195369896534633688473574924400369105079631872.000 mm', 'x_zero_rxy': '228599999999999990153278795644557513325662450416567920472097684948267316844236787462200184552539815936.000 to -228599999999999990153278795644557513325662450416567920472097684948267316844236787462200184552539815936.000 = -457199999999999980306557591289115026651324900833135840944195369896534633688473574924400369105079631872.000 mm', 'y': '228599999999999990153278795644557513325662450416567920472097684948267316844236787462200184552539815936.000 to -228599999999999990153278795644557513325662450416567920472097684948267316844236787462200184552539815936.000 = -457199999999999980306557591289115026651324900833135840944195369896534633688473574924400369105079631872.000 mm', 'y_zero_rxy': '228599999999999990153278795644557513325662450416567920472097684948267316844236787462200184552539815936.000 to -228599999999999990153278795644557513325662450416567920472097684948267316844236787462200184552539815936.000 = -457199999999999980306557591289115026651324900833135840944195369896534633688473574924400369105079631872.000 mm', 'z': '228599999999999990153278795644557513325662450416567920472097684948267316844236787462200184552539815936.000 to -228599999999999990153278795644557513325662450416567920472097684948267316844236787462200184552539815936.000 = -457199999999999980306557591289115026651324900833135840944195369896534633688473574924400369105079631872.000 mm', 'z_zero_rxy': '228599999999999990153278795644557513325662450416567920472097684948267316844236787462200184552539815936.000 to -228599999999999990153278795644557513325662450416567920472097684948267316844236787462200184552539815936.000 = -457199999999999980306557591289115026651324900833135840944195369896534633688473574924400369105079631872.000 mm', 'machine_unit_sys': 'Metric', 'gcode_units': 'mm'}`

hansu, 2023-03-15:
1. Fixed
2. I noticed that too, but that went out of my focus. I quick investigation lead to that it appeared with #2313 when gremlin was fixed (at least for Bookworm). But probably the print command was introduced earlier, but didn't show up because gremlin didn't run (on Bookworm). I'll dig into that...

hansu, 2023-03-16:
2. was introduced in https://github.com/linuxcnc/linuxcnc/commit/97f8cb071c1a7976b9c27cadf9b99303aba19c1d. Now fixed as well.

closed 2023-03-16 by hansu