LinuxCNC/linuxcnc #2399: Fix man pages of axis, gscreen, gmoccapy, touchy or fix the GUIs

hansu, 2023-03-17:
The man pages say the following, but none of those are working with those arguments. I also wonder how that should work together with the `display = xxx` in the INI file.  So I think it's best to adapt the man pages.
```
SYNOPSIS
       axis -ini INIFILE
```
       
```
SYNOPSIS
       gscreen -ini INIFILE
```

```
SYNOPSIS
       gmoccapy -ini <INI file>
```

```
SYNOPSIS
       touchy -ini <INI file>
```

SebKuzminsky, 2023-04-23:
The `linuxcnc` launcher script runs the GUIs with the `-ini` arguments as shown in the manpages, e.g.:
```
$ ps auxw
...
seb        51400  7.7  0.1 226500 55764 pts/13   Sl+  16:05   0:03 /usr/bin/python3 /usr/bin/touchy -ini /usr/share/doc/linuxcnc/examples/sample-configs/sim/touchy/touchy.ini
...
```
This is not a bug.

closed 2023-04-23 by SebKuzminsky

hansu, 2023-04-24:
But you cannot type
   `gmoccapy -ini <INI file>`  amd expect that to work that I meant.