LinuxCNC/linuxcnc #2400: Audio player from #2350 introduces delay in Gmoccapy touch off dialog

hansu, 2023-03-17:
## Here are the steps I follow to reproduce the issue:

 1. Switch on and reference Gmoccapy
 2. Click on an axis letter

## This is what I expected to happen:

Before #2350 was merged, the touch off dialog popped up immediately. Now there is a delay of about 1-2 seconds.
I wonder if it waits there for sound to finish playing? I don't know,  I don't have sound on my VM. But this delay is a bit annoying.

phillc54, 2023-03-17:
Yep, that seems to be the reason.
To prevent the need to create another module as a dependency we could do something like this in the audio test:
```
    _AUDIO_AVAILABLE = False
    try:
        if subprocess.run(["lspci | grep -i audio"], shell=True, capture_output=True).stdout.decode():
            from gmoccapy import player        # a class to handle sounds
            _AUDIO_AVAILABLE = True
    except:
        pass
```
 If users have sound then they would still get the delay.
Another way would be to add an option in the settings page to turn sounds off. 
Maybe even redo sounds so they run in a separate thread or process.

hansu, 2023-03-18:
Ok, but I wonder why:
- There is sound played for the calculator. This is no warning dialog, so there should be no sound.
- The delay seems only to happen in this case. Even for the calculator there is no delay if you click on the settings button (in case you have an unlock code defined). And in Settings -> Appearance -> Show test message, the test message is displayed immediately with sound.

To add an option to disable sound would be a good one independent of this behaviour here.

gmoccapy, 2023-03-19:
When I got the first time in contact with sound and at that time with audio player, I noticed that too. There was a solution at that time with an command line option telling audio player not to wait to finish the sound an give back handling immidiately.

The delay would be longer as longer the sound is, so selecting a long sound for the calculator could lead also there to a delay. Might be worse to test.

Norbert

hansu, 2023-03-19:
Yeah but there is no delay when the calculator is opened to enter the unlock code. There the sound seems to be played in parallel. But I also think the sound makes only sense for notifying dialogs, not for the calculator.

gmoccapy, 2023-10-08:
The entry dialog that opens if you click on CombiDRO Axis letter is defined in dialogs.py and does not send any signal to play sound. So I do not know why you have sound on that.
All test I was able to do, do not lead to a noticeable delay on my PC.

So I close this issue, but feel free to reopen it, if the problem still persist.

closed 2023-10-08 by gmoccapy

hansu, 2023-10-08:
I added a setting to disable sound in general (c7bcd7930282b9f7d0613641061a0be7c07326da) and I also disabled the sound when opening the calculator.
This was more a workaround but I could have closed this issue, sorry.