LinuxCNC/linuxcnc #2402: LinuxCNC crashes when it parses numeric PyVCP widgets of the “x” format.

issuefiler, 2023-03-20:
# Bug

 When it processes a `<number>`, `<s32>`, or `<u32>` PyVCP widget whose format is `x`, it throws an error and crashes.

[image]

## The LinuxCNC version

The latest version. LinuxCNC 2.10.0~pre0.

## The inputs

 Let LinuxCNC open a PyVCP panel layout XML file that contains such numeric widgets.

```
    <hbox>
	<u32> <halpin>"code0"</halpin> <format>"02x"</format> </u32>
	<u32> <halpin>"code1"</halpin> <format>"02x"</format> </u32>
	<u32> <halpin>"code2"</halpin> <format>"02x"</format> </u32>
	<u32> <halpin>"code3"</halpin> <format>"02x"</format> </u32>
	<u32> <halpin>"code4"</halpin> <format>"02x"</format> </u32>
	<u32> <halpin>"code5"</halpin> <format>"02x"</format> </u32>
    </hbox>
```

https://github.com/LinuxCNC/linuxcnc/blob/fc3bfe9d2c3cb7b380e2addf19b2b3d50c15986e/configs/by_machine/smithy/8i20.xml#L120-L123

## The reason

https://github.com/LinuxCNC/linuxcnc/blob/fc3bfe9d2c3cb7b380e2addf19b2b3d50c15986e/lib/python/pyvcp_widgets.py#L1065-L1067
https://github.com/LinuxCNC/linuxcnc/blob/fc3bfe9d2c3cb7b380e2addf19b2b3d50c15986e/lib/python/pyvcp_widgets.py#L1089-L1091
https://github.com/LinuxCNC/linuxcnc/blob/fc3bfe9d2c3cb7b380e2addf19b2b3d50c15986e/lib/python/pyvcp_widgets.py#L1113-L1115

```python
 self.value=0.0 
 dummy = "%(b)"+self.format 
 self.v.set( str( dummy  % {'b':self.value} ) ) 
```

 When `lib/python/pyvcp_widgets.py`’s `pyvcp_number`, `pyvcp_s32`, or `pyvcp_u32` instance is instantiated, it tries to `format` `0.0`, that is, `'%(b)FORMAT' % {'b': 0.0}`. If the format is “`x`” (hexadecimal), it does `'%(b)x' % {'b': 0.0}`, or simply put, `'%x' % 0.0`.

 While Python scripts in the previous versions of LinuxCNC are in Python 2, those in this version of LinuxCNC are in Python 3. Unlike Python 2, Python 3 forbids formatting a floating-point value in the `x` (hexadecimal) format, and attempts to do so raise a `TypeError: %x format: an integer is required, not float`.

ALatSMT, 2023-05-10:
Yes pyvcp_number uses a float and pyvcp_u32 & s32 are int, python3 also needs a int to format a hex

andypugh, 2023-09-30:
Sorry this took so long to merge.

closed 2023-09-30 by andypugh