LinuxCNC/linuxcnc #2411: The LinuxCNC configuration selector does not support Unicode README files.

issuefiler, 2023-03-29:
## The version

The latest. LinuxCNC 2.10.0~pre0.

## The system environment

```
$ locale 
LANG=C
LANGUAGE=
LC_CTYPE="C"
LC_NUMERIC="C"
LC_TIME="C"
LC_COLLATE="C"
LC_MONETARY="C"
LC_MESSAGES="C"
LC_PAPER="C"
LC_NAME="C"
LC_ADDRESS="C"
LC_TELEPHONE="C"
LC_MEASUREMENT="C"
LC_IDENTIFICATION="C"
LC_ALL=
```

# The issue

 The LinuxCNC configuration selector does not support Unicode README files.

[image]

“`’`” appears as “`â`.”

SebKuzminsky, 2023-04-23:
This does look annoying :-(
PR gratefully solicited...

havardAasen, 2024-03-27:
The application `pickconfig.tcl`  supports UTF-8, it does this by reading in the `locale`setting. The problem is that you set the locale to `C` which limits its ability to correctly display and manage all characters. You can't expect it to display UTF-8 characters if you tell it to display a different character set.

luzpaz, 2024-09-24:
Is there a workaround ?

andypugh, 2024-10-02:
Interestingly Unicode support is one of the few new things in the new version of Tcl.
(And a new version of Tcl is a rare event: https://www.theregister.com/2024/10/02/tcltk_version_9/?utm_medium=share&utm_content=article&utm_source=linkedin )

andypugh, 2024-10-02:
The workaround is to use a real locale, rather than the basic "C" locale, which only supports 8-bit standard ASCII. 
I don't know if there is any way to work around this inside the LinuxCNC code.

hansu, 2026-05-07:
We can add `encoding system utf-8` to  pickconfig.tcl but that forces this script to use utf8 encoding regardless of the locale setting.
So I would rather keep it as it is because I see no reason why not to use `C.utf8`.