LinuxCNC/linuxcnc #2414: Jog velocity with keyboard not continuous

dberndt391, 2023-03-30:
## Here are the steps I follow to reproduce the issue:

 1. Start linuxcnc 2.9pre-1 sim, power on.
 2. halscope joint.x.vel-cmd
 3. jog with arrow keys

## This is what I expected to happen:
Expect a constant jog velocity

## This is what happened instead:
Velocity drops sometimes. Velocity drop is roughly centered around the key repeat occurrence.

## It worked properly before this:
It worked on 2.8 on a entirely different computer.  I have no baseline with this hardware. It seems 2.8 can't be gone back to without major python2 related work.

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Linux Mint 21
 * I am using this kernel version (shown by `uname -a`): 6.1.0-5-rt-amd64
 * I am running ...
   * [x] A binary I built myself
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): 2.9~pre1
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): Axis
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card):  Simulation

## More details:

Issue does not happen on every key-repeat, but happens often. See attached image.
[image]

I know this looks similar to #434 However I'm fairly sure it's not a hardware/usb/keyboard issue. xev shows the key press up and down signals happening on the same ms every time. 

I instrumented the jog_on and jog_off functions in axis with some debug statements. Here is what pressing the jog keys with the machine off looks like:

*note times are deltas from the last jog debug event, and apparently they're *-1 as well as I wrote bad debug math apparently. 
<details><summary>Details</summary>
<p>
---f: jog_on start,                                 -4.98560
---f: jog_on stop-not manual_ok,                    -0.00004
^^^f: jog_off start,                                -0.52913
^^^f: jog_off stop-jog_after[a],                    -0.00003
---f: jog_on start,                                 -0.00115
---f: jog_on stop-not manual_ok,                    -0.00003
^^^f: jog_off start,                                -1.99852
^^^f: jog_off stop-jog_after[a],                    -0.00003
---f: jog_on start,                                 -0.00025
---f: jog_on stop-not manual_ok,                    -0.00002
^^^f: jog_off start,                                -1.99801
^^^f: jog_off stop-jog_after[a],                    -0.00003
---f: jog_on start,                                 -0.00033
---f: jog_on stop-not manual_ok,                    -0.00002
^^^f: jog_off start,                                -1.99839
^^^f: jog_off stop-jog_after[a],                    -0.00003
---f: jog_on start,                                 -0.00040
---f: jog_on stop-not manual_ok,                    -0.00002
^^^f: jog_off start,                                -1.99876
^^^f: jog_off stop-jog_after[a],                    -0.00003
---f: jog_on start,                                 -0.00039
---f: jog_on stop-not manual_ok,                    -0.00002
^^^f: jog_off start,                                -1.21582
^^^f: jog_off stop-jog_after[a],                    -0.00003
</p>
</details> 
As per above, the time delta between jog_off function starting and the jog_on function starting is pretty small. I'm not really sure what the requirements are but it seems the keyboard are certainly arriving for processing at the same millisecond.

Here is a sample of the same debug output but with the machine on and exhibiting the symptom. As per above, this is just 1 key press and hold, for a few seconds. I did speed the key repeat up some from above to try to capture a combined view with halscope, but creating a compelling visual explanation hasn't happened yet.
--f: jog_on start,                                 -4.29440
JOG GLITCH WILL OCCUR  -- not really true, this debug statement does mark the glitch every time after the first call though.
<details><summary>Details</summary>
<p>
---f: jog_on step jog calling,                      -0.00013
***f: jog start,                                    -0.00001
***f: jog end,                                      -0.02018
---f: jog_on step jog called,                       -0.00004
---f: jog_on stop-end,                              -0.00013
^^^f: jog_off start,                                -0.50813
^^^f: jog_off stop-end,                             -0.00011
---f: jog_on start,                                 -0.00033
---f: jog_on stop-jog_after[a],                     -0.00028
^^^f: jog_off start,                                -0.53068
^^^f: jog_off stop-end,                             -0.00006
***f: jog start,                                    -0.00009
***f: jog end,                                      -0.02021
---f: jog_on start,                                 -0.00045
JOG GLITCH WILL OCCUR
---f: jog_on step jog calling,                      -0.00011
***f: jog start,                                    -0.00001
***f: jog end,                                      -0.02017
---f: jog_on step jog called,                       -0.00003
---f: jog_on stop-end,                              -0.00010
^^^f: jog_off start,                                -0.48692
^^^f: jog_off stop-end,                             -0.00005
***f: jog start,                                    -0.00010
***f: jog end,                                      -0.02019
---f: jog_on start,                                 -0.00038
JOG GLITCH WILL OCCUR
---f: jog_on step jog calling,                      -0.00015
***f: jog start,                                    -0.00001
***f: jog end,                                      -0.02018
---f: jog_on step jog called,                       -0.00004
---f: jog_on stop-end,                              -0.00011
^^^f: jog_off start,                                -0.48975
^^^f: jog_off stop-end,                             -0.00005
---f: jog_on start,                                 -0.00011
---f: jog_on stop-jog_after[a],                     -0.00006
^^^f: jog_off start,                                -0.52906
^^^f: jog_off stop-end,                             -0.00006
---f: jog_on start,                                 -0.00026
---f: jog_on stop-jog_after[a],                     -0.00006
^^^f: jog_off start,                                -0.52983
^^^f: jog_off stop-end,                             -0.00006
---f: jog_on start,                                 -0.00027
---f: jog_on stop-jog_after[a],                     -0.00007
^^^f: jog_off start,                                -0.52961
^^^f: jog_off stop-end,                             -0.00005
---f: jog_on start,                                 -0.00011
---f: jog_on stop-jog_after[a],                     -0.00006
^^^f: jog_off start,                                -0.52984
^^^f: jog_off stop-end,                             -0.00005
---f: jog_on start,                                 -0.00033
---f: jog_on stop-jog_after[a],                     -0.00006
^^^f: jog_off start,                                -0.21854
^^^f: jog_off stop-end,                             -0.00005
***f: jog start,                                    -0.00008
***f: jog end,                                      -0.02020

</p>
</details> 

Excerpt from axis.py:
```python
def jog_on(a, b):
    if not manual_ok(): return
    if not manual_tab_visible(): return
    if a < 3 or a > 5:
        if vars.metric.get(): b = b / 25.4
        b = from_internal_linear_unit(b)
    if jog_after[a]:
        root_window.after_cancel(jog_after[a])
        jog_after[a] = None
        return
    jogincr = widgets.jogincr.get()
    jjogmode = get_jog_mode()
    if jogincr != _("Continuous"):
        s.poll()
        if s.state != 1: return
        distance = parse_increment(jogincr)
        jog(linuxcnc.JOG_INCREMENT, jjogmode, a, b, distance)
        jog_cont[a] = False
    else:
----------------------------------------------this is the branch of interest------------------
        global continuous_jog_in_progress,cjogindices
        continuous_jog_in_progress = True
        if not a in cjogindices: cjogindices.append(a)
        jog(linuxcnc.JOG_CONTINUOUS, jjogmode, a, b)
        jog_cont[a] = True
        jogging[a] = b
    activate_ja_widget(a)
``` 

Any time we end up in the branch of interest marked above the problem occurs. If jog_on runs and doesn't execute the last else case jogging remains smooth. I suspect this code is more of a sign of the issue rather than the cause, but I don't have a very good understanding of the code base.

samcoinc, 2023-03-30:
I think this has to do with the key repeat in the os.. I have played with
this a bit a while ago - but a quick search shows

https://forum.linuxcnc.org/38-general-linuxcnc-questions/46373-keyboard-jogging-is-there-a-fix-for-stuttering

so look for auto-repeat or keyboard repeat settings in the os..

dberndt391, 2023-03-30:
I have played with the keyboard control. I can turn repeat off and that resolves the issue, but it also makes using the computer feel weird when you can't do things like hold a key to scroll or insert multiple characters.

I proved out in xev that the events are closely tied time wise (key release and key press happen in the same ms). And I show above that if the machine is off the jog events happen in quick succession. So I do not believe this to actually be os/keyboard related.  

Setting the repeat rate faster makes the problem worse in that there are more chances for it to happen. Reducing it to the minimum rate of 2 seconds decreases the issue but there is still a good chance for it to happen at the 2 second event.

andypugh, 2023-03-31:
Are you asking for linuxCNC to ignore key release events? That does sound do-able, but I am not sure I would want to be the one responsible for that code. If it goes wrong then there is much potential for damage, 
The difficulty is in LinuxCNC differentiating between real key release events and key-repeat events.

dberndt391, 2023-03-31:
After more investigating I see that the jog_off function is using after_idle to detect an empty queue and thus button release. Well that pinpoints the problem a bit more and lets me do more research. I guess my complaint would be that after_idle is firing too quickly or some variation of that theme. I'll investigate further into how the tkinter event queue works.

dberndt391, 2023-03-31:
It looks like the TK event queue is a black box, and TCL underneath it is the same without getting really into the weeds.

I still don't know why the issue is actually happenning at a TCL/OS level but may continue to investigate if time and curiousity allow. But for now here's a proposed in axis fix. The idea is that if the jog ends due to a keyrelease but no keypress being in the event queue, then use after to wait a few more ms and re-check the queue. 

The downside is when you release a jog button it is effectively held down for the additional configured time period.

```python
def jog_off(a):
    if jog_after[a]: return
    jog_after[a] = root_window.after_idle(lambda: jog_off_actual_delay(a))

def jog_off_actual_after_delay_stage_two(a):
    # delay complete, put the after_idle check back in queue
    jog_after[a] = root_window.after_idle(lambda: jog_off_actual(a))

def jog_off_actual_delay(a):
    """
    If axis was jogging and a key_release occurs and there is no matching
    keypress event to keep the jog going then queue a delay using tk.after, and
    then that function will requeue to after_idle.
    This gives a chance for any missing keypress event to arrive in the
    keyrepeat_grace_time to be interpreted as a continuous jog.
    It add keyrepeat_grace_time to the length of any jogs
    """

    keyrepeat_grace_time = 10 # milliseconds
    jog_after[a] = root_window.after(keyrepeat_grace_time, jog_off_actual_after_delay_stage_two, a)

def jog_off_actual(a):
    print('afteridle firing')
    global continuous_jog_in_progress
    continuous_jog_in_progress = False
    if not manual_ok(): return
    jog_after[a] = None
    jogging[a] = 0
    jjogmode = get_jog_mode()
    if jog_cont[a]:
        jog(linuxcnc.JOG_STOP, jjogmode, a)
```

phillc54, 2023-03-31:
Personally I would like a jog to stop if the key was released rather than working around stuttering in jogging. The same issue happens in PyQt as well and we found the safest way to prevent stuttering was to inhibit key repeat but the user has the [option to disable](http://linuxcnc.org/docs/devel/html/plasma/qtplasmac.html#_keyboard_jogging) the inhibit.

I think the reason is that python has no real access to the keyboard. It can be done with the [keyboard module](https://github.com/boppreh/keyboard) (unmaintained)  but this requires root priviliges.

SebKuzminsky, 2023-04-23:
Key repeat is useful in other parts of the computer's UI, so I prefer not to turn it off.

rmu75, 2023-05-17:
Tht Tcl wiki has an elaborate discussion how to disable/ignore auto-repeated keys here: https://wiki.tcl-lang.org/page/Disable+autorepeat+under+X11

andypugh, 2023-05-19:
Top marks for linking to a Tcl-relevant page on the subject, as that means that the ideas there ought to work.

andypugh, 2023-06-18:
@SebKuzminsky This issue is present on previous versions of LinuxCNC. Can you decide why we decided it was a 2.9 Must-fix? Would not fixoing this really prevent a 29 release?

SebKuzminsky, 2023-06-18:
No, this should not block 2.9

grotius-cnc, 2024-02-21:
This is what happening when you have debug in ini on : DEBUG = 0x00000010

```
Issuing EMC_JOG_CONT --          (  +124,+40,  +248,    +0,31.116667,    +1,)
Issuing EMC_JOG_STOP --          (  +136,+32,  +249,    +0,    +1,)
Issuing EMC_JOG_CONT --          (  +124,+40,  +250,    +0,31.116667,    +1,)
Issuing EMC_JOG_STOP --          (  +136,+32,  +251,    +0,    +1,)
Issuing EMC_JOG_CONT --          (  +124,+40,  +252,    +0,31.116667,    +1,)
Issuing EMC_JOG_STOP --          (  +136,+32,  +253,    +0,    +1,)
Issuing EMC_JOG_CONT --          (  +124,+40,  +254,    +0,31.116667,    +1,)
Issuing EMC_JOG_STOP --          (  +136,+32,  +255,    +0,    +1,)
```
In the jog planner, the tp_simple.c It's requesting a jog stop every cycle, for every axis.

Indeed it's caused by  a keyboard auto-repeat function in X11.
Changing the keyboard settings in the os reflexts the problem.

This is a tiny fix to make it work. It only involves the axis.py file. 
The keyboard autorepeat is turned off when axis is running, it's turned on again 
when axis is closing. It uses a system command "xset r on-off".

In the axis.py file

line 126:
```
os.system("xhost -SI:localuser:gdm -SI:localuser:root > /dev/null 2>&1")
os.system("xset r off")
```

line 157 : 
```
def General_Halt():
    text = _("Do you really want to close LinuxCNC?")
    if not root_window.tk.call("nf_dialog", ".error", _("Confirm Close"), text, "warning", 1, _("Yes"), _("No")):
    	os.system("xset r on")
        root_window.destroy()
```

closed 2024-03-10 by andypugh