LinuxCNC/linuxcnc #2416: OpenGL error when starting AXIS on X11

allistar, 2023-04-01:
# The issue tracker is not a support forum

## Here are the steps I follow to reproduce the issue:

 1. Build and run the latest linuxCNC on an X11 (xorg) machine
 2. Run LinuxCNC
 3.

## This is what I expected to happen:
The Axis GUI should start as expected.

## This is what happened instead:
This error is happening:

`Traceback (most recent call last):
  File "/home/allistar/Projects/linuxcnc-test/bin/axis", line 3964, in <module>
    get_coordinate_font(vars.dro_large_font.get())
  File "/home/allistar/Projects/linuxcnc-test/bin/axis", line 3842, in get_coordinate_font
    glnav.use_pango_font(coordinate_font, 0, 128)
  File "/home/allistar/Projects/linuxcnc-test/lib/python/glnav.py", line 65, in use_pango_font
    glBitmap(1, 0, 0, 0, 0, h-d, bytearray([0]*4))
  File "src/latebind.pyx", line 39, in OpenGL_accelerate.latebind.LateBind.__call__
  File "src/wrapper.pyx", line 318, in OpenGL_accelerate.wrapper.Wrapper.__call__
  File "src/wrapper.pyx", line 311, in OpenGL_accelerate.wrapper.Wrapper.__call__
  File "/usr/lib/python3.11/site-packages/OpenGL/platform/baseplatform.py", line 415, in __call__
    return self( *args, **named )`

## It worked properly before this:
(If the behavior changed after making a particular change in hardware or
software, describe the change you think is responsible.  E.g., "after upgrading
from LinuxCNC 2.7.3 to 2.7.4")

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Gentoo Linux 2.13
 * I am using this kernel version (shown by `uname -a`): Linux 6.1.19-rt-rt8 #3 SMP PREEMPT_RT
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [x] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): 2.10.0-pre0-1093-g7ba312e3d8
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): AXIS
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): Parallel port

I've attached a full error file of the issue.
I have tracked this issue down to [this](https://github.com/LinuxCNC/linuxcnc/commit/710ce7e39dcbc584615771947bf26658808ce8e3) commit. When I revert that commit the error no longer happens.

Does the second value passed in need to be determined based on whether we're running Wayland or X11?
[linuxcnc_error.log](https://github.com/LinuxCNC/linuxcnc/files/11128376/linuxcnc_error.log)

This OpenGL problem has been mentioned in a few other issues and PRs:
https://github.com/LinuxCNC/linuxcnc/pull/2259
https://github.com/LinuxCNC/linuxcnc/issues/1599

SebKuzminsky, 2023-04-23:
The current code works on Debian Buster through Bookworm.  I think there should be a better solution that works on all platforms, and i think the person who knows what that solution is, is not me.  I look forward to a PR from a cluefull OpenGL hacker.  Until then, I prefer to keep the current code, which works on the officially supported Debian platforms, sorry...

rene-dev, 2023-04-23:
try https://github.com/LinuxCNC/linuxcnc/pull/2446

rene-dev, 2023-09-26:
workaround by https://github.com/LinuxCNC/linuxcnc/commit/bde2cdc5051647e36435136adee2c13768b5f52b

closed 2023-09-26 by rene-dev