LinuxCNC/linuxcnc #2423: halscope: add OK button for "Select Channel Source"

hansu, 2023-04-06:
The improvements of the Select Channel Source dialog which allows to 
filter the list by just typing (which is really great, thanks @havardAasen) makes it necessary to double click the selected signal or press enter.  An OK button like this would be a great improvement:

[image]

But to connect the OK button is a bit tricky and I am not so familiar  with the Gtk Tree model. So maybe you @havardAasen can help out here as you have done the complete Gtk3 conversion?

havardAasen, 2023-04-07:
Yes, I completely agree, an OK button would be nice.

I looked at the code, and if I remember correct, it was getting the signal name, reliably, that I struggled with As long as we have access to the name of the element, of the selected row, I hope the rest will be more or less straight forward.

I can take a look and see if I can't find a better way to deal with this.

I assume we should have all three methods to accept a new signal, double click, enter and OK button?

And for something partly related.
Do you believe we need the informational message at the top of the dialog? It isn't present in Halmeter, and it would be nice if we could share more of the code between the applications. I can't understand why these two windows shouldn't be identical, perhaps aside from the title.

hansu, 2023-04-07:
>I can take a look and see if I can't find a better way to deal with this.

Thanks

>I assume we should have all three methods to accept a new signal, double click, enter and OK button?

Maybe we can drop the double-click when we have an ok button, but I would keep at least enter. But if it is not too much work, I would also keep the double click.

>Do you believe we need the informational message at the top of the dialog? 

No. Maybe we can add the channel number to the title (e.g. "Select Channel **1** Source"), but other than that this info text gives no information.

PR #2427 (2023-04-08): halscope: Minor changes for 'Select Channel Source' dialog
This pr consist of three changes to the 'Select Channel Source' dialog.

- Add 'OK' button in the Select Channel Source' dialog. Fixes: #2423 
- Add channel number in dialog title.
- Remove information label
closed 2023-04-08 by hansu