LinuxCNC/linuxcnc #2426: gremlin - black ghost

zz912, 2023-04-07:
Tested with:
linuxcnc-2.9/configs/sim/gmoccapy/gmoccapy.ini
[image]
I also tested it with gscreen, so the bug is in gremlin, not in Gmoccapy.

hansu, 2023-04-07:
Also visible in Axis, but not relevant there as the DRO cannot be hidden:

[image]

c-morley, 2023-04-07:
This is the overlay area.
It is coded in glcanon.py

There is an attribute: show_overlay that can turn it on and off from subclasses.
I would bet it got set to default on in one of the subclasses.

hansu, 2023-04-08:
Can we add something that the attribute show_overlay will be automatically set to false when no DRO or similar is shown?

c-morley, 2023-04-09:
You could for sure.
probably right here:
https://github.com/LinuxCNC/linuxcnc/blob/2.9/lib/python/gladevcp/hal_gremlin.py#L179

closed 2023-05-06 by rene-dev

SebKuzminsky, 2023-05-06:
Thanks @rene-dev !

commit 0b1be6e0 (2023-05-06): glcanon: only show overlay when DRO is enabled. closes #2426