LinuxCNC/linuxcnc #2431: mqtt-publisher user hal component is not sending keys

mrbubble62, 2023-04-11:
The mqtt-publisher is connecting to mqtt broker but I cannot get the mqtt-publisher component to send any key values.

pip install paho-mqtt

In HAL file:
loadusr -W mqtt-publisher --mqtt-broker=[MQTT]BROKER --mqtt-user=[MQTT]USERNAME --mqtt-password=[MQTT]PASSWORD keys=Xpos,Ypos,Zpos,halui.machine.is-on,halui.mode.is-auto,halui.mode.is-manual,halui.mode.is-mdi,halui.mode.is-teleop,halui.program.is-running

## This is what I expected to happen:
In MQTT Explorer under devices/linuxcnc/machine
{
  "mqtt-publisher.period": 10
  "Xpos": 1234
  "Ypos": 1234
  ...
}

## This is what happened instead:
MQTT Explorer devices/linuxcnc/machine
{
  "mqtt-publisher.period": 10
}

## Information about my hardware and software:

Machine: VBox Debian GNU/Linux 11 (bullseye)

Config: configs/sim/axis/axis_mm

LinuxCNCDEV 5.10.0-21-amd64 #1 SMP Debian 5.10.162-1 (2023-01-21) x86_64 GNU/Linux

Git version:
v2.9.0-pre1-673-gffb4e43998

mrbubble62, 2023-04-11:
Looking at the code there appears to be nothing more than mqtt-publisher.period being added to the array "data"

```
    def update_mqtt(self):

        "Run the endless loop fetching pin data and sending it to MQTT."

        running = True

        while hal.component_exists('halui') and running:

            data = {}

            if self.hal['enable']:

                data['mqtt-publisher.period'] = self.hal['period']

                if not self.dryrun:

                    print(f"info: Publishing MQTT message ({self.mqtt_prefix}):", json.dumps(data))

                    self.mqttc.publish(self.mqtt_prefix, json.dumps(data))

                else:

                    print("info: Not publishing MQTT message ({self.mqtt_prefix}):", json.dumps(data))

                self.hal['lastpublish'] = time.time()

            try:

                time.sleep(self.hal['period'])

            except KeyboardInterrupt as e:

                running = False
```

SebKuzminsky, 2023-04-23:
It's great you're working on a fix for this issue, a PR against 2.9 is welcome!

mrbubble62, 2023-04-24:
Sorry, I'm not sure how to do that. Do I need to make a new fork of 2.9?

hansu, 2023-04-24:
You have to
1. make a fork of the LinuxCNC repo (it you haven't done that already),
2. checkout the 2.9 branch,
3. create a new branch,
4. commit & push your changes into that branch and
5. create a pull request.

andypugh, 2023-04-24:
This tutorial seems to start at the beginning: https://www.freecodecamp.org/news/how-to-make-your-first-pull-request-on-github-3/

mrbubble62, 2023-05-04:
> It's great you're working on a fix for this issue, a PR against 2.9 is welcome!

Completed
Update mqtt-publisher.py #2449

commit 63d7f6ad (2023-05-04): Merge pull request #2449 from mrbubble62/mqtt-publisher

Update mqtt-publisher.py fixes #2431