LinuxCNC/linuxcnc #2435: Wrong axis direction with xyzbc-trt and maxkins kinematics

DerAndere1, 2023-04-12:
# The issue tracker is not a support forum

The LinuxCNC issue tracker is to report bugs in the software.
If you have a question about how to use the software, use one of the other methods detailed on our community support page: http://linuxcnc.org/community/

(delete this section before submitting your bug report)

## Here are the steps I follow to reproduce the issue:

 1. See source code at https://github.com/LinuxCNC/linuxcnc/blob/76b870db31a541a4c4e6b1e1deb02724b5c51961/src/emc/kinematics/maxkins.c#L63-L94 and https://github.com/LinuxCNC/linuxcnc/blob/76b870db31a541a4c4e6b1e1deb02724b5c51961/src/emc/kinematics/maxkins.c#L63-L94
 3. Compare with documentation at https://linuxcnc.org/docs/html/gcode/machining-center.html#_rotational_axes
 4. Observe that with maxkins and with xyzbc-trt kinematics, axes B and C move in wrong direction  

## This is what I expected to happen:

inverse kinematics should ensure that rotational axes move CCW relative to the work piece when viewed from positive. For a tilting/rotating table, this means the rotational joint of the table must move clockwise when viewed from positive. 

These are some expected joint positions calculated from given cartesian positions using fixed xyzbc-trt inverse kinematics:

|Cartesian position XYZBC| Joint position XYZBC|
|--- |--- |
|(1, 0, 2, 0, 0) | (1, 0, 2, 0, 0) |
|(1, 0, 2, 90, 0) | (2, 0, 1, 90, 0) |
|(1, 0, 2, 0, 90) | (0, -1, 2, 0, 90) |

## This is what happened instead:

Axes B and C move in wrong direction.

These are some results using current xyzbcKinematicsInverse() from trtfuncs.c:

|Cartesian position (X,Y,Z,B,C) | Joint position (X,Y,Z,B,C) |
|--- |--- |
|(1, 0, 2, 0, 0) | (1, 0, 2, 0, 0) |
|(1, 0, 2, 90, 0) | (-2, 0, 1, 90, 0) |
|(1, 0, 2, 0, 90) | (0, 1, 2, 0, 90) |

## It worked properly before this:

Never

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): None
 * I am using this kernel version (shown by `uname -a`): None
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): None
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): None

andypugh, 2023-04-12:
Whilst you may be technically correct, there are configurations out there, making parts, with this kinematics module. 
If it was to change, then bad things would happen to those machines. 

I think that changing this module is a bad idea because of the installed userbase, but would support the inclusion of an _alternative_ kins with the correct behaviour.

SebKuzminsky, 2023-04-23:
I agree we shouldn't change the default behavior of existing kins.  Either add new kins with new names that have the documented behavior, or add optional switches or something to the existing kins.

Some illustrative pictures that support @DerAndere1's observation are here: https://linuxcnc.org/docs/html/user/user-concepts.html#sec:Machine-Configurations

DerAndere1, 2025-06-21:
Videos of LinuxCNC with unusual axis directions:
maxkins (unusual B axis direction): https://www.youtube.com/watch?v=qiCD0MxI1_I
5axiskins (unusual B direction): https://www.youtube.com/watch?v=65Wd6ZGoWys&themeRefresh=1
xyzac-trt (unusual A and C direction); https://youtu.be/BevZm46cTQw?t=347

multiple sources showing conventional axis direction:
Siemens / FANUC:
- Video of TRT  machine using FANUC's G43.4 command, simulated using Siemens NX CAM software: https://www.youtube.com/watch?v=-JP0MAO7ySM
- Siemens AG, SINUMERIK, Manual, 5-axis machining, Edition 05/2009, DocOrderNo. 6FC5095-0AB10-0BP1, page 51

Haas Automation: 
- Video of an actual machine with a TRT: https://youtu.be/HxPjH4v5iEg?t=213
- Video of a simultion: https://www.youtube.com/watch?v=lazNvWOSYns
- Manual: https://www.haascnc.com/service/troubleshooting-and-how-to/how-to/umc-750---b-axis-and-c-axis-intersection---alignment.html

HAIDENHAIN:
Video tutorial: https://youtu.be/oMyzYUOJrxM?t=76

closed 2025-06-26 by andypugh