LinuxCNC/linuxcnc #2440: gladevcp icon symlinks

SebKuzminsky, 2023-04-14:
# The issue tracker is not a support forum

The LinuxCNC issue tracker is to report bugs in the software.
If you have a question about how to use the software, use one of the other methods detailed on our community support page: http://linuxcnc.org/community/

(delete this section before submitting your bug report)

## Here are the steps I follow to reproduce the issue:

 1.
 2.
 3.

## This is what I expected to happen:

## This is what happened instead:

## It worked properly before this:
(If the behavior changed after making a particular change in hardware or
software, describe the change you think is responsible.  E.g., "after upgrading
from LinuxCNC 2.7.3 to 2.7.4")

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
 * I am using this kernel version (shown by `uname -a`):
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card):

hansu, 2023-04-14:
>The first thing that comes to mind is to just usethe libgladeui-common icons directly

Yeah, just copying the icons from libgladeui-common to the LinuxCNC repo would have been the easiest solution. But it felt not ideal to have those duplicates in the repo, so I came up with the symlinks. But if lintian does not like them, the only solution I see here is to copy the icons directly.
I'll prepare a PR.

>Because lintian fails, we do not publish debs.

Regarding the debs I wonder about two things: 
- Publishing debs to debian is made manually, right? Because the version in debian unstable was fairly old when I updated the last time
- We are building debs for Bookworm and Bullseye, right? Why don't we have them on http://buildbot.linuxcnc.org/dists/ ?

SebKuzminsky, 2023-04-14:
> > The first thing that comes to mind is to just usethe libgladeui-common icons directly
> 
> Yeah, just copying the icons from libgladeui-common to the LinuxCNC repo would have been the easiest solution.

I didn't mean copy the icons into our repo, i meant use the icons installed by libgladeui-common.

>     * Publishing debs to debian is made manually, right? Because the version in debian unstable was fairly old when I updated the last time

Yes, the linuxcnc debs on debian.org are manually prepared as dsc (source packages), then uploaded and automatically built into deb (binary packages) by debian's CI infrastructure.  I believe those debs need to be lintian clean to be accepted, especially now that Bookworm is in Freeze. 
 
>     * We are building debs for Bookworm and Bullseye, right? Why don't we have them on http://buildbot.linuxcnc.org/dists/ ?

Because the buildbot is (still!) in transition from the old one (at http://buildbot.linuxcnc.org) that doesn't know how to deal with anything newer than Buster, to the new one (at http://buildbot2.highlab.com) that knows how to deal with all supported distros.

hansu, 2023-04-14:
>I didn't mean copy the icons into our repo, i meant use the icons installed by libgladeui-common.

As far as I know, the icon names are derived from the widget names. Let me investigate if we can assign different names.

> to the new one (at http://buildbot2.highlab.com/) that knows how to deal with all supported distros.

Great, I didn't know that the debs are available there for download

hansu, 2023-04-15:
Setting the icon name in the glade catalogue didn't work for all icons. Really weird.
Do you think copying symbolic links would also trigger the linter?

PS: Great to have also Bullseye and Bookworm on https://www.linuxcnc.org/dists/ :+1:

hansu, 2023-05-01:
@SebKuzminsky can be closed now, right?

PR #2441 (2023-04-27): Remove gladevcp symlink creation
Fixes #2440

This removes the links from the .links file to not create symlinks with `dh_link`.
For some of them it worked to set alternative icons in the glade catalogue file, but not for all.
So for the other ones I created symlinks which are just copied by the makefile. 
If you think this also triggers the linter, I can simply drop the last commit.
closed 2023-09-30 by andypugh