LinuxCNC/linuxcnc #2445: Axis get_max_jog_speed should check if axis in jog_order[]

dberndt391, 2023-04-21:
I believe axis should check jog_order before returning a value in the get_max_jog_speed function. Or any other fix that prevents the function from erroring when an axis isn't defined

## Here are the steps I follow to reproduce the issue:

 1. Create a machine config with only 1 Axis
 2. use shift + jog hot key to jog an access that doesn't exist (Shift + Up arrow for example, would jog an undefined Y axis)
 3. receive error something like:
 Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/home/niigata/lcnc-dev/bin/axis", line 3339, in <lambda>
    root_window.bind("<Shift-KeyPress-%s>" % b, lambda e: jog_on_map(axis_num, get_max_jog_speed_map(axis_num)))
  File "/home/niigata/lcnc-dev/bin/axis", line 1870, in get_max_jog_speed_map
    return get_max_jog_speed(a)
  File "/home/niigata/lcnc-dev/bin/axis", line 1862, in get_max_jog_speed
    if joint_type[a] == 'LINEAR':
IndexError: list index out of range

## This is what I expected to happen:
Jog keys for undefined axis should not generate an error

## This is what happened instead:
Jog keys for undefined axis do generate an error

## Information about my hardware and software:
Linuxm Mint 21
kernel 6.1.0-7rt
2.9-pre1 build from source

I looks to me like get_jog_speed_map is protected from this issue by the line:
if a >= len(jog_order): return

Perhaps the get_max_jog_speed function should have the same protective line? But I'm just guessing, I haven't dug too deeply into the issue.

SebKuzminsky, 2023-04-23:
Confirmed with sim, start sim/axis/lathe, home, then Shift-PgUp and Shift-PgDown emit the error.  PgUp and PgDown without Shift does not cause the problem.  The fix proposed by @dberndt391 looks right.

petterreinholdtsen, 2023-07-06:
Is this the change you are talking about?

```
diff --git a/src/emc/usr_intf/axis/scripts/axis.py b/src/emc/usr_intf/axis/scripts/axis.py
index c49c1feb30..d3cb2d7693 100755
--- a/src/emc/usr_intf/axis/scripts/axis.py
+++ b/src/emc/usr_intf/axis/scripts/axis.py
@@ -1848,7 +1848,9 @@ def get_jog_speed_map(a):
         a = "XYZABCUVW".index(axis_letter)
     return get_jog_speed(a)
 
+
 def get_max_jog_speed(a):
+    if a >= len(jog_order): return 0
     max_linear_speed = vars.max_speed.get()
     max_linear_speed = to_internal_linear_unit(max_linear_speed)
     if vars.metric.get(): max_linear_speed = max_linear_speed * 25.4
@@ -1863,6 +1865,8 @@ def get_max_jog_speed(a):
             return max_linear_speed
         else:
             return vars.max_aspeed.get()
+
+
 def get_max_jog_speed_map(a):
     if not get_jog_mode():
         axis_letter = jog_order[a]
```

closed 2023-09-30 by andypugh