LinuxCNC/linuxcnc #2448: Remap - random bug

zz912, 2023-04-24:
I have a bug on a real machine:
https://forum.linuxcnc.org/38-general-linuxcnc-questions/48870-atc-random-bug

The good news is that I am able to simulate this error on a virtual machine.

Simulate the problem:
1) Download and Buid branche 2.9
2) I followed this:
http://linuxcnc.org/docs/2.9/html/remap/remap.html#remap:upgrading-an-existing

cd /home/user/linuxcnc/linuxcnc-2.9/configs/sim/gmoccapy/python/
rm remap.py
rm toplevel.py 
rm stdglue.py
cp ../../../../nc_files/remap_lib/python-stdglue/stdglue.py .
echo 'from stdglue import *' >remap.py
echo 'import remap' >toplevel.py

3) edit /home/zdenek/linuxcnc/linuxcnc-2.9/configs/sim/gmoccapy/gmoccapy.ini
add:

[RS274NGC]
REMAP = M6  modalgroup=6  prolog=change_prolog ngc=atc epilog=change_epilog

[PYTHON]
TOPLEVEL=./python/toplevel.py
PATH_APPEND=./python

4) copy  ngc files to to folder /home/zdenek/linuxcnc/linuxcnc-2.9/configs/sim/gmoccapy/macros/
[atc.ngc.txt](https://github.com/LinuxCNC/linuxcnc/files/11311261/atc.ngc.txt)
[atc-down.ngc.txt](https://github.com/LinuxCNC/linuxcnc/files/11311262/atc-down.ngc.txt)
[atc-up.ngc.txt](https://github.com/LinuxCNC/linuxcnc/files/11311264/atc-up.ngc.txt)
remove .txt :-)

5) run linuxcnc gmoccapy.ini
6) home all axes
7) in MDI try:
M6 T1
M6 T2
M6 T3
....
[image]

Please be patient. On a physical machine, the error appears approximately once in 10 attempts, but on a virtual PC, this error appears approximately once in 40 attempts.

I wanted to make your work easier and write a program that would call M6 commands. Unfortunately, this error does not appear here. It is necessary to write M6 commands in MDI.

I have already cheered many times that I have removed the error with some modification of Remap, but in the end this message always appears.

zz912, 2023-05-05:
I made the same attempt with GUI AXIS. I executed the remapped command M6 T?? more than 100 times. The error did not appear. So I think the bug is with Gmoccapy when it works with AXIS.

gmoccapy, 2023-10-08:
I took a look in your subs and as I expected it seems to be an error in your config.
You do end all your subs with an M2 command, but in remap you should not use M2. You may take a look in gmoccapy sim folder within macros the change.ngc.

I was not able to reproduce the described behavior with two different remaps of mine.
Thats why I do close this issue.

closed 2023-10-08 by gmoccapy

zz912, 2023-10-08:
>You do end all your subs with an M2 command, but in remap you should not use M2.

Look here:
http://linuxcnc.org/docs/2.9/html/remap/remap.html
Every ngc file for remap has M2.

Caution | Never use an M2 in a O-word subroutine, including this one. It will cause hard-to-find errors. For instance, using an M2 in a subroutine will not end the subroutine properly and will leave the subroutine NGC file open, not your main program.
-- | --

If I understand correctly M2 cant be IN subroutine. I have M2 out of subroutine like in LCNC remap documentation.

> I was not able to reproduce the described behavior with two different remaps of mine.
> Thats why I do close this issue.

I have this problem:
- on my physical PC on the machine
- on my PC at home
- on a Virtual Machine

The reason why this topic should remain open is that the message itself is not ideal:

>0: not in a subroutine definition: ' '

What is ' ' ???

It would help if this message included the file name and line number where the error is.

gmoccapy, 2023-10-08:
Change the o<word> to lower letters. I had similar problem in one sub. Do not know why this helped, but may be a solution. Next try to comment out the M2 and test again.

Until I am not able to reproduce, I will not reopen the issue. But of course I will try to help you to get your system running.