LinuxCNC/linuxcnc #2450: Profibus support request

Alexfehr1987, 2023-04-26:
To use profibus Hardware I currently use pyprofibus. Unfortunatly its not realtime and needs an extra Script to start linuxcnc. As I saw modbus Implementation is worked on currently.
Wouldnt this be a good Feature to add profibus Support directly in linuxcnc?
There are many industrial grade components available with profibus(valve Islands, IO modules)

andypugh, 2023-04-26:
From a quick inspection, this looks plausible. 
Is the physical layer compatible with RS422 or RS485?

Alexfehr1987, 2023-04-27:
Yes its RS485. You can use the Serialport(old comport) directly or a USB to RS485 Adapter. I am currently running it with 6Mbit via USB to RS485 at 2-7ms delay. Profibus can run up to 12Mbit. Using the Mesa Serial could bring improvement delaywise.
https://github.com/mbuesch/pyprofibus is the repository i used.

andypugh, 2023-04-27:
I don't know how fast the Mesa UART can run. In smart-serial mode it runs at 2.5Mbit. 
I don't think that using the Mesa UART will be faster, and it will always take 3ms per packet.
However it will be realtime, which should be a major advantage.

Alexfehr1987, 2023-04-27:
Yes, realtime and the usage of a mesacard are the key features. Most Profibus Devices can run slower at 500kbit, 1,5Mbit too. The rate must be set from the master, e.g. 1,5Mbit or 3Mbit would work, 2,5Mbit not. I think for the application of most Profibus IO components 3ms is fast enough. Using Profibus encoders, which would require low latencyis, is probably not so common.

mbuesch, 2023-05-28:
There is pbmaster, which is a C implementation of Profibus DP. It still seems to be available on Sourceforge.
But it doesn't seem to have as wide spread use as pyprofibus, although pyprofibus being very incomplete and non-rt.
Not sure how hard it would be to adapt and integrate that into LinuxCNC. But it's probably worth a try.
I personally never looked at pbmaster in detail, though.

pcw-mesa, 2023-06-01:
The PktUART should be good up to about 25 Mbits/sec